openscad/openscad #2372: Difference on extrusion of difference results in wrong preview

themactep, 2018-04-07:
Sometimes I get weird preview of crosscut primitives in my models which then renders fine. I narrowed the case to this little sample. 
~~~
id = 30; // inner diameter
od = 40; // outer diameter

difference() {
  color("silver") tube3(); // <--- change tube type here
  color("tomato") rotate([90, 0, 0]) hole();
  color("orchid") rotate([0, 90, 0]) hole();
}

module tube1() {
  difference() {
    cylinder(d = od, h = id, center = true);
    cylinder(d = id, h = od, center = true);
  }
}

module tube2() {
  translate([0, 0, -(id / 2)]) difference() {
    linear_extrude(id) circle(d = od, center = true);
    translate([0, 0, -(od - id) / 2]) linear_extrude(od) circle(d = id, center = true);
  }
}

module tube3() { // <--- this one is broken
  translate([0, 0, -(id / 2)]) linear_extrude(id) difference() {
    circle(d = od, center = true);
    circle(d = id, center = true);
  }
}

module hole() {
  cylinder(d = (od / 2), h = id);
}
~~~

These are three approaches to create the same model. Two of them render fine while the last one displays impossible geometry in preview (screenshots attached). 

[image]
with tube1()

[image]
with tube2()

[image]
with tube3

Tested on Linux with different builds from stable 2015.03 to the most recent git snapshot.

t-paul, 2018-04-07:
That's a [convexity issue](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#Why_are_some_parts_(e.g._holes)_of_the_model_not_rendered_correctly?) I think. Changing `tube3()` to use `linear_extrude(id, convexity=3)`, the display issues are gone for me.

nophead, 2018-04-07:
You need to set convexity in the linear extrude.

themactep, 2018-04-09:
Why not to set a default convexity for linear_extrude to something higher than 1, then? linear_extrude turns a 2D object into a 3D object which means the new object will have at least two sides for a ray to intersect, front and back. Even value of 2 would fix the issue. OpenJSCAD, I believe, have default convexity set at 5.

kintel, 2018-04-09:
A higher convexity value would decrease preview performance. We _could_ however, implement some methods for automatically determine/guess convexity.
Since OpenJSCAD doesn't implement a preview mode, they don't have a concept of convexity.

t-paul, 2019-11-19:
Closing, issue seems solved.

closed 2019-11-19 by t-paul