openscad/openscad #2375: Noise when rendering complete #featurerequest

dorkmo, 2018-04-12:
I think it'd be nice if there was an optional microwave beep when a rendering is done. 

thanks for the great software

MichaelAtOz, 2018-04-12:
+1.
Surely it should be the sound of trumpets and angels singing, or something...user configurable, maybe one of the built in sounds, 'Fax Error' (Windows) is probably not used much any more.

JustVic, 2018-05-03:
I made a try to implement it in pulrequest #2393.

MichaelAtOz, 2018-05-06:
Note for future, when this is expanded to file based, we probably want to keep one built-in sound in case users do strange things to the installation.

MichaelAtOz, 2018-05-06:
I was thinking about what size budget we should target for sound files.
So I checked past distributions, then got carried away:
```
(64bit unless noted - this is the Program Files size) 
Version		Size-MB		Mainly Due To 
2013.06		23.1
2015.03-1\32	30.0		.exe, fonts (~4MB)
2015.03-2\64	29.7
2016.06.08	29.6
2016.07.20	30.8 		.exe
2016.09.18	31.1		.exe
2016.10.04\32	31.1		-
2017.01.20	31.4		.exe
2017.08.31	34.5		.exe
2017.10.28	34.7		.exe
2018.01.06	34.5		.exe
2018.04.01	34.6		~
2018.05.05	39.0		.exe (presumably sound, note .wav file is 0.345MB)

```

So presumably we can afford a small number of MB, say low target of 2MB initially???
That would allows 2 big sounds and a few mediums and some more tinys.
And I can resample some sounds to squeeze them in if required.

MichaelAtOz, 2018-05-09:
> And I can resample some sounds to squeeze them in if required.

I got 1 800K+ file (at the unusual 96000 rate) down to ~10K with little noticeable sound difference BTW.

Lenbok, 2018-05-20:
I think that rather than adding a special-purpose feature like this, it would be much more useful to provide a general purpose hook that could run any user specified command. If my rendering is taking ages, I'm likely to have actually wandered out of earshot of the computer. With a general hook I could send a notification to my phone (e.g. via pushover/pushbullet).

MichaelAtOz, 2018-05-20:
Thanks Len, feel free to add another feature request, that may be something the community wants too.
This issue has a Bounty attached, and work has been done, so this should complete as specified.

JustVic, 2018-05-28:
Can this issue be closed or Do We need to do something else?

t-paul, 2018-05-28:
No, I think everything in scope of this ticket is done and merged to master. I'm having trouble with getting the new build out on OBS but that's not related to the code, but seems to be an OBS issue. So Linux updates may need a day or too to be available.

closed 2018-05-28 by t-paul

MichaelAtOz, 2018-05-28:
@JustVic I approved the Bounty claim, the other backer is Anonymous, so I don't know how that works, whether it takes that into account, or has to timeout on 12th June.

JustVic, 2018-05-29:
@MichaelAtOz Thanks.