openscad/openscad #2376: Fedora 28: linear_exturde gives SIGABRT

hroncok, 2018-04-12:
After upgrading to Fedora 28, I have SIGABRTs when i do linear_exturde.

 1. type: linear_extrude() square();
 2. press F6

```
/usr/include/c++/8/bits/stl_vector.h:932: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = Eigen::Matrix<float, 3, 1>; _Alloc = Eigen::aligned_allocator<Eigen::Matrix<float, 3, 1> >; std::vector<_Tp, _Alloc>::reference = Eigen::Matrix<float, 3, 1>&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__builtin_expect(__n < this->size(), true)' failed.
SIGABRT (core dumped)
```

Reported bugs from users of Fedora:

 * https://bugzilla.redhat.com/show_bug.cgi?id=1564734
 * https://bugzilla.redhat.com/show_bug.cgi?id=1563897

There are useful attachments.

(I'll try a rebuild first to see if it wont fix this.)

hroncok, 2018-04-12:
[What I have upgraded when updated to Fedora 28](https://github.com/openscad/openscad/files/1902490/upgradelog.txt) (texlive packages grepped out).

t-paul, 2018-04-12:
First bug report is "Access denied". Please make sure the linked stuff is available, or I'll close this.

kintel, 2018-04-13:
Is this issue also present in our current master?

t-paul, 2018-04-18:
Bug report link is still "access denied". Please re-open once the information is publicly available.

hroncok, 2018-04-19:
I haven't noticed thet the bug was private. The user reporting this checked that box and I cannot make it public without the user acking that. The second bug is public.

Apparently I cannot reopen.

@kintel I'll try master.

kintel, 2018-04-19:
uh, sorry, this doesn't need to stay open (Assuming @hroncok that you're investigating the possibility for a Fedora-specific patch)

closed 2018-04-19 by kintel

kintel, 2018-04-19:
@hroncok Looks a bit similar to #1755

hroncok, 2018-04-20:
bisect says 684d283b1e090242681b66564e3570a4379686c1

kintel, 2018-04-20:
@hroncok That makes sense; looks like it was fixed by this commit: https://github.com/openscad/openscad/commit/2b3392266fe59c3eb32750469f34fddebd54a296

hroncok, 2018-04-24:
https://github.com/openscad/openscad/pull/2387

PR #2387 (2018-04-24): #1867 Fixed problem with vertices having an evil, downscaled-to-float…
…, twin, if a polyhedron contains occasional non-planar polygons

Backporting this fixes #2376