openscad/openscad #2377: OPENSCADPATH relative to opened .scad file (for project-specific libraries)

sadr0b0t, 2018-04-15:
Hello,

I want to have project-specific libraries in one dir in the same dir where .scad file is located.

For example, we have project1 with files:

project1/model.scad 
project1/modules/lib1/lib1.scad
project1/modules/lib2/lib2.scad

I want to import lib1 and lib2 libraries transparently without providing modules subdir name like:

~~~scad
use <lib1/lib1.scad>
use <lib2/lib2.scad>
~~~

currently I can use full relative path including "modules" dir name like:
~~~scad
use <modules/lib1/lib1.scad>
use <modules/lib2/lib2.scad>
~~~
But this looks dirty.

I have tried to use OPENSCADPATH env variable with relative path value to achieve that but it gives only partial result.

If I run openscad from console from project dir like this:
~~~
export OPENSCADPATH=modules
cd /home/user/projects/project1
openscad model.scad
~~~

I can see relevant library path in the build info dialog:
>OPENSCADPATH: modules
OpenSCAD library path:
/home/user/projects/project1/modules

and can include modules without providing "modules" dir name (this is exactly what I want).

But if I run openscad by clicking on the model.scad file in file manager, importing library without "modules" dir name willl NOT work

add to .profile or global env varialbles
~~~
export OPENSCADPATH=modules
~~~

Then logout/login (on Linux) and open .scad file from file manager

>OPENSCADPATH: modules
    OpenSCAD library path:
    /home/user/modules

So, OpenSCAD will not use OPENSCADPATH value relative to current project dir in this case, it will attach it to $HOME dir value at startup, so "include <lib1/lib1.scad>" without "modules" will not work.

The use-case here is to have nodejs-like module dependency management.

There are at least 2 projects
nodejs npm-based: https://github.com/RobertFach/Openscad-Modules
ruby gems-based: https://github.com/lostapathy/scad_bundler#draft-conventions

The 1st one suggests to import libraries with full relative path like "use <node_modules/lib1/lib1.scad>" (which looks quite dirty to me), the 2nd one suggest to run openscad with script that sets OPENSCADPATH (which is quite inconvenient).

So my request here is:
- either hardcode project-related library dir name "modules"/"scad_modules"/"lib"/"libraries" or any other, so OpenSCAD would also search modules inside this dir if it exists in current project (like nodejs does with "node_modules" dir, for example)
- or allow OPENSCADPATH to be relative to currently opened scad file

t-paul, 2018-04-15:
`OPENSCADPATH` allows multiple path entries, similar to Unix `PATH`, so for example `OPENSCADPATH=/tmp/project1/modules/lib1:/tmp/project1/modules/lib2 openscad-nightly project1/model.scad` does work.

It might still make sense to think about allowing path relative to the top level .scad file. The information to handle that case should be there I think.

sadr0b0t, 2018-04-19:
Hello, I have created a small patch which handles this feature and adds something more.

The 1st simple fix for the relative path in OPENSCADPATH is here:

~~~patch
void parser_init()
...
-			add_librarydir(fs::absolute(fs::path(boost::copy_range<std::string>(*it))).string());
+			add_librarydir(fs::path(boost::copy_range<std::string>(*it)).string());
~~~

Just remove fs::absolute call whe add dirs from OPENSCADPATH and things will work as I have described above:

In a project like
project1/model.scad
project1/modules/lib1/lib1.scad
project1/modules/lib2/lib2.scad

one will be able to import lib1 and lib2 like:

~~~openscad
use <lib1/lib1.scad>
use <lib2/lib2.scad>
~~~

if OPENSCADPATH is set to 'modules' (or contains it among multiple values)
~~~
OPENSCADPATH=modules
~~~

But after checking this option I have come across another issue with a bit more complex usecase.

Imagine, we still have project
project1/model.scad
project1/modules/lib1/lib1.scad
project1/modules/lib2/lib2.scad

model.scad references lib1.scad like above
~~~openscad
use <lib1/lib1.scad>
~~~

and lib1.scad, in turn, references lib2.scad also in a similar way:
~~~openscad
use <lib1/lib2.scad>
~~~

This is common case if we use nodejs-like modules which depend on each other and refer to each other while being loaded to the same plain node_modules dir.

But with current OpenSCAD library load strategy this will NOT work, because while being included from lib1.scad, lib2.scad should be located in the same dir as lib1.scad or in a subdir or somewhere in system libraries.

https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Libraries
> In the case of a library file itself having use <...> or include <...> the directory of the library .scad file is the 'calling' file, i.e. when looking for libraries within a library, it does not check the directory of the top level .scad file.

But if we change library search strategy:

once lib2.scad is included from lib1.scad like:
~~~openscad
use <lib1/lib2.scad>
~~~

first, try to search in relatively to lib1.scad dir in:
/home/user/project1/modules/lib1/modules/lib2/lib2.scad (NO)
if not found, then do the same for parent dir:
/home/user/project1/modules/modules/lib2/lib2.scad (NO)
and recursively until we reach root or find library:
/home/user/project1/modules/lib2/lib2.scad (YES)

in case, we did not find it above:
/home/user/modules/lib2/lib2.scad (NO)
[... 84 lines cut ...]
-			add_librarydir(fs::absolute(fs::path(boost::copy_range<std::string>(*it))).string());
+			add_librarydir(fs::path(boost::copy_range<std::string>(*it)).string());
 		}
 	}
 
 	add_librarydir(PlatformUtils::userLibraryPath());
 
 	add_librarydir(fs::absolute(PlatformUtils::resourcePath("libraries")).string());
+	
+	add_librarydir("scad_modules");
 }
diff --git a/src/parsersettings.h b/src/parsersettings.h
index f1a9db1..4b8c04e 100644
--- a/src/parsersettings.h
+++ b/src/parsersettings.h
@@ -12,7 +12,7 @@ extern int parser_error_pos;
  */
 void parser_init();
 
-fs::path search_libs(const fs::path &localpath);
+fs::path search_libs(const fs::path &localpath, const fs::path &sourcepath);
 fs::path find_valid_path(const fs::path &sourcepath, 
                          const fs::path &localpath, 
                          const std::vector<std::string> *openfilenames = nullptr);
~~~

MichaelAtOz, 2018-04-21:
I'm wondering if a source based 'scad_modules' concept would be helpful?
```
//main.scad
scad_path <./../../modules> // prepend to search paths 
use <lib1/lib1.scad>
use <lib2/lib2.scad>
```

sadr0b0t, 2018-04-21:
NodeJS modules architecture (ror reference):
https://nodejs.org/api/modules.html#modules_loading_from_node_modules_folders

tsingi, 2018-04-22:
That works for me.  If I source OPENSCADPATH='../' it finds my library where I like to have it.  if I make a lib dir in my project and source OPENSCADPATH='../:lib/' it searches the lib directory too.

Thanks sadr0b0t.

Would it be worthwhile to inform the user in console messages what library was linked?  It works very well but even working properly it could do something the user did not intend.

lostapathy, 2018-04-23:
> The 1st one suggests to import libraries with full relative path like "use <node_modules/lib1/lib1.scad>" (which looks quite dirty to me), the 2nd one suggests to run openscad with script that sets OPENSCADPATH env value (which is quite inconvenient).

There's a lot more to scad_bundler than just setting OPENSCADPATH - that's just how it tells OpenSCAD about the libraries.

The bigger utility is that it manages versioned dependencies, so that libraries are able to change over time without breaking existing code.  You're able to have multiple versions of a library installed on your system at once and different OpenSCAD projects can use different versions of the libraries.  This and #2384 really don't seem to address that aspect at all.

sadr0b0t, 2018-04-23:
> that's just how it tells OpenSCAD about the libraries.

yes, that's exactly what I say.

>The bigger utility is that it manages versioned dependencies [...] This and #2384 really don't seem to address that aspect at all.

This patch is not intended to become another package manager, this patch is only about that OpenSCAD will be able to search and load libraries in a package-manager-friendly way without any additional pre-configurations or startup scripts.

doug-moen, 2018-04-23:
@MichaelAtOz asks: "How will this affect the future namespace developments? Perhaps @doug-moen may have a view?"

In my opinion, package management would work better if there were language extensions to support it. I like the way package management works in Rust. The two language extensions that I think would be beneficial are:
* Hierarchical namespaces.
* The ability to directly import a specific version of a specific package, using a URL and a version #, like you do in a Rust cargo.toml file.

For example, suppose that we had some syntax like this:
```
package arch = "https://github.com/doug-moen/architecture-scad#1.0";
```
This defines `arch` to be a namespace that is bound to version 1.0 of the architecture-scad package. Then you use dotted names to invoke modules like `arch.ogee` and `arch.flying_buttress`.

(Note that I don't have a fully speced out proposal for package management in OpenSCAD, I'm just suggesting a future direction. I do have a spec for package management in Curv, and it does work the way I'm describing.)

I think this kind of approach would be simpler and more robust than using OPENSCADPATH to locate packages that are stored in the file system. I'm a bit worried about supporting an arbitrary mixture of absolute and relative pathnames in OPENSCADPATH. I think it leads to chaos if different authors require OPENSCADPATH to be customized in different ways in order to use their packages.

MichaelAtOz, 2018-04-23:
>> recursively until we reach root
>
>I think this could have undesirable nondeterministic outcomes.

I still think this needs debate. Do you really want it picking up any-old-odd file in the volume?
Also if you do a typo in the library name, it will search the whole disc hierarchy.
On Windows that would take forever.

sadr0b0t, 2018-04-24:
>I think it leads to chaos if different authors require OPENSCADPATH to be customized in different ways in order to use their packages.

The user will just drop libraries to "project/scad_modules" dir and they will all start to see each other as if they were dropped to $HOME/.local/share/OpenSCAD/libraries .

>Also if you do a typo in the library name, it will search the whole disc hierarchy.
On Windows that would take forever.

It will not search the whole disk, it will only search from current project dir up to fs root.

For example, on Windows you have project in:
c:\Users\user1\Documents\myproject\myopenscadprojects\myproject1\mysubproject1\mything.scad

Let mything.scad include lib1.scad, which does not exist

The search will go as:
c:\Users\user1\Documents\myproject\myopenscadprojects\myproject1\mysubproject1\scad_modules\lib1.scad (NO)
c:\Users\user1\Documents\myproject\myopenscadprojects\myproject1\scad_modules\lib1.scad (NO)
c:\Users\user1\Documents\myproject\myopenscadprojects\scad_modules\lib1.scad (NO)
c:\Users\user1\Documents\myproject\scad_modules\lib1.scad (NO)
c:\Users\user1\Documents\scad_modules\lib1.scad (NO)
c:\Users\user1\scad_modules\lib1.scad (NO)
c:\Users\scad_modules\lib1.scad (NO)
c:\scad_modules\lib1.scad (NO)
lib1.scad NOT found

This is 8 checks file.exist() for any library. Would run instantly on any OS.

sadr0b0t, 2018-04-24:
>tsingi
> Would it be worthwhile to inform the user in console messages what library was linked? It works very well but even working properly it could do something the user did not intend.
> Not sure how to extract the load path but just putting that in there would do it.

If you want to show load path for every found library, try this patch:

~~~diff
diff --git a/src/parsersettings.cc b/src/parsersettings.cc
index e8164c7..bbc003c 100644
--- a/src/parsersettings.cc
+++ b/src/parsersettings.cc
@@ -111,7 +111,10 @@ fs::path find_valid_path(const fs::path &sourcepath,
                if (fs::exists(fpath)) fpath = boosty::canonical(fpath);
                if (check_valid(fpath, openfilenames)) return fpath;
                fpath = search_libs(localpath, sourcepath);
-               if (!fpath.empty() && check_valid(fpath, openfilenames)) return fpath;
+               if (!fpath.empty() && check_valid(fpath, openfilenames)) {
+                       PRINTB("INFO: load lib: %s", fpath);
+                       return fpath;
+               }
        }
        return fs::path();
 }
~~~

tsingi, 2018-04-24:
Thanks dude.  I had something like that without the check.
Some trouble sorting out the % syntax of the macro.

jordanbrown0, 2020-11-13:
2¢:  Do the same as C double-quote style:  search the same directory that the immediately-including file is in, then search a specified path.  (Probably that path would be partially user-specified and partially system-defined.)

That's a simple and well-understood rule.  It allows libraries to have "private" include files without colliding with other components.

uDude, 2021-09-17:
ditto for @jordanbrown0 

If not changing semantics for either  <path_spec> or "path_spec" then at least add  getcwd() into the library path as the file managers I have checked set cwd to the directory containing the .scad file when double clicked.