openscad/openscad #2378: Previewing doesn't work after different issues

kwendenarmo, 2018-04-15:
I'm developing a project where I need a lot of 3D design, and I found that my version of OpenSCAD stopped previewing with some frequency.
OpenSCAD was previously running, although I can't determine what has changed since then, or if it's the same version. So I decided to compile the master branch again and install it, in case it fixed the problem. It didn't fix it, but it has allowed me to detect the causes that cause it to enter this error state.
I have recently found that the same thing happens to me on my work computer.

When it enters this mode, the only way out is to close the program. The preview doesn't usually work again, but it can still be rendered. It is very similar to what was discussed in Issue #1727. Once it stops previewing, most of the time it doesn't do it again and in fact, the small color coordinate axis of the corner is also no longer displayed.
The times it does preview again, it skips in the preview the items that are part of differences or intersections. And if it previews again, it is because there are no differences or intersections in the part that was previewing.
Whenever you try to preview again, you see how the console information is regenerated, but nothing is shown in the design area, apart from the scale.

It has taken me hours of frustration to design several iterations of the model with these circumstances.

The reasons I have found this to be the case are as follows:
-  After a syntax failure, which happens to me frequently because my laptop keyboard is quite deteriorated.
-  After rendering. After that, it never previews again.
-  I just discovered it right now: After a preview failure, if another OpenSCAD instance is opened (File -> New), and a syntax failure occurs that causes a preview failure in this new instance, and you go back to the first instance, then the preview works again in both instances. Weird. And if syntax errors are made in this first instance, it ignores them and continues to preview them. Even weirder.

Example codes with which this happens:
```
// This previews correctly.
difference() {
    cube([10,10,10]);
    translate([7,7,7]) cube([5,5,5]);
}
```
Remove the last semicolon to produce a syntax error. When you put it back on, stop previewing.

```
$fn=36;
difference() {
    sphere(10);
    cylinder(r=4, h=22, center=true);
    rotate([90,0,0]) cylinder(r=4, h=22, center=true);
    rotate([0,90,0]) cylinder(r=4, h=22, center=true);
}
```
Render the model, once it has been rendered, it does not preview again. (I don't know if the complexity of the piece has anything to do with it, because with the previous example of the cubes it still previews.)

About my system:
```
OpenSCAD Version: 2018.03.25
System information: Linux 4.10.0-42-generic #46~16.04.1-Ubuntu SMP Mon Dec 4 15:57:59 UTC 2017 x86_64 Linux Mint 18.1 8 CPUs 15.58 GB RAM
Compiler: GCC "5.4.0 20160609" 64bit
Boost version: 1_58
Eigen version: 3.2.92
CGAL version, kernels: 4.7, Cartesian, Extended_cartesian, Epeck
OpenCSG version: OpenCSG 1.4.0
Qt version: 5.5.1
QScintilla version: 2.9.1
MingW build: No
GLib version: 2.48.2
libzip version: 1.0.1
Application Path: /usr/local/bin
Documents Path: /home/kwendenarmo/.local/share
Resource Path: /usr/local/share/openscad
User Library Path: /home/kwendenarmo/.local/share/OpenSCAD/libraries
User Config Path: /home/kwendenarmo/.config/OpenSCAD
Backup Path: /home/kwendenarmo/.local/share/OpenSCAD/backups
OPENSCADPATH: 
OpenSCAD library path:
/home/kwendenarmo/.local/share/OpenSCAD/libraries
/usr/local/share/openscad/libraries

OPENSCAD_FONT_PATH: 
OpenSCAD font path:
GLEW version: 1.13.0
OpenGL Version: 3.0 Mesa 17.2.8
GL Renderer: Mesa DRI Intel(R) Haswell Mobile 
GL Vendor: Intel Open Source Technology Center
RGBA(8880), depth(0), stencil(0)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
[... 216 lines cut ...]
GL_KHR_robust_buffer_access_behavior
GL_KHR_robustness
GL_MESA_pack_invert
GL_MESA_shader_integer_functions
GL_MESA_texture_signed_rgba
GL_MESA_window_pos
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_depth_clamp
GL_NV_light_max_exponent
GL_NV_packed_depth_stencil
GL_NV_primitive_restart
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_NV_texture_env_combine4
GL_NV_texture_rectangle
GL_OES_EGL_image
GL_OES_read_format
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_border_clamp
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays
```

t-paul, 2018-04-15:
Sounds like a duplicate of  #1875 - see also http://forum.openscad.org/Object-appears-in-F6-render-but-not-F5-preview-td23809.html

kwendenarmo, 2018-04-15:
Thank you, looking at the other issue you mentioned, I noticed that even though it is a git-master version and recently compiled, it still uses an older version of OpenCSG.
I assumed that this dependency was included in the repository and updated, or that my system would have updated it from its package repositories. so I'd overlooked this.

To update OpenCSG you need to download the latest sources from http://opencsg.org/#download, and compile and install OpenCSG (`make && sudo make install`). It may not compile because it is missing `gl/glut.h`. This is fixed by installing the package `freeglut3-dev`.
Having done this, I downloaded OpenSCAD from git-master and compiled it again.
Once compiled and installed again, it' s done!
```
OpenSCAD Version: 2018.04.15
System information: Linux 4.10.0-42-generic #46~16.04.1-Ubuntu SMP Mon Dec 4 15:57:59 UTC 2017 x86_64 Linux Mint 18.1 8 CPUs 15.58 GB RAM
Compiler: GCC "5.4.0 20160609" 64bit
Boost version: 1_58
Eigen version: 3.2.92
CGAL version, kernels: 4.7, Cartesian, Extended_cartesian, Epeck
OpenCSG version: OpenCSG 1.4.2
Qt version: 5.5.1
QScintilla version: 2.9.1
MingW build: No
GLib version: 2.48.2
libzip version: 1.0.1
Application Path: /usr/local/bin
Documents Path: /home/kwendenarmo/.local/share
Resource Path: /usr/local/share/openscad
User Library Path: /home/kwendenarmo/.local/share/OpenSCAD/libraries
User Config Path: /home/kwendenarmo/.config/OpenSCAD
Backup Path: /home/kwendenarmo/.local/share/OpenSCAD/backups
OPENSCADPATH: 
OpenSCAD library path:
/home/kwendenarmo/.local/share/OpenSCAD/libraries
/usr/local/share/openscad/libraries

OPENSCAD_FONT_PATH: 
OpenSCAD font path:
GLEW version: 1.13.0
OpenGL Version: 3.0 Mesa 17.2.8
GL Renderer: Mesa DRI Intel(R) Haswell Mobile 
GL Vendor: Intel Open Source Technology Center
RGBA(8880), depth(0), stencil(0)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes

Qt graphics widget: QOpenGLWidget
QSurfaceFormat: RGBA(8880), depth(0), stencil(0)

GL Extensions:
GL_3DFX_texture_compression_FXT1
GL_AMD_conservative_depth
GL_AMD_draw_buffers_blend
GL_AMD_seamless_cubemap_per_texture
GL_AMD_shader_trinary_minmax
GL_ANGLE_texture_compression_dxt3
GL_ANGLE_texture_compression_dxt5
GL_APPLE_object_purgeable
GL_APPLE_packed_pixels
GL_ARB_ES2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
[... 191 lines cut ...]
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_depth_clamp
GL_NV_light_max_exponent
GL_NV_packed_depth_stencil
GL_NV_primitive_restart
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_NV_texture_env_combine4
GL_NV_texture_rectangle
GL_OES_EGL_image
GL_OES_read_format
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_border_clamp
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays
```

Anyway, it would be nice to update the list of dependencies in the README, as well as in the `check-dependencies.sh` script, which continues to validate from 1.3.2.
Maybe add an explicit note to that it may be necessary to manually update OpenCSG.

Does this mean that the original problem has been fixed?
No. No. It continues to fail in the same way as above.

t-paul, 2018-04-15:
The version number shown is the library it's compiled against, so if you still have the one from the distribution installed, it might pick up that at runtime still.
Can you check via `ldd openscad | grep opencsg` where the library is actually loaded from?

kwendenarmo, 2018-04-15:
Of course.

```
ldd `which openscad` | grep opencsg
libopencsg.so.1 => /usr/local/lib/libopencsg.so.1 (0x00007f556134b000)
```
Point to the library I compiled today, the latest version:
```
lrwxrwxrwx  1 root root    19 abr 15 19:55 libopencsg.so.1 -> libopencsg.so.1.4.2
```

t-paul, 2018-04-15:
I'm running out of ideas. I can't reproduce the issue on my system, which is not that different just a bit newer as it's running Debian/Testing.

I also tried the [AppImage](https://download.opensuse.org/repositories/home:/t-paul/AppImage/OpenSCAD-Nightly-latest-x86_64.AppImage) which also did not show the issue. Would you give that a try to compare results with the same binary?

kwendenarmo, 2018-04-15:
Interesting, running the AppImage all seems to work fine. I will test it more intensively soon.
Btw, has the color set in the rendering been changed? The lower (-z) part of the preview appears much darker from the AppImage.

kwendenarmo, 2018-04-15:
At least it seems to be working fine. I'll try it in the next few days and tell you something. Anyway, any idea why my compiled version wasn't working properly?

Philius, 2018-11-02:
I've been getting preview issues for over a month.
I'm using nightly on Debian Stretch amd64.
In case this helps the AppImage I just downloaded is
OpenSCAD-Nightly-0-Build2531.1.glibc2.14-x86_64.AppImage
Animation still blanks out if I change values in the Customizer.
[appimage-library-info.txt](https://github.com/openscad/openscad/files/2543319/appimage-library-info.txt)
[nightly-library-info.txt](https://github.com/openscad/openscad/files/2543332/nightly-library-info.txt)
A cursory glance shows that some libraries in the AppImage are older than those installed on my system.
The issues happen if I use Intel or NVidia through optirun.

t-paul, 2018-11-02:
@Philius can you please be a bit more specific? What issues? Exactly what is described by the OP? Those happen on Stretch or with the AppImage or with both?

As discussed above, one of the important things is OpenCSG which specifically is built for the AppImage to be at version 1.4.2. We can't really have the means to fix that in all older distributions, that would need to be handled by the respecitive maintainers.

MichaelPFrey, 2018-11-03:
> Animation still blanks out if I change values in the Customizer.

Customizer and animations are not really optimized for each other, as both try to influence variables and recompile the design. Recompiling the design requires rebuilding the customizer. I mean: That could be tweaked, but there is the old feature, that you can change the code while the animation is running - and this is where things start to get painful.

t-paul, 2018-11-03:
If the interaction between animation and customizer is the only issue, that seems pretty unrelated to everything discussed here, neither of the two things were mentioned before. Hence my questions to be a bit more specific.

We should have customizer/animation interaction as separate issue with a clear description and not mixed up in a different discussion.

Philius, 2018-11-06:
@t-paul [hypocycloid.scad.gz](https://github.com/openscad/openscad/files/2554222/hypocycloid.scad.gz)
Curious that .scad uploads aren't supported by github :(
Version 2018.11.05.nightly (git e1f043d)
Anyway I open hypocycloid.scad and the screen shows the axes and rulers but no model.
I press reset (in the customizer) and the model appears.
I press animate and the model disappears.
I enter a frame rate and steps and the console flashes but no model animation.

t-paul, 2018-11-06:
The library info of your nightly shows OpenCSG 1.4.0, this version has issues with the drawing buffers, so there's not much to do about this. The issues described in the original post seem to be caused by that.

If your problem shows with the AppImage too (which has OpenCSG 1.4.2), please add your info to the new ticket (#2548) @MichaelPFrey created specifically about interaction between Customizer and Animation.

AndKe, 2018-12-06:
same issue here, highly frustrating.   Always one error away from forced restart.
(openscad-nigtly on snap)

t-paul, 2018-12-06:
Snap (on [snapcraft.io](https://snapcraft.io/openscad-nightly)) builds against "Ubuntu Core 16" which has OpenCSG 1.4.0. There might be a way to also build an updated OpenCSG library.

t-paul, 2019-02-19:
Looks like all cases are caused by older OpenCSG with known bug, so I'm closing this here. If there are new reports with OpenCSG >= 1.4.2, please open a new issue.

closed 2019-02-19 by t-paul