openscad/openscad #2380: Malware in latest Windows snapshot?

aricblunk, 2018-04-17:
Hi team, thanks for your hard work, I've really been enjoying Openscad. However, I went to download your latest windows x64 portable snapshot, 2018.04.06, but windows defender instantly removes it. I had virustotal scan the file, and none of the 67 AVs on there detected it, but windows defender is claiming to detect it using machine learning. Can you please security audit your build system? I'd like to try out the latest version :)

File in question:
http://files.openscad.org/snapshots/OpenSCAD-2018.04.06-x86-64.zip

Virustotal results:
https://www.virustotal.com/#/url/0c0e9efaa334d41aa442d0db8730e394d894fad336f4c4a8e6f39bc0ea1acdd3/detection

Windows defender detection:
https://www.microsoft.com/en-us/wdsi/threats/malware-encyclopedia-description?name=Trojan%3aWin32%2fSprisky.U!cl&threatid=2147719836

Screenshot of detection:
https://lh3.googleusercontent.com/-RG8LazGByaA/WtYAV-7Y9WI/AAAAAAAA5xM/BpVTMLH7T8ofn9H9_lgSv_B61iR6xctyQCHMYCw/s0/ApplicationFrameHost_2018-04-17_10-10-33.png

kintel, 2018-04-17:
@aricblunk I'm quite confident this is a bug in Windows defender. Could you file a bug report with them?

aricblunk, 2018-04-17:
It looks like the detection was for the file openscad.com in the archive. I downloaded it on my linux machine, moved out that file, re-compressed it, and redownloaded it on windows, and it went fine. Additionally no engines on virustotal detect anything in the exe. The program appears to work just fine without the .com file, it renders scripts and exports them to stl files which open fine in Cura. Easy enough fix I guess.

kintel, 2018-04-17:
The `openscad.com` executable is for cmd-line usage. The source code is here: https://github.com/openscad/openscad/blob/master/winconsole/winconsole.c

If it works for you, I'll just close this. Feel free to report this your virus checker vendor if it becomes annoying.

closed 2018-04-17 by kintel