openscad/openscad #2383: Feature Request: Add layer= for import() of SVG files.

MichaelAtOz, 2018-04-20:
Not urgent, can be worked around handraulicly , but would be useful.

MichaelAtOz, 2018-04-20:
See files [here](https://github.com/jacoburge/TouchPad/tree/master/Overlays).
Open *.svg in Inkskape, `Layer/Layers...`

kintel, 2018-04-20:
Ah, I see; "layer" is an Inkscape concept. In svg they have to use Inkscape-specific attributes to distinguish between Inkscape layers and svg `<g>` objects.

MichaelAtOz, 2018-04-20:
There is a looong discussion here: https://github.com/w3c/svgwg/issues/68 with some background, it doesn't get to a point. I'm still looking.

MichaelAtOz, 2018-04-20:
> Do we need other filter mechanisms that are more SVG-centric?

Well not for me just yet. But having a look at the spec it seems there could be a use for something equivalent to `dxf_dim()`.

tsingi, 2018-04-20:
Just as a forward, you could add this into an SVG image with metadata and then use that as data for post processing.  This doesn't help unless you are willing to had code the SVG, and of course teach OpenSCAD to recognize it.  It might be worth investigating a plugin for InkScape.  I wrote one once to clean things up a bit, don't have that code any more but I recall that it wasn't difficult.  I stopped using InkScape a while back when it started crapping polylines for various operations that it handled well previously.  I don't know if it still does that, I did report it.

SVG does not have layers, it has groups but they are not treated like layers.

<?xml version="1.0" encoding="UTF-8" standalone="no"?>
<svg xmlns="http://www.w3.org/2000/svg">
	<rect fill="green" x="100" y="20" width="200" height="40"/>
    <g id="group_1" stroke="red" transform="translate(200,200) scale(30)">
        <circle r="4"/>
        <polygon points="1,2 3,4 6,5 4,3"/>
    </g>
</svg>

That's it.  Groups are for stringing items together that are treated as a group.  This is similar to OpenSCAD.  SVG uses a painter's model, it's linear.  (The SVG root element has group semantics)

  Layers, as in DXF, are a flag, the object can be anywhere in the drawings and it is always flagged as belonging to a specific layer.  It's required.

As I said above, it might possible to simulate layer semantics to SVG using metadata but intrinsically SVG has no layers and no concept of layers.  I do know that the SVG working group has tossed the idea around so it is possible that SVG may have a layer concept in the future.

Now since SVG is 2d and there is no concept of layers it seems to me that the most useful thing you might want to do is describe a 2d polyline for extrusion.  How you position that data and reference that data are things that would be very easy to implement but difficult to make intuitive to an author.

SVG is designed for page display graphics even though it does have possibilities for engineering.  The engineering drawing community wasn't interested in being represented on the group.  It's basically an open source PDF but more geared towards geometry.

Continuing on with the topic of importing SVG, if I load the example above import("example1.svg"); I don't see anything even though the viewer shows one object.  What should it do?  I'd expect it to show up as a 2d geometry does, all of the objects in that file will translate to 2d geometry objects.

I'm good at hand coding and generating SVG so I'm interested in seeing SVG import work.  I'm also one of the authors of the SVG spec.  I could do things with this feature.

Let me know how I can help move it forward.

I've attached example1.svg, an image of it and an scad file that does nothing but import it.

[svg_import.zip](https://github.com/openscad/openscad/files/1933396/svg_import.zip)

t-paul, 2018-04-20:
I would prefer if there's a general concensus about layers, but from that linked issue it seems every application has it's own way of specifiying this and not much chance of changing that anytime soon. Talking open source, it would make some sense to look at Inkscape and just ignore the others. That's not ideal but for OpenSCAD the layer would basically work as an import filter.

But yes, the main problem with the existing (experimental) SVG import is that it's a graphics and not a CAD format. So it will never work 100% for all files. I think that's a limitation we can easily live with, the import is still a very useful feature.

tsingi, 2018-04-20:
You could start with a Z=0 import of a 2d shape, ignoring anything that doesn't translate well.  Text can be exploded into polylines in InkScape.  I suspect this would cover 90%+ of what anyone wants to do with it.

Regarding creation tools, I avoid Illustrator, not only is it prohibitively expensive, it likes to put proprietary stuff in SVG that only Illustrator recognizes.  I don't have experience with other creation tools but if they're free I can take a look at them.

t-paul, 2018-04-20:
That's what we are doing right now, some (ancient) results of the test suite are still on the file server: http://files.openscad.org/svglibtest/index-spec.html - including some random samples from openclipart.

tsingi, 2018-04-20:
Nice test suite!  I created the initial test frame and a lot of the tests when I was on the working group.
Does the suite answer to your requirements?  I can write some targeted tests easily enough.

MichaelAtOz, 2018-04-21:
> I would prefer if there's a general concensus about layers, but from that linked issue it seems every application has it's own way of specifiying this and not much chance of changing that anytime soon. Talking open source, it would make some sense to look at Inkscape and just ignore the others. 

I wrote the following before looking at some SVG source.
> From the linked issue, 5 of 6 use `<g id="layer..."` (in the forms `Layern`, `layern`, `Layer_n`, `Layer<space>n`), disregarding caps, and generecising to 'layer' & '?n', is a fallback mechanism to be able to select by layer number.

Unfortunately, viewing an Inkscape SVG source, it uses `<g id="Layer_n"...` (without the `inkscape:label="Layer n" inkscape:groupmode="layer"`) in other contexts. This one is caps and the other is not, but I don't think that is something to hand a hat on.

So, yes, lets support the Inkscape method, used by two of the above. Gradually establishing an 'industry' standard ;)

tsingi, 2018-04-21:
The inkscape:label="Layer n" was what I meant by using metadata, yes, that is the way to do it.

I'm confused about the logic of using <g id="Layer_n"... though.  This concept is disassociating layers from groups, alternatively, if the id is meant to be used for layer identification, it is illegal to use a duplicate id and it will break scripting at the level of XML.  Anyway, the metadata is the way to go.

MichaelAtOz, 2018-04-21:
> I'm confused about the logic of using <g id="Layer_n"... though. 

If you look in https://github.com/w3c/svgwg/issues/68, referenced above, you will see how 5 programs other than Inkscape apparently do 'layers'. 
I was musing on a way to allow a more wider generic support for 'layers', using just 
`<g id="layer<something>"`, but it appears that SVG groups with `id="layer<something>"` are used for other than 'layers', at least in the two simple Inkscape files I looked at. So that is a NOGO.

> illegal to use a duplicate id

I was using Layer_`n` to represent a generic `n` e.g. Layer_1 Layer_2 etc. Not duplicate Layer_n

MichaelAtOz, 2018-04-21:
@tsingi 
> Continuing on with the topic of importing SVG, if I load the example above import("example1.svg"); I don't see anything even though the viewer shows one object. What should it do? I'd expect it to show up as a 2d geometry does, all of the objects in that file will translate to 2d geometry objects.

You need a recent development [snapshot](http://www.openscad.org/downloads.html#snapshots), and enable the experimental feature in preferences:
[image]

Then you get 2D like: (note - it looks like 1 unit high 3D, but that is just the way preview shows it)
[image]

tsingi, 2018-04-21:
> I was using Layer_n to represent a generic n e.g. Layer_1 Layer_2 etc. Not duplicate Layer_n

My point there was that using "layer" id's on groups is antithetical to the semantics of disassociating groups from layering.  Confusing because groups and layers would be independent associations.  On the other hand, if the author can group all layer objects and actually organize the objects that way the process is simpler.  But it's not using the layer concept in that case.  However my discovery of what InkScape actually does means this is what is being done...

> Also note that apparently you can have nested 'layers' in Inkscape. I haven't done that yet.
How to import...thinking cap on...gears are still grinding...

That is how groups behave.  Curious.  I should a look at layering in Inkscape.  Looking at the manual...

    *Internally, Layers are just SVG Groups with a few extra Inkscape specific parameters that Inkscape uses to control the Layer interface. Like Groups, Layers can contain sub-Layers.*

So that is what they are doing, attempting to apply layer semantics to groups.  That doesn't change anything really but it might make it easier to understand by an author who thinks 'layers' not 'groups'  It's the SVG painter's model though and not the CAD concept of layers.  Still, we can work with that and it means that *giving a group an ID of "layer_1" is not only appropriate it's explicitly correct*.  I should have looked at it before.  Sorry.  :)  It will raise questions and likely ire from people who want (correctly) to put disparate objects in a layer.  You can't do that.

Mean time, Michael, is it prohibitively difficult to set up a dev build on Ubuntu?  I only see a master branch in the repo.

MichaelAtOz, 2018-04-22:
Let me preface that I'm no Inkscape expert. Inkscape also has objects and object grouping (not named like layers tho). I haven't looked but presume they also use '<g' groups, in the GUI I was able to select objects from two layers and group them. I'll have to see how that is done in SVG. However that is somewhat off topic for this issue (maybe).

> Mean time, Michael, is it prohibitively difficult to set up a dev build on Ubuntu? I only see a master branch in the repo.

Sorry, I'm Windows. @t-paul perhaps can point you to the doco. Reading between the lines here, didn't sound too difficult.

@kintel 
>  Do we need other filter mechanisms that are more SVG-centric?

I did have a thought about `display: none`.
This is a snip of an Inkscape `<g` layer:
```
  <g
     inkscape:label="Print"
     inkscape:groupmode="layer"
     id="layer1"
     style="display:inline"
     sodipodi:insensitive="true">
    <g
       id="g100"
       ...
```
In Inkscape you can hide a layer, then it is `style="display:none"`.
If that was supported by `import()` by default, that could perhaps be an interim means of importing a specific layer, by hiding all others in Inkscape.
I doubt that any program that sets a `<g style="display:none"` would expect that to be imported. ?

MichaelAtOz, 2018-04-22:
> an interim means of importing a specific layer

BTW, ATM it is pretty easy to save a new copy, delete all unwanted layers & import(). Which is why I said 'Not urgent' in the OP.

MichaelAtOz, 2018-04-22:
> <g style="display:none"

Firefox & IE do not display hidden layers.
So I'm suggesting that should be the import() behaviour too.
Unless someone objects, I'll raise a separate issue for that tomorrow.

tsingi, 2018-04-22:
> Let me preface that I'm no Inkscape expert. Inkscape also has objects and object grouping.

[SVG g element](https://www.w3.org/TR/SVG/struct.html#Groups)

It's in the spec.

> However that is somewhat off topic for this issue (maybe).

Well, InkScape, as I said above, is using groups to represent layers.  So they aren't really layers.

> In Inkscape you can hide a layer, then it is style="display:none".

display = "none" requires that the user agent treat that part of the tree as of it weren't there.

> I doubt that any program that sets a <g style="display:none" would expect that to be imported. ?

The proper behaviour is to not render those elements, in any user agent, so yes, it must be ignored.

[*A value of display: none indicates that the given element and its children shall not be rendered directly (i.e., those elements are not present in the rendering tree.*](https://www.w3.org/TR/SVG/painting.html#DisplayProperty)

@t-paul 
Can you tell me how to turn on debugging in the compile?  I found a debug target in openscad.pro but I don't know how to call it.  I put it inline and it didn't help.

Cheers!

kintel, 2018-04-22:
@tsingi `qmake CONFIG+=debug CONFIG+=experimental`

tsingi, 2018-04-23:
@kintel 
That worked, thank you.

RichyT, 2020-01-29:
Perhaps we're looking at this the wrong way. Perhaps it would be better to do something broader that _can_ be used for layering. Perhaps instead, filter by tag so you _could_ filter on **id="Layer_n"** but you could equally filter on **stroke="red"**. This allows for other layering schemes from other software (since it's not part of the standard) but also allows some other powerful options. For example, I might make a design in black but make my annotations in blue.

PR #4042 (2022-10-09): SVG import: Allow selecting parts by id
This adds an `id` parameter to the `import` module that allows selecting parts of an SVG to import, either single elements or whole groups (including layers).

In particular, Inkscape layers can be selected using `import("file.svg", id="layer2")`.

I was unsure whether to reuse the `layer` parameter for this, which already performs a similar function for DXF import (although layers in DXF are a different thing than layers in vector graphics applications), but since this is more general than selecting layers, I decided to introduce a new one. They are ali [... cut]
closed 2022-10-09 by t-paul