openscad/openscad #2388: Openscad does not launch

Montibus, 2018-04-28:
Openscad (OpenSCAD-2018.04.06-x86-64-Installer.exe) crashed during rendering. It was my fault - I realized I did not save the source for couple of hours (yes I know) and in a panic I clicked on the editor Ctrl A, Ctrl C.
The text did not copy and Openscad crashed. Fair enough.
However it would not start again. All I get is the circle rotating for about three seconds.
What I tried:  
Reboot
Uninstall with the utility and reboot
Install again
Uninstall with the utility and reboot
Install again
Uninstall and clean registers with c cleaner (there were about 18 items)
Install again
Uninstall with the ccleaner.
Cleared system cashes with ccleaner.
I tried all this with both OpenSCAD-2018.04.06-x86-64version and OpenSCAD-2015.03-2-x86-64.
I installed it to a different folder and again to the original one.
I excluded the OpenSCAD folder from defender.
I tried to run it from command prompt, both .com and .exe.

[prompt info.txt](https://github.com/openscad/openscad/files/1957787/prompt.info.txt)

Sometimes the .exe running in the prompt generated a window saying "openscad.exe stopped working".

I did not upadate any software, I did not connect any hardware...
It seems openscad will never run on my computer again.
There must be some simple solution, delete a damaged file or registers item.
I am out of ideas, what to try.

rcolyer, 2018-04-28:
For completeness, can you try renaming the C:\Users\marek\AppData\Local\OpenSCAD configuration folder, and trying again?  If this fixes it, the difference between the two configurations might be insightful as feedback here.

t-paul, 2018-04-28:
Note that if you did render sometime before the crash, there should be a copy of the script in the Backup Folder.

Montibus, 2018-04-28:
Latest attempt to run it a minute ago produced the following log::
Faulting application name: openscad.exe, version: 0.0.0.0, time stamp: 0x564d2b59
Faulting module name: atig6txx.dll, version: 22.19.677.257, time stamp: 0x59c5720e
Exception code: 0xc0000005
Fault offset: 0x00000000000196db
Faulting process id: 0x3114
Faulting application start time: 0x01d3df043c71cbf5
Faulting application path: C:\Program Files\OpenSCAD\openscad.exe
Faulting module path: C:\WINDOWS\System32\DriverStore\FileRepository\c0318486.inf_amd64_11ba0b4b7cc81d52\atig6txx.dll
Report Id: e065e696-244a-460d-b02d-94f0c9389e0b
Faulting package full name: 
Faulting package-relative application ID:

Montibus, 2018-04-28:
"Note that if you did render sometime before the crash, there should be a copy of the script in the Backup Folder."
I realized it as soon as I unistalled for the first time. It was not that important - i tried to model a magnetic Rubic's cube 2x2x2 with a steel ball inside just for fun. It will be much easier to write it again. :-)

Montibus, 2018-04-28:
The first crash produced the following log:
Faulting application name: openscad.exe, version: 0.0.0.0, time stamp: 0x5ac7f7d8
Faulting module name: atig6txx.dll, version: 22.19.677.257, time stamp: 0x59c5720e
Exception code: 0xc0000005
Fault offset: 0x00000000000196db
Faulting process id: 0x1c0c
Faulting application start time: 0x01d3de65e897310b
Faulting application path: C:\Program Files\OpenSCAD\openscad.exe
Faulting module path: C:\WINDOWS\System32\DriverStore\FileRepository\c0318486.inf_amd64_11ba0b4b7cc81d52\atig6txx.dll
Report Id: 60b8f78f-2919-4fed-8a25-56d3ee188320
Faulting package full name: 
Faulting package-relative application ID:

t-paul, 2018-04-28:
Both crashes are in the ATI/AMD graphics driver. Is there an update available for that? Not sure how it could be triggered by the actions mentioned above though.

Montibus, 2018-04-28:
I installed a new graphics driver right now, unfortunately with the same result.

MichaelAtOz, 2018-04-28:
This probably won't work, but maybe the installer is stuffing something.
Try downloading the .zip version extract to Program Files [or ...(x86) depending].

t-paul, 2018-04-29:
Options on Windows are stored in registry, wiping that would be another random try.
I believe it would be somewhere in
HKEY_CURRENT_USER\Software\OpenSCAD
HKEY_LOCAL_MACHINE\Software\OpenSCAD

Montibus, 2018-04-29:
Thank you  I am going to try.

Montibus, 2018-04-29:
There were none openscad keys in local machine, there were some related to openscad in current user. Most of them related to harmful settings of openscad. Under window there were four size, position, state and Defaoult. I deleted size and position, renamed state.
Nothing happened and a failed start does not create new keys.

kintel, 2018-04-29:
Just for completion, are you able to run OpenSCAD fully on the cmd-line? e.g.
`openscad.exe mydesign.scad -o out.png`

Montibus, 2018-04-29:
Great!
I have no idea, what happened, but it helped!

Thank you very much!!!
I tried to run openscad before from the command line, but with no success, but not with a concrete file.
I tried now with a file and since it is possible to start it in a normal way from a desktop icon. 
I am glad it works, but does anybody know WHY?!?

kintel, 2018-04-29:
No idea why that works. My only guess is that, in cmd-line mode, we still initialize Qt, and that may cause some registry or other settings to be written.
If we have a bug related to interpreting or expecting certain settings to be present when launching the GUI, that could explain why running the cmd-line version helps.

If you're ever able to reproduce this, that may prove helpful. Then, we just have to reproduce that on some developer machine : /

Montibus, 2018-04-30:
Thank you very much again!

closed 2018-04-30 by Montibus