openscad/openscad #2389: Ignore: I'm fp

MichaelAtOz, 2018-04-29:


MichaelAtOz, 2018-04-29:
@kintel 
It look like they [fixed it](https://github.com/bountysource/core/issues/1096).
[image]

Can you check the plugin at [bountysource/tools](https://www.bountysource.com/tools), or via your icon at top/right if the link doesn't get you there.
[image]

It isn't clear, but I suspect you should see the install button then authorise, if not try unclicking all the options, logout of both bounty source & github, then back in and see what it looks like then.

kintel, 2018-04-29:
@MichaelAtOz It was showing up as "locked". I unlocked it and it should (hopefully) work again now.

MichaelAtOz, 2018-04-29:
It even went back, and eg did this issue, see OP.

I did a bounty cycle for #2284 and it works.
It left the message below, seems incomplete, but maybe intentional in case another claimant sees it??
`Did you help close this issue? Go claim the $10 bounty on Bountysource.`
Small wins...

closed 2018-04-29 by MichaelAtOz