openscad/openscad #2391: no easy way to run it in Ubuntu 18.04 ?

AndKe, 2018-04-29:
OpenSCAD is no longer in Ubuntu repositories :(
- so I tried PPA., 

the newest PPA is for older Ubuntu versions, and fail miserably:
```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 openscad : Depends: libboost-filesystem1.58.0 but it is not installable
            Depends: libboost-program-options1.58.0 but it is not installable
            Depends: libboost-regex1.58.0 but it is not installable
            Depends: libboost-system1.58.0 but it is not installable
            Depends: libboost-thread1.58.0 but it is not installable
            Depends: libcgal11v5 but it is not installable
            Depends: libglew1.13 (>= 1.12.0) but it is not installable
            Depends: libmpfr4 (>= 3.1.3) but it is not installable
            Depends: libqt5scintilla2-12v5 (>= 2.8.4) but it is not installable
E: Unable to correct problems, you have held broken packages.
```

So I downloaded the binary package openscad-2014.03.x86-64.tar.gz
installed.. and failed with: 
```
$ openscad 
/usr/local/bin/../lib/openscad//openscad: relocation error: /usr/local/bin/../lib/openscad/librt.so.1: symbol __vdso_clock_gettime version GLIBC_PRIVATE not defined in file libc.so.6 with link time reference

```

AndKe, 2018-04-29:
appimage works fine.  :)

t-paul, 2018-04-29:
Thanks. Good news.