openscad/openscad #2392: Dot bug

plumbum, 2018-05-02:
Please look this code.

```
l = 61;
w = 44;
h = 78 + 7;

thick = 1;

difference() {
    cube([l+thick*2, w+thick*2, h+thick]);
    translate([thick, thick, thick])
       cube([l, w. h+thick*2]);
}
```
It is have errors, but not generate errors afer parsing.
Detail look string `cube([l, w. h+thick*2]);`. Here i place dot instead comma after `w`. It's not generate error, but not generate vaild model.

t-paul, 2018-05-02:
Yes, it's using Z = 0 as only 2 values are given in the vector. That will never generate a parse error.

I'll set it as "Discussion" if it maybe should give a warning.

nophead, 2018-05-02:
w. h is not a syntax error because this like p.x are allowed. It gives
undef.

MichaelPFrey, 2018-10-27:
So, to simplify the issue to its essential:
`cube([1,1]);` should give a warning, as the third value of the vector is missing (and is thus assumed tobe zero)

Looking at the source code, it gets tricky quickly.

Oddly enough, getVec3 specifically defaults the z-value, when a Vector with two elements is supplied [[1]](https://github.com/openscad/openscad/blob/master/src/value.cc#L375), but cube does not supply a default [[2]](https://github.com/openscad/openscad/blob/master/src/primitives.cc#L195), so 0.0 is used a default [[3]
](https://github.com/openscad/openscad/blob/master/src/value.h#L198)

This whole defaulting the Z-Value is coming from scale: [[4]](https://github.com/openscad/openscad/blob/master/src/transform.cc#L89)

Well: That is tricky, even before I try black box testing:
```
translate([-6,0,0])
cube([1]);

translate([-4,0,0])
cube([1,1]);

translate([-2,0,0])
cube([1,1,1]);

translate([0,0,0])
cube([true,true,true]);

translate([2,0,0])
cube([true,2,"3"]);

translate([4,0,0])
cube(["1","2","3"]);
```
Yes that needs some attention.

@plumbum Technically, this has nothing to do with the parser. At least the way our parser is setup, we can only detect this issue later in the process. Breaking the process always breaks backward compatibility, so a warning is more adequate.

nophead, 2018-10-27:
getVec3 is also used by translate and that should accept two elements without warning, not just scale.

t-paul, 2018-10-27:
I think we should close this issue as there is no "dot bug" as @nophead already explained. The point about the need for more warnings in specific cases still stands and those would be better handled in a separate issue (and there are a number of existing issues already).