openscad/openscad #2394: Select sounds to include with distribution packages

MichaelAtOz, 2018-05-04:
#2375 Needs sounds. 
It seems Qt Qsound supports WAVE files on all platforms.
There are obviously sounds available in various OSs, but I think, just like fonts, we probably need a small set of sounds to distribute with OpenSCAD.

I raise this issue to facilitate the selection, and discussions to achieve this.

Criteria?
Obviously, they need to be Open Source, preferably Public Domain.
Not too big, or too long.
Inoffensive.
Other criteria?

Types?
I think there should be at least one Microwave ping.
Some generic bleep boing typical finish tones.
A couple of comic relief sounds?
Bonus points if anyone finds something suitable that is topic related.

Unless someone finds a better source, the Freesound.org site seems to be a good source, limit sounds to that source?

Do we open this to the forum, or limit it here to developers?

MichaelAtOz, 2018-05-04:
Maybe I'm overthinking this, but perhaps people could nominate a single sound, as a post here with a short description of why it suits, and we can use the up/down vote emojis.
Or would people want me to build a strawman set?

dorkmo, 2018-05-04:
https://freesound.org/people/Atkom/sounds/347686/

this ones kinda nice

MichaelAtOz, 2018-05-04:
That last one is .mp3

MichaelAtOz, 2018-05-04:
[Microwave sound with 3 pings](https://freesound.org/people/bormane/sounds/257435/), <3s, 260KB possibly too big.

MichaelAtOz, 2018-05-04:
[Harp sound](https://freesound.org/people/luckylittleraven/sounds/415447/) quite nice, bit big 500KB 3s CC0

MichaelAtOz, 2018-05-04:
[Quirky squeak](https://freesound.org/people/propthis/sounds/24731/), 1s 40KB CC0

MichaelAtOz, 2018-05-04:
[Drum Roll with Cymbal Clash](https://freesound.org/people/adriann/sounds/191718/), 6s 116KB CC0

MichaelAtOz, 2018-05-04:
[Halleluja](https://freesound.org/people/edsward/sounds/341867/#)(as in finally it's over...), 3s 540KB CC0

MichaelAtOz, 2018-05-05:
[Eerie Alien Horn](https://freesound.org/people/Mr_KeybOred/sounds/350964/) load & bassy, 75K 8s CC0

MichaelAtOz, 2018-05-05:
[Chicken Single Alarm Call](https://freesound.org/people/Rudmer_Rotteveel/sounds/316920/), 200K 1s CC0

MichaelAtOz, 2018-05-05:
[Boing Spring Mouth harp](https://freesound.org/people/pan14/sounds/334033/), 1s 110K CC0

MichaelAtOz, 2018-05-05:
[Bass riff](https://freesound.org/people/NOTOTO84/sounds/275509/), 2s 440K CC0

MichaelAtOz, 2018-05-05:
[Tambourine](https://freesound.org/people/marvman/sounds/58876/), 3s 280K CC0

MichaelAtOz, 2018-05-05:
[Mystery of the cosmos](https://freesound.org/people/Fynixx/sounds/400744/), 4s 577K CC0. Trim ends.

MichaelAtOz, 2018-05-06:
OK we need one for the initial release, criteria short & simple notification beep-like, CC0 (public domain)

I'm suggesting
A. [SweetAlertSound4.wav](https://freesound.org/people/KIZILSUNGUR/sounds/72128/), 1.2s 205K
B. [Signal-Ring 1](https://freesound.org/people/Vendarro/sounds/399315/), 1.4s 240K
C. [Powerup/success.wav](https://freesound.org/people/GabrielAraujo/sounds/242501/), 0.9s 78K
D. [Soft Alert.wav](https://freesound.org/people/Ethraiel/sounds/362650/), 0.7s 58K
E. [Retro "Accomplished" SFX](https://freesound.org/people/suntemple/sounds/253177/), 0.34s 30K

I like A, but balancing size, as ATM the file in in the exe, I recommend C.

@dorkmo @kintel @t-paul @JustVic Thoughts?

MichaelAtOz, 2018-05-06:
Alternatively I was playing with Audacity, This is 'openscad' in DTMF & manipulated a bit.
Possibly too high pitched? 10K 0.7s
[OpenSCAD DTMF2.zip](https://github.com/openscad/openscad/files/1977237/OpenSCAD.DTMF2.zip)
Public domain.

t-paul, 2018-05-06:
That one is nice. I like that. I'm voting for this as initial file. I'll drop some more links from freesound.org later, just browsing still....

MichaelAtOz, 2018-05-06:
Note that the tone sounds on cancelling render. 
I think that is good, as cancelling can sometimes take a while.
Also sounds with warning & errors, a later change may be to have a 'bad' tone for errors??

MichaelAtOz, 2018-05-06:
[SFX » SynthPing](https://freesound.org/people/DigiSample/sounds/210871/), I like this one, 1.5s 530K
I think that's enough for now...

MichaelAtOz, 2018-05-13:
[Sound of Work - Blacksmith ed1. - Pneumatic tools - Drill](https://freesound.org/people/ldezem/sounds/386288/), 1.5s 830K (this is the one I got to ~10K) CC0

MichaelAtOz, 2018-05-14:
[Distant Explosion](https://freesound.org/people/Mr_KeybOred/sounds/372086/), 4.3s 750K CC0
Sort of in the soft range, but the ambience makes it noticeable. Just a nice sound IMO.

OK Ill stop now. 
Also for those who just read emails, I edited a few posts above, including adding two variants of the DTMF openscad tones. I prefer No. 4 (1st) & 3 over the original posted (No. 2).

donbright, 2018-08-12:
"I think, just like fonts, we probably need a small set of sounds to distribute with OpenSCAD."

I think we need to clarify **need** vs **want**. Many open source projects have died or become unusuable because of feature creep and bloat while straying away from the fundamentals. It is nice to have fun things but it should be weighed against the costs.

* Does the project need to build? 

Currently, the windows build of master branch is borken. The sound patch was accepted but the cross build does not actually work because the current mxe scripts did not include qtmultimedia.(among other things )  Im not talking about special hacks we have to build special branches, im talking about if the average user follows the README.md instructions to cross build windows, it doesnt work. A surprising number of people do actually want to build it themselves. 

The release-build on master branch also does not work with the sound because the resulting packages do not actually play sounds because the release-common script doesnt include the sounds subdirectory into the final package. I.e. it copies everything under examples and fonts, but not under sounds. If the sound feature is in the master, then it should be in the builds of the packages (like the windows installer, mac package). 

* Do we need to watch package bloat?

Rough addup of all the sounds above is about 5 megabytes.  Current windows installer is about 15 MB. If you add all these sounds that would be increasing the size by 33% for a fun feature  that vast majority of people will never use. But a vast majority of people i would wager do want faster downloads and smaller packages.  

This kind of bloat also makes it very hard for developers without fast internet who want to test builds. Download speeds are slow in places, good testing means repeated downloads, git clone, etc. 

* Do we need to watch dependencies bloat?

Including qt multimedia means that we now depend on qtsvg, qtxmlpatterns, and qtdeclarative which alltogether is over 480 Megabytes of source code, and it adds approximately 1 hour 40 minutes to the cross-build of dependencies for windows.  

All of the extra build time, download time, etc, makes it harder to test, harder to find bugs, harder to do the normal work of maintenance on a project.

JustVic, 2018-08-12:
@donbright Your words make sense, but sound notification is a small but nice and convenient feature. It will be sad to refuse this function. If you know how to implement this function more simple and lightweight offer your thoughts.

t-paul, 2018-08-12:
Regarding builds, that's all a bit throwing out the whole project with the bathtub. If you have a look at the Windows builds on the download page, those work just fine with sound. And same all the Linux builds as far as I'm aware of (except AppImage builds are currently broken for us on OBS but that's a different issue).

The missing script update adding qtmultimedia is just an oversight on my part, I will fix as soon as possible, now that I'm aware of it. I see no need to implement a huge campaign against a tiny feature just because of that.

As noted in the other issue, the sound folder currently does not need to be copyied, as it's included though `openscad.qrc`. No issue here either.

And bloat? The Windows binary is still pretty much unchanged at ~17MB, that is less than most icon folders of other programs. I do agree with the dependency concern, but letting OpenSCAD die as there are no new features seems not a good plan. It's not changing much lately anyway as not much people seem to have the time to work on it.

I don't think the plan is to include all the sounds linked above. There was some more discussion, not sure if on some other issue or in the forum which pretty much concluded that there should be at most a minimal collection to suite a wider audience. Everything else could be handled via a user folder which is in place for other files already. That way users could add different sounds without the need for a huge configuration GUI.

MichaelAtOz, 2018-08-12:
> Rough addup of all the sounds above is about 5 megabytes. Current windows installer is about 15 MB. If you add all these sounds 

I have recoded some of the above (ie only tried some) to get significant size reduction without noticeable quality loss. I don't think size will be an issue.

t-paul, 2018-08-13:
Yes, we certainly should not just dump big files into the distribution. What's a good limit to go for? Something like max. 5 files, max. 500kB?

MichaelAtOz, 2018-08-13:
Alternatively, include one small default as a qt resource (like now).
At startup look for some standard file name (eg RenderSound.wav) in the OpenSCAD directory (ie where backups go) & let users copy/rename their sound there. 

Minimal GUI work, tho will need a command-line option perhaps, or let people script it??

We could host some in the forum or web site and only include one default.

MichaelAtOz, 2018-08-13:
> tho will need a command-line option perhaps,

No, just the same RenderSound.wav method.

MichaelAtOz, 2019-02-19:
> Alternatively, include one small default as a qt resource (like now).
At startup look for some standard file name (eg RenderSound.wav) in the OpenSCAD directory (ie where backups go) & let users copy/rename their sound there.

With prospects of a release, should there be some simple means to allow users to choose their preferred sound?

t-paul, 2019-02-19:
Would be nice, but I think for the release-in-progress, it's too late for new features.

kintel, 2019-02-20:
Also, I'd be ok with adding minor UI features like this to patch-level releases as long as we don't change core behavior.