openscad/openscad #2395: Render abort in CGAL line 326

poindexter9, 2018-05-13:
It seems that I can crash the renderer in OpenSCAD Nightly using two rectangles and a cube.

Code:
```
points = [
	[15, 1, 5], // short-rect bottom right -and- tall-rect bottom left
	[14, 2, 5], // short-rect bottom left
	[14, 2, 6], // short-rect top left
	[15, 1, 6], // short-rect top right
	[15, 1, 9], // tall-rect top left
	[14, 0, 9], // tall-rect top right
	[14, 0, 5], // tall-rect bottom right
];

faces = [
	[1, 2, 3, 0], // short-rect
	[4, 5, 6, 0], // tall-rect
];

polyhedron(points, faces); // two adjoined rectangles

cube();
```

Render Abort Message:
```
ERROR: CGAL error in CGAL_Nef_polyhedron3():
CGAL ERROR: assertion violation!
Expr: e->incident_sface() != SFace_const_handle()
File: /usr/include/CGAL/Nef_S2/SM_const_decorator.h
Line: 326
```
The two rectangles are joined on one side, but one is a bit taller (Z direction) than the other.

The two sides / lines that "join" the rectangles are defined with these point coordinates:

The shared point at the bottom is:
[15, 1, 5]

The two end points at the top are:
[15, 1, 6]
[15, 1, 9]

Since the X and Y coordinates are the same everywhere (see points above), the two lines must be parallel.

However, the crash can seemingly be avoided by adding one extra point somewhere on the taller line.

I have added the point where the shorter rectangle ends, but other points may work as well.  Modified code:

```
faces = [
	[1, 2, 3, 0], // short-rect
	[3, 4, 5, 6, 0], // tall-rect
];
```

Now we get the correct "not a valid 2-manifold" message and the model renders.

(PS. The original model _is_ 2-manifold; the above code just represents the smallest test case I could manage to produce, which isn't.)

(PPS. You have to "flush caches" before you render again, otherwise the error hides.  Not sure why, maybe cache invalidation does not work correctly when the renderer aborts.  It might be worthwhile to always automatically flush caches after a render abort event?)

(PPPS. I could probably adapt the sample code into a test file if you point me in the right direction of the test suite being used.)

t-paul, 2018-05-13:
Non-manifold (and 2 single faces can't be 2-manifold) input may cause errors in the underlying library or just generate broken geomentry. In any case the result is undefined.

poindexter9, 2018-05-13:
As mentioned, the crash occurs whether the input is 2-manifold or not.

t-paul, 2018-05-13:
Then please try to add an example of that (as minimal as possible). With the test case above, there should be a better error communication - which is a long standing, difficult to solve point on the wish list - but it is going to fail in some form or another.

poindexter9, 2018-05-13:
Hmm, okay.  How about this then.

```
points = [
	[15, 2, 5], // short-triangle bottom right -and- tall-triangle bottom left
	[14, 3, 5], // short-triangle bottom left
	[14, 3, 6], // short-triangle top left
	[14, 1, 9], // tall-triangle top right
	[14, 1, 5], // tall-triangle bottom right
	[16, 3, 5], // short-triangle-2 bottom right
	[15, 4, 5], // short-triangle-2 bottom left
	[15, 4, 6], // short-triangle-2 top left
	[16, 1, 5], // tall-triangle-2 bottom left
	[15, 0, 9], // tall-triangle-2 top right
	[15, 0, 5], // tall-triangle-2 bottom right
];

faces = [
	[1, 2, 0], // short-triangle
	[3, 4, 0], // tall-triangle
	[6, 7, 5], // short-triangle-2 (displaced a bit)
	[9, 10, 8], // tall-triangle-2 (displaced a bit)
	[1, 2, 7, 6], // back cover, short triangle pair
	[2, 0, 5, 7], // top cover, short triangle pair
	[0, 1, 6, 5], // bottom cover, short triangle pair
	[3, 4, 10, 9], // back cover, tall triangle pair
	[4, 0, 8, 10], // top cover, tall triangle pair
	[0, 3, 9, 8], // bottom cover, tall triangle pair
];

polyhedron(points, faces); // two adjoined triangles + two displaced triangles + 6 cover plates

cube();
```

I've replaced rectangles with triangles to make the test code a bit smaller.

No more 2-manifold warnings seen (well, here on my end, at least).

t-paul, 2018-05-13:
The faces are wrong orientation. Everything shows up in pinkish color when using "thrown-together" view. See [FAQ article](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#Why_is_my_model_showing_up_with_F5_but_not_F6?).

Also I think both parts are touching in a single vertex? I believe that's also by definition not 2-manifold.

poindexter9, 2018-05-13:
Here you go.  All surfaces are yellow instead of pink now.

```
points = [
	[15, 2, 5], // short-triangle bottom right -and- tall-triangle bottom left
	[14, 3, 5], // short-triangle bottom left
	[14, 3, 6], // short-triangle top left
	[14, 1, 9], // tall-triangle top right
	[14, 1, 5], // tall-triangle bottom right
	[16, 3, 5], // short-triangle-2 bottom right
	[15, 4, 5], // short-triangle-2 bottom left
	[15, 4, 6], // short-triangle-2 top left
	[16, 1, 5], // tall-triangle-2 bottom left
	[15, 0, 9], // tall-triangle-2 top right
	[15, 0, 5], // tall-triangle-2 bottom right
];

faces = [
	[1, 2, 0], // short-triangle
	[3, 4, 0], // tall-triangle
	[5, 7, 6], // short-triangle-2 (displaced a bit)
	[8, 10, 9], // tall-triangle-2 (displaced a bit)
	[6, 7, 2, 1], // back cover, short triangle pair
	[7, 5, 0, 2], // top cover, short triangle pair
	[5, 6, 1, 0], // bottom cover, short triangle pair
	[9, 10, 4, 3], // back cover, tall triangle pair
	[10, 8, 0, 4], // top cover, tall triangle pair
	[8, 9, 3, 0], // bottom cover, tall triangle pair
];

polyhedron(points, faces); // two adjoined triangles + two displaced triangles + 6 cover plates

cube();
```

Changed in this version: ordering of points making up faces, to make the faces yellow.

Rendering aborts with same error message.

poindexter9, 2018-05-13:
> Yeah, the problems are related to the two sub-objects of your polyhedron touching in a single vertex. Since we pass such polyhedrons as-is to CGAL, it will barf.

Aha, very interesting.

The original model is one big hollow object (it produces the same error as this reduced test case).  I guess I should make the test case larger again until it is once more not disconnected.  Or post the original model.

> Workaround in the meantime: Split this into to polyhedrons.

Hmm, easily done with the test case, but the original model is 20000 points and 6000 faces.  It is a hollow circular ramp, splitting it would make a gaping hole in it, which I imagine is not good either, or?

kintel, 2018-05-13:
I believe hollow polyhedrons should work, but I cannot find a simple test case atm. - you may need to reverse winding order for the negative shapes for this to work - that should be testable by building a hollow cube.

Word of warning though: OpenSCAD isn't too happy about large meshes; things may take some time to render...

poindexter9, 2018-05-13:
> I believe hollow polyhedrons should work

Yep, if one triangle (from code above) is offset so that it doesn't touch the other, the render succeeds.

It chugs happily along with the render if you punch a couple holes in the shapes as well by removing some of the cover plates (although it does emit a 2-manifold warning).

Seems that 2-manifold is not so important for a successful render, but making sure that shapes do not touch each other is.

> Word of warning though: OpenSCAD isn't too happy about large meshes; things may take some time to render...

Fast enough for my use cases :-).  If only it would render...

NateTG, 2018-05-18:
> ... Hmm, easily done with the test case, but the original model is 20000 points and 6000 faces. ...

It is possible (if somewhat slow and painful) to have OpenSCAD detect and mitigate face winding issues in userspace provided it's an oriented surface.  Similarly, simple cases of the 'shared vertex' problem can be detected and mitigated.

poindexter9, 2018-05-21:
Renders now.

Converted all rectangles to triangles ([a,b,c,d] -> [a,b,c] & [c,d,a]), and this had the effect of turning the line 326 assertion into a line 1149 assertion.

The latter was more easily fixable by taking eg. the cube() and when doing the union() making sure that two planes do not overlap, by pushing the cube 0.000001 into the polyhedron.

(It does give a 2-manifold warning now, but only when rendering from cache.  First render, or render after cache flush, does not give the warning.  Bit odd.)

Kudos, and thanks for the help.

poindexter9, 2018-05-21:
> It is possible (if somewhat slow and painful) to have OpenSCAD detect and mitigate face winding issues

Interesting - in practical terms, how would I go about checking a shape for face winding issues?

NateTG, 2018-05-21:
Basically, you just walk the faces, making sure that the edges are all compatible.

I wrote something that checks for duplicate edges & holes that way:

http://forum.openscad.org/Mesh-Polyhedron-Test-Script-td23106.html

You'd also want to split the faces into connected components and do shell counting to get the correct orientation for 'inside' surfaces though that takes a little more work.

poindexter9, 2018-05-28:
Excellent - checking that out now, thanks!

kintel, 2018-12-07:
Closing as this seems to work as expected

closed 2018-12-07 by kintel