openscad/openscad #2396: Segfault with overlapping polyhedrons (PolySet has nonplanar faces)

aaronevers, 2018-05-18:
[pyramid.txt](https://github.com/openscad/openscad/files/2018572/pyramid.txt)

Getting a segfault with the attached code. Tested on OSX 10.12.6  and Windows 7 and OpenSCAD 2015.03 and 2018.03.17.

t-paul, 2018-05-18:
All faces seem to have wrong winding order ([showing up in pink-ish color](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#Why_is_my_model_showing_up_with_F5_but_not_F6?)).

Also most of the faces seem to list points twice, I'm not sure that can generate valid geometry.

Still, it should not crash.

aaronevers, 2018-05-18:
Oops, I thought pink was good. ;)

aaronevers, 2018-05-18:
[pyramid.txt](https://github.com/openscad/openscad/files/2018729/pyramid.txt)

Fixed reversed face winding. Still seeing "PolySet has nonplanar faces" warning. Still crashes if I increase the number of polyhedrons.

donbright, 2019-05-02:
i can reproduce in 6 lines (basically it boils down to the polyhedron being created twice, triggering some cgal ss.circle.has_on assertion)

```points=[[0, 1, 0],[0, 1, 1],[1, 0, 1],[0, 0, 0]];
faces=[[0,1,2,3]];
polyhedron(points, faces);

points=[[0, 1, 0],[0, 1, 1],[1, 0, 1],[0, 0, 0]];
faces=[[0,1,2,3]];
polyhedron(points, faces);

ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation!
Expr: ss_circle.has_on(sp)
File: /usr/include/CGAL/Nef_3/polygon_mesh_to_nef_3.h
Line: 263
ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation!
Expr: ss_circle.has_on(sp)
File: /usr/include/CGAL/Nef_3/polygon_mesh_to_nef_3.h
Line: 263
Current top level object is empty.

```

donbright, 2019-05-02:
note: example two .txt file above doesnt just crash with cgal assertion, it literally segfaults. 

also there is a .scad code workaround:

change 

```
pyramid(20,1)
```

 to this

```
hull() pyramid(20,1)
```

since pyramid is a convex shape the hull wont hurt anything (i think?!?!?!?!)

very cool shape:

[image]

kintel, 2024-03-29:
Status as of 2024-03-24: 
* No longer crashes
* Still broken with CGAL
* Works with Manifold