openscad/openscad #2398: render crash with polyhedron + cylinder union()

poindexter9, 2018-05-21:
It seems that this code:

```
polyhedron([
	[1000, 1000, 1000], [4, 0, 1], [0, 4, 1], [-4, 0, 1], [0, -4, 1],
	[1000, 1000, 1000], [4, 0, 5], [0, 4, 5], [-4, 0, 5], [0, -4, 5],
	[1000, 1000, 1000], [15, 0, 5], [0, 15, 5], [-15, 0, 5], [0, -15, 5],
	[1000, 1000, 1000], [15, 0, 6], [0, 15, 12], [-15, 0, 18], [0, -15, 24],
	[1000, 1000, 1000], [0, 15, 6], [-15, 0, 12], [0, -15, 18], [15, 0, 24],
	[1000, 1000, 1000], [50, 0, 6], [0, 50, 12], [-50, 0, 18], [0, -50, 24],
	[1000, 1000, 1000], [0, 50, 6], [-50, 0, 12], [0, -50, 18], [50, 0, 24],
	[1000, 1000, 1000], [50, 0, 0], [0, 50, 0], [-50, 0, 0], [0, -50, 0],
	[1000, 1000, 1000], [0, 0, 0],
	[1000, 1000, 1000], [0, 0, 1],
], [
	[26, 31, 37],
	[37, 36, 26],
	[27, 32, 38],
	[38, 37, 27],
	[28, 33, 39],
	[39, 38, 28],
	[29, 34, 36],
	[36, 39, 29],
	[36, 37, 41],
	[37, 38, 41],
	[38, 39, 41],
	[39, 36, 41],
	[1, 4, 43],
	[4, 3, 43],
	[3, 2, 43],
	[2, 1, 43],
	[7, 6, 1],
	[1, 2, 7],
	[8, 7, 2],
	[2, 3, 8],
	[9, 8, 3],
	[3, 4, 9],
	[6, 9, 4],
	[4, 1, 6],
	[12, 11, 6],
	[6, 7, 12],
	[13, 12, 7],
	[7, 8, 13],
	[14, 13, 8],
	[8, 9, 14],
	[11, 14, 9],
	[9, 6, 11],
	[21, 16, 11],
	[11, 12, 21],
	[22, 17, 12],
	[12, 13, 22],
	[23, 18, 13],
	[13, 14, 23],
	[24, 19, 14],
	[14, 11, 24],
	[31, 26, 16],
	[16, 21, 31],
	[32, 27, 17],
	[17, 22, 32],
	[33, 28, 18],
	[18, 23, 33],
	[34, 29, 19],
	[19, 24, 34],
	[27, 31, 21],
	[21, 17, 27],
	[28, 32, 22],
	[22, 18, 28],
	[29, 33, 23],
	[23, 19, 29],
	[26, 34, 24],
	[24, 16, 26],
]);

cylinder(d=120, h=0.3);
```

Crashes both OpenSCAD-Nightly and OpenSCAD with this error message:
```
terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::bad_any_cast>>'
what(): boost::bad_any_cast: failed conversion using boost::any_cast
Aborted
```

When you click Render.

kintel, 2018-05-21:
The backtrace indicates a crash inside CGAL's union code:
```
#8	0x000000010055a192 in void boost::throw_exception<boost::bad_any_cast>(boost::bad_any_cast const&) at /Users/kintel/code/OpenSCAD/openscad/../libraries/install/include/boost/throw_exception.hpp:69
#9	0x000000010074038d in unsigned int boost::any_cast<unsigned int>(boost::any&) at /Users/kintel/code/OpenSCAD/openscad/../libraries/install/include/boost/any.hpp:268
#10	0x000000010073ff62 in CGAL::SNC_FM_decorator<CGAL::SNC_structure<CGAL::Cartesian<CGAL::Gmpq>, CGAL::SNC_indexed_items, bool> >::determine_facet(CGAL::internal::In_place_list_iterator<CGAL::SNC_in_place_list_shalfedge<CGAL::SNC_indexed_items::SHalfedge<CGAL::SNC_structure<CGAL::Cartesian<CGAL::Gmpq>, CGAL::SNC_indexed_items, bool> > >, std::__1::allocator<CGAL::SNC_in_place_list_shalfedge<CGAL::SNC_indexed_items::SHalfedge<CGAL::SNC_structure<CGAL::Cartesian<CGAL::Gmpq>, CGAL::SNC_indexed_items, bool> > > > >, std::__1::vector<CGAL::internal::In_place_list_iterator<CGAL::SNC_in_place_list_shalfedge<CGAL::SNC_indexed_items::SHalfedge<CGAL::SNC_structure<CGAL::Cartesian<CGAL::Gmpq>, CGAL::SNC_indexed_items, bool> > >, std::__1::allocator<CGAL::SNC_in_place_list_shalfedge<CGAL::SNC_indexed_items::SHalfedge<CGAL::SNC_structure<CGAL::Cartesian<CGAL::Gmpq>, CGAL::SNC_indexed_items, bool> > > > >, std::__1::allocator<CGAL::internal::In_place_list_iterator<CGAL::SNC_in_place_list_shalfedge<CGAL::SNC_indexed_items::SHalfedge<CGAL::SNC_structure<CGAL::Cartesian<CGAL::Gmpq>, CGAL::SNC_indexed_items, bool> > >, std::__1::allocator<CGAL::SNC_in_place_list_shalfedge<CGAL::SNC_indexed_items::SHalfedge<CGAL::SNC_structure<CGAL::Cartesian<CGAL::Gmpq>, CGAL::SNC_indexed_items, bool> > > > > > > const&, CGAL::Unique_hash_map<CGAL::internal::In_place_list_iterator<CGAL::SNC_in_place_list_shalfedge<CGAL::SNC_indexed_items::SHalfedge<CGAL::SNC_structure<CGAL::Cartesian<CGAL::Gmpq>, CGAL::SNC_indexed_items, bool> > >, std::__1::allocator<CGAL::SNC_in_place_list_shalfedge<CGAL::SNC_indexed_items::SHalfedge<CGAL::SNC_structure<CGAL::Cartesian<CGAL::Gmpq>, CGAL::SNC_indexed_items, bool> > > > >, int, CGAL::Handle_hash_function, std::__1::allocator<int> > const&, std::__1::vector<CGAL::internal::In_place_list_iterator<CGAL::SNC_in_place_list_shalfedge<CGAL::SNC_indexed_items::SHalfedge<CGAL::SNC_structure<CGAL::Cartesian<CGAL::Gmpq>, CGAL::SNC_indexed_items, bool> > >, std::__1::allocator<CGAL::SNC_in_place_list_shalfedge<CGAL::SNC_indexed_items::SHalfedge<CGAL::SNC_structure<CGAL::Cartesian<CGAL::Gmpq>, CGAL::SNC_indexed_items, bool> > > > >, std::__1::allocator<CGAL::internal::In_place_list_iterator<CGAL::SNC_in_place_list_shalfedge<CGAL::SNC_indexed_items::SHalfedge<CGAL::SNC_structure<CGAL::Cartesian<CGAL::Gmpq>, CGAL::SNC_indexed_items, bool> > >, std::__1::allocator<CGAL::SNC_in_place_list_shalfedge<CGAL::SNC_indexed_items::SHalfedge<CGAL::SNC_structure<CGAL::Cartesian<CGAL::Gmpq>, CGAL::SNC_indexed_items, bool> > > > > > > const&) const at /Users/kintel/code/OpenSCAD/openscad/../libraries/install/include/CGAL/Nef_3/SNC_FM_decorator.h:424
#11	0x000000010073acbc in CGAL::SNC_FM_decorator<CGAL::SNC_structure<CGAL::Cartesian<CGAL::Gmpq>, CGAL::SNC_indexed_items, bool> >::create_facet_objects(CGAL::Plane_3<CGAL::Cartesian<CGAL::Gmpq> > const&, std::__1::__list_iterator<CGAL::Object, void*>, std::__1::__list_iterator<CGAL::Object, void*>) const at /Users/kintel/code/OpenSCAD/openscad/../libraries/install/include/CGAL/Nef_3/SNC_FM_decorator.h:654
#12	0x0000000100732b75 in CGAL::SNC_external_structure<CGAL::SNC_indexed_items, CGAL::SNC_structure<CGAL::Cartesian<CGAL::Gmpq>, CGAL::SNC_indexed_items, bool> >::categorize_facet_cycles_and_create_facets() const at /Users/kintel/code/OpenSCAD/openscad/../libraries/install/include/CGAL/Nef_3/SNC_external_structure.h:1280
#13	0x00000001007aefca in void CGAL::SNC_external_structure<CGAL::SNC_indexed_items, CGAL::SNC_structure<CGAL::Cartesian<CGAL::Gmpq>, CGAL::SNC_indexed_items, bool> >::build_after_binary_operation<CGAL::ID_support_handler<CGAL::SNC_indexed_items, CGAL::SNC_decorator<CGAL::SNC_structure<CGAL::Cartesian<CGAL::Gmpq>, CGAL::SNC_indexed_items, bool> > > >(CGAL::ID_support_handler<CGAL::SNC_indexed_items, CGAL::SNC_decorator<CGAL::SNC_structure<CGAL::Cartesian<CGAL::Gmpq>, CGAL::SNC_indexed_items, bool> > >&) at /Users/kintel/code/OpenSCAD/openscad/../libraries/install/include/CGAL/Nef_3/SNC_external_structure.h:1404
#14	0x00000001007a6107 in void CGAL::Binary_operation<CGAL::SNC_structure<CGAL::Cartesian<CGAL::Gmpq>, CGAL::SNC_indexed_items, bool> >::operator()<CGAL::Nef_polyhedron_3<CGAL::Cartesian<CGAL::Gmpq>, CGAL::SNC_indexed_items, bool>::OR>(CGAL::SNC_point_locator<CGAL::SNC_decorator<CGAL::SNC_structure<CGAL::Cartesian<CGAL::Gmpq>, CGAL::SNC_indexed_items, bool> > >*, CGAL::SNC_structure<CGAL::Cartesian<CGAL::Gmpq>, CGAL::SNC_indexed_items, bool> const&, CGAL::SNC_point_locator<CGAL::SNC_decorator<CGAL::SNC_structure<CGAL::Cartesian<CGAL::Gmpq>, CGAL::SNC_indexed_items, bool> > > const*, CGAL::SNC_structure<CGAL::Cartesian<CGAL::Gmpq>, CGAL::SNC_indexed_items, bool> const&, CGAL::SNC_point_locator<CGAL::SNC_decorator<CGAL::SNC_structure<CGAL::Cartesian<CGAL::Gmpq>, CGAL::SNC_indexed_items, bool> > > const*, CGAL::Nef_polyhedron_3<CGAL::Cartesian<CGAL::Gmpq>, CGAL::SNC_indexed_items, bool>::OR const&) at /Users/kintel/code/OpenSCAD/openscad/../libraries/install/include/CGAL/Nef_3/Binary_operation.h:581
[... cut ...]
#18	0x0000000100791179 in CGAL::Nef_nary_union_3<CGAL::Nef_polyhedron_3<CGAL::Cartesian<CGAL::Gmpq>, CGAL::SNC_indexed_items, bool> >::add_polyhedron(CGAL::Nef_polyhedron_3<CGAL::Cartesian<CGAL::Gmpq>, CGAL::SNC_indexed_items, bool> const&) at /Users/kintel/code/OpenSCAD/openscad/../libraries/install/include/CGAL/Nef_nary_union_3.h:57
#19	0x0000000100790905 in CGALUtils::applyOperator(std::__1::list<std::__1::pair<AbstractNode const*, std::__1::shared_ptr<Geometry const> >, std::__1::allocator<std::__1::pair<AbstractNode const*, std::__1::shared_ptr<Geometry const> > > > const&, OpenSCADOperator) at /Users/kintel/code/OpenSCAD/openscad/src/cgalutils-applyops.cc:101
```

poindexter9, 2018-05-21:
@t-paul: do you see something wrong with this model?

I am thinking that this rule may have been broken:

> exactly two faces should meet at any polyhedron edge.

Specifically in each of the four corners, where the taller block has one triangle towards the vertical "corner edge", and the less tall block has one as well, but there is also a third triangle pointing inwards towards the model center.

I'm not sure about this rule:

> if two faces have a vertex in common, they should be in the same cycle face-edge around the vertex.

I have not (yet at least) understood the second part of that sentence.

poindexter9, 2018-05-28:
So anyway,

I still do not understand the second rule:

> if two faces have a vertex in common, they should be in the same cycle face-edge around the vertex.

So I simply ignored it.

I then made a model where I followed this rule:

> exactly two faces should meet at any polyhedron edge.

The logical consequence of this rule is that I have to "triangulate" everything manually for any model slightly more complex than a cube, far as I can tell.  I did, and it works great.  No more crashes, no more assertion failures either.

I "triangulated" in a way that I suppose is logical to a human being, and that is simple to implement in code.  That ended up increasing the amount of triangles by several magnitudes though.  So both preview and render is really, really slow.

It was also a hell of a lot of work.  Using this method, about 10% of the work in using polyhedron() is defining points and shapes, an 90% is slicing things up into triangles where each side touch one other triangle.

I also made another model using this home-made rule for polyhedron():

"For each line of each face you define, if there is a point along that line which is used in another face, make sure to include that point in the definition of the current face as well."

It turns out that OpenSCAD is much better at turning faces into triangles than I am, as one can see after turning on "View --> Show Edges".  Following the above rule worked exactly as well as doing everything manually, no assertions or failures either.  It also previews and renders a lot faster than the manual method.

Long story short, I think the wiki should probably state somewhere, for other newbies like me, the following rules:

"1) For each edge of each face you define, if there is a point along that line which is used in another face, make sure to include that point in the definition of the current face as well.  This ensures that OpenSCAD can properly turn the face into a triangle where each side touch exactly one other triangle.

2) Make sure not to define two faces that meet at only a single point.  This ensures that CGAL can render the shape without raising any assertions."

Instead of the current rules that a) force a lot of manual work and b) is incomprehensible.

Another idea would be to have OpenSCAD do the above automatically?

In theory, it should be possible to take any faces that share a single point and check if any of their edges are exactly parallel.  If so, and iff one of the edges is shorter than the other, automatically include the (previously non-shared) end point of the shorter edge in the face with the longer edge.

If afterwards two faces have at least one point in common, check that they also have at least two points in common.  If not, show a warning during preview (because CGAL will certainly throw an assertion during render).

(After those two checks, turn faces into triangles as usual.)

Not sure if it is realistic to implement, but if it was, it could potentially save a bunch of time for the next newbie...

kintel, 2018-05-28:
Thanks for being patient!
Some notes/responses:
* We have a meta-issue for automatic mesh repair. It contains some ideas and work in progress: https://github.com/openscad/openscad/issues/1580
* Another way of describing what you mention in the end is that any mesh should be "topologically watertight": If you were to move _any_ one point in space, it will never cause a crack to appear in the mesh.
* If you feel you have a good way of explaining this to the next victim, it would be very much appreciated if you'd take the opportunity to edit the wiki; most of the wiki is user-contributed. Projects like this exist because of distributed volunteer effort!

nophead, 2018-05-28:
>if two faces have a vertex in common, they should be in the same cycle face-edge around the vertex.

I think that just means faces that share a vertex should have the same winding order. I.e. all clockwise when viewed from the outside.

kintel, 2024-03-29:
Status as of 2024-03-24:
* No longer crashes
* The model still doesn't render since the design isn't a manifold topology

jordanbrown0, 2026-09-27:
Is there anything left to do here?  It seems that the original problem (the crash) has been fixed, and the secondary problem (that a render fails) is that the polyhedron is, according to our picky rules, malformed... end of story.