openscad/openscad #2400: Hang when compiling (or initial preview) with large number of objects

MichaelAtOz, 2018-05-23:
@t-paul 
can you build a Windows snapshot at the same point as the multi-thread is based on, but without the multi-thread changes.

I've got a hang but not in render, so I need to see if it is master or some side-effect of the multi-thread stuff.
It doesn't happen with current snapshot .04.06 or the .05.05 sound version.
Perhaps #2343?

MichaelAtOz, 2018-05-23:
Below is the code, designed for multi-thread testing, which hangs, after 'Compiling design (CSG Products generation)...'
Start with n=500, go up by 400 at a time.
It is slower with increasing n, but it is pretty obvious when it hangs. CPU use stops.
The hang happens at different n on the 2 PCs I used. 
In earlier scad versions, n=10,000+ is handled easily.
```
// random-threads
x=1920; // smaller values encourages overlap v's big n
y=1080; // 10% HiDef 16:9 ;)

// key changes to use different geometry each time
key=floor(rands(1000,999999,1)[0]);

// add a previous key here to reproduce same output
//key=1; 

s=rands(-20,20,7,key);     // seed

// Number of objects (#thread ~= 2-4 times n)
n=500;    // fixed
//n=ceil(rands(50000,99000,1,s[0])[0]);  // random

log=1;

// random position & size
rx=rands(-x,x,n,s[1]);
ry=rands(-y,y,n,s[2]);
rz=rands(-2,2,n,s[3]);
rs=rands(5,35,n,s[4]);      // size
rt=rands(0,3.999,n,s[5]);    // type
rr=rands(0,359,n,s[6]);      // rotation
echo(""); // visusal note

// limits thread concurrency if < #cores 
parts=12;

// 2D v's 3D
D=3; 

// affects speed of render
fn=7;
$fn=(D==3) ? fn : fn*20; // make 2D work harder 
echo(D=D,key=key,n=n,$fn=$fn,parts=parts);

module D2(i) {
  s=rs[i];
  t=3; //floor(rt[i]);
  rotate(rr[i])
  if (t==0)
    circle(s);
  else if (t==1)
    square(s);
  else if (t==2)
    scale(0.05*s)
      polygon(points=[[0,0],[100,0],[0,100],[10,10],[80,10],[10,80]]
            , paths=[[0,1,2],[3,4,5]],convexity=10); // from wiki
  else if (t>=3)
    hull() {
      circle(s);
      translate([rz[i]*5,sign(rz[i])*s*3 ]) square(s);
      translate([-sign(rz[i])*s*5,rz[i]*3 ]) square(s);
   }
}

module D3(i) { 
  s=rs[i];
  t=floor(rt[i]);
  rotate([rr[i],-rr[i],rr[i]/s])
  if (t==0)
    sphere(s);
  else if (t==1)
    cube(s);
  else if (t==2)
    cylinder(r=s,h=s);
  else if (t>=3)
    scale(1/24*s)
      polyhedron
      (points = [
       [0, -10, 60], [0, 10, 60], [0, 10, 0], [0, -10, 0], [60, -10, 60], [60, 10, 60], 
       [10, -10, 50], [10, 10, 50], [10, 10, 30], [10, -10, 30], [30, -10, 50], [30, 10, 50]
       ], 
      faces = [
       [0,3,2],  [0,2,1],  [4,0,5],  [5,0,1],  [5,2,4],  [4,2,3],
       [6,8,9],  [6,7,8],  [6,10,11],[6,11,7], [10,8,11],
       [10,9,8], [3,0,9],  [9,0,6],  [10,6, 0],[0,4,10],
       [3,9,10], [3,10,4], [1,7,11], [1,11,5], [1,8,7],  
       [2,8,1],  [8,2,11], [5,11,2]
       ]
      );  // from wiki
}

module object(i) 
  if (D==2)
    D2(i);
  else
    D3(i);

for (p=[0:parts-1]) {
  if (log) echo(p=p);
  for (i=[p:parts:n]) { 

    translate([rx[i],ry[i],(D==2) ? 0 : rz[i] ])
      object(i);
  }
}

translate([x,y+90])
  if (D==2)
    text(str("KEY# ",key),size=10,halign="right",valign="bottom");
  else
    #linear_extrude(height=3, convexity=32)
      text(str("KEY# ",key),size=80,halign="right",valign="bottom");
```

MichaelAtOz, 2018-05-23:
[image]
Prob. [this](https://github.com/openscad/openscad/commit/e94eaff)?

t-paul, 2018-05-23:
Sounds good, if you have an update to build, let me know. Ideally with the latest master merged, so I could build both master and the multithread branch for @MichaelAtOz to cross-check.

justbuchanan, 2018-05-24:
I think I have this fixed as of https://github.com/openscad/openscad/pull/2399/commits/212caccb00438b1e3d32d2b83eeeca4c04e9662d. My usage of condition variables wasn't quite right, which could sometimes cause a thread to wait forever for new work to do, preventing compilation from ever finishing.

Let me know if you're able to cause this to happen again.

Also, I merged the latest master branch.

closed 2018-05-24 by justbuchanan

t-paul, 2018-05-24:
http://files.openscad.org/snapshots/OpenSCAD-2018.05.24-x86-32-Installer.exe
http://files.openscad.org/snapshots/OpenSCAD-2018.05.24-x86-32.zip
http://files.openscad.org/snapshots/OpenSCAD-2018.05.24-x86-64-Installer.exe
http://files.openscad.org/snapshots/OpenSCAD-2018.05.24-x86-64.zip

http://files.openscad.org/snapshots/OpenSCAD-2018.05.24-x86-32_multithread-Installer.exe
http://files.openscad.org/snapshots/OpenSCAD-2018.05.24-x86-32_multithread.zip
http://files.openscad.org/snapshots/OpenSCAD-2018.05.24-x86-64_multithread-Installer.exe
http://files.openscad.org/snapshots/OpenSCAD-2018.05.24-x86-64_multithread.zip

MichaelAtOz, 2018-05-25:
No longer hanging, either version, so it was the MT issue.

nophead, 2018-05-25:
I tried snapshots/OpenSCAD-2018.05.24-x86-64_multithread.zip
<http://files.openscad.org/snapshots/OpenSCAD-2018.05.24-x86-64_multithread.zip>
on Win7 64 but I don't get more than 25% CPU usage on a 4 core machine.
This is with thread-traversal enabled in preferences.

However the behaviour of difference has changed. If the result is empty
because the second object is bigger than the first I get a warning WARNING:
No 2D or 3D child objects, 1 empties.

If the first object is empty I get the second object returned as the
difference. Previously the result would be empty.

justbuchanan, 2018-05-25:
Hmm, that's odd. Is the low cpu usage for previewing the model or for a full render? Previews are still mostly single-threaded (except for "render" nodes).

Is difference() doing this for all models you've tried or just some? I tried something simple like:

~~~{.scad}
difference() {
    cube(5);
    translate([-1,-1,-1]) cube(10);
}
~~~

For me, this correctly shows an empty model and prints out the "Warning: No 2d or 3D child  objects". Do you have a model that reliably returns the second object instead of doing a difference?

justbuchanan, 2018-05-25:
Sorry, misread your last comment. If I give an empty first object to difference(), it does return the second. I'll investigate.

t-paul, 2018-05-25:
Added to the pull request TODO so it's not going to be lost in this closed issue.

nophead, 2018-05-26:
I do get all my cores used with Michael's test above with F6.

When testing on my own projects I never get more than one CPU used. They are complex assembles that I view with F5. All the objects have a render at the top level, so the preview is just displaying precomputed polyhedra.  I don't use F6 much at all.

To make STLs I use the command line and that doesn't seem to use more than one core either. Presumably I would need to set a command line option to enable it.

The new warning about no 2d or 3d child objects is an irritation. I have lots of them in my projects.

I also get a new warning that says localisation initialisation failed when run from the command line.

nophead, 2018-05-26:
Actually it does make very brief excursions to using more than one core during F5 but most of the time is spent using one core.

nophead, 2018-05-26:
And it is slightly slower.

nophead, 2018-05-27:
This is the simplest example I could distil from my code that goes wrong:
```
render() difference() {
        translate([0, 0, 0])
            cylinder(d = 0, h = 1);
    
        cylinder(d = 3, h = 100, center = true);
    }
```

It needs the render() and oddly it needs the translate. If the first cylinder has a diameter less than 3 but bigger than zero the result is empty. When the diameter is zero the result is the second cylinder.

It is code that displays PCB pads around mounting holes, or not if the pad diameter is zero.

justbuchanan, 2018-05-31:
I have this fixed as of 3d2df8f436efbd59b8ab487b885b69da059ee41e. I also added this to the test examples.