openscad/openscad #2401: Preferences CGAL & Polyset cache size limit

MichaelAtOz, 2018-05-24:
[Forum](http://forum.openscad.org/CPU-and-RAM-Resources-Allocations-Increase-for-OpenSCAD-tp24031p24080.html) found that the max input is limited to signed int max, 2^31 bytes.

Having looked at the code, this may be a hard limit for 32 bit support. But I'm no expert on the C internals.

cache.h has `void setMaxCost(int m) { mx = m; trim(mx);` making that size_t or unsigned int may get another 2GB?

There is also an anomaly src/Preferences.cc used toULong, src/mainwin.cc uses toUInt

64bit should presumably handle more?

Also while there I noticed [maxCost=100](https://github.com/openscad/openscad/blob/b9de27fbf1e3c73243c5453ad18dbe106e24b2b7/src/cache.h#L97), I'm wondering with larger cache sizes, whether that needs to be examined?

doug-moen, 2018-05-24:
`size_t` is unsigned, 32 bits on a 32 bit system and 64 bits on a 64 bit system. Just change the type to `size_t` everywhere.

MichaelAtOz, 2018-05-24:
[QIntValidator](http://doc.qt.io/qt-5/qintvalidator.html#details), [here](https://github.com/openscad/openscad/blob/7e22a7da586617a29c1e7a3082e43e5d6656a1b2/src/Preferences.cc#L197), appears to be the limiter.

kintel, 2018-05-27:
I agree with @doug-moen.
When fixing this, perhaps it's quick to look into presenting that number in a more human readable format in Preferences (e.g. using a `MB` postfix) ?

peter64, 2019-01-05:
Hey guys I posted a $15 bounty for this. I'de be entirely content with just having the QIntValidator removed (if that is all it takes to get the cache limit unleashed). If it requires more work to get >4GB on x64 systems I am happy to forgo that if it would be overly troublesome. 
 
The last nice to have would be if this can get integrated back into the source so that it is included in the multithreaded build or windows as that is what I use. https://github.com/openscad/openscad/pull/2399
Specifically so it ends up in the next build associated with 2399. The one I currently use is https://339-1049088-gh.circle-artifacts.com/0/64-bit/OpenSCAD-2018.11.04.ci339-x86-64_PR2399.zip

Also if $15 seems like too low a bounty let me know and I can up it slightly.
The biggest hurdle to me trying to fix this myself is building all the dependencies from source under mingw64 (or whatever the required compiler is) (specifically QT), so if there is some remote auto build system already setup for windows x64 for people who are forking and want to play with making changes let me know and I may be able to make a marginal fix myself.

t-paul, 2019-01-05:
All pull requests are now built on CircleCI which produces both 32bit and 64bit Windows binaries. So basically pushing a change to the pull request should result in a new binary about 15 minutes later.
Otherwise the wiki docs on "[Compiling on Windows](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Building_on_Microsoft_Windows)" should be still working. This is just installing all the dependencies via MSYS2 environment, optionally including the QtCreator IDE.

MichaelAtOz, 2019-01-05:
@peter64 just so you know, there is not a lot of uptake on the Bountysource enticements in recent times.

Bounty's not showing on the issue, it looks like the Bountysource/GitHub integration is broken again...

MichaelAtOz, 2019-01-05:
@kintel can you have a look at https://www.bountysource.com/tools, when I look it (any of the openscad/ gits) does not show 'installed' as it does for my git which I setup to test it last time.

peter64, 2019-02-18:
Hey @thehans, Thanks so much, now that #2778 (your fix) is merged down to master, it shouldn't take too long before master is merged to the branch I use As soon as this get's merged down to #2399 (multi-thread support) I will give it a try :) I'm not sure if the bounty can be claimed yet or not but assuming you've fixed it I don't see any reason you shouldn't be able to. Hopefully master will get merged down to #2399 again soon so I can get this change :)

t-paul, 2019-02-18:
The bounty can be claimed once the ticket is closed. With #2778 merged, it should be good to go, so I'm closing here and @thehans can claim the bounty on BountySource.

closed 2019-02-18 by t-paul