openscad/openscad #2409: Feature Request - Plugins like Atom

alex-d-fox, 2018-05-28:
Feature Request to allow plugins similar to http://atom.io
for example https://wakatime.com/ for time tracking, and adding custom file formats... etc...

t-paul, 2018-05-28:
I'm voting for closing as much too broad request. Implementing some specfic interface like the one mentioned should be possible and even going into the direction of plugins would be interesting (support for [Language Server](http://Langserver.org))?
But change C++/QScintilla based editor to be like a totally different JavaScript/Electron editor is a bit strange request.

Note that OpenSCAD specfically supports [external editors](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Using_an_external_Editor_with_OpenSCAD), so actually using Atom is easily possible.

thehans, 2018-05-28:
Agreed, OpenSCAD is not a general purpose IDE, its a CAD program first and foremost.  Adding a plugin system is not the kind of thing you do on a whim, it would likely require a complete re-write/overhaul of the application.

rumatoest, 2018-06-30:
IMHO OpenSCAD does not need any plugins.

rumatoest, 2018-08-04:
It is very nice to see another JS psycho. I think that @donbright should notice, that there is actually almost no UI in OpenSCAD, and this is the point. Adding just 50Mb or more libraries + 100Mb memory consumption + VM just to draw menu and slipt main window in 3 sections is totaly insane. 

> use some kind of scripting language to create their own addons.
Users MUST use only OpenSCAD scripting to create they own libraries and this is the way how it should be. Extension == Libraray.
The only way to improve here is to add more features to the OpenScad language itself to make libraries more flexible.

nstielau, 2018-08-10:
I agree the extension should be more focused on libraries than general purpose plugins.

I could see a command like `openscad library get foobar` that could download `https://github.com/openscad/libraries/blob/master/foobar.scad` into the library directory.  That would be an improvement on the library UX (at least for me), which now involves downloading them from thingiverse or random websites, renaming extensions sometimes, copying them into the library directory.  This would require maintaining the openscad/libraries repo with user submissions.

This would also enable OpenSCAD could also parse library names from code and attempt to automatically download libraries from that repository, making it share libraries and build on other people's code.

jurra, 2018-10-12:
Maybe is a bit out of scope, but anyways: I use the atom editor to program in OpenScad, it its way better than the OpenScad editor, and is open source as well, so that is good for openscad.
Maybe also a bit out of topic, but also related to what @donbright mentioned. Electron is a multiplatform framework, that allows to write applications for every OS like web applications, no need to make versions for Linux, MAC, etc..  I see that eventually moving to these kind of framework can make things in the long run easier for OpenScad developers. 
On the other hand, OpenScad currently performs really fast, this is why is so powerfull. I still have problems with electron based apps, probably because they require a lot of PC resources.

kintel, 2018-12-07:
Closing as this is way off the radar for an issue tracker

closed 2018-12-07 by kintel