openscad/openscad #2411: Error  inr mirror function

gsohler, 2018-05-28:
Appearently there is an error in the mirror function:

Using 2015.03-2

and this test code:

for(z=[0:1])
{
    for(y=[0:1])
    {
        for(x=[0:1])
        {
            mirror([x,y,z]) color([x,y,z]) cube([1,1,1]);
        }
    }    
}
I'd really love to put this picture ... but you will see that the cube at 1:1:1 is out of order

thehans, 2018-05-29:
The parameter for mirror is a vector about which the object is mirrored.  It is not a list of which axes to mirror.
https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Transformations#mirror

If you mirror about [1,1,1], that is a vector which is not aligned with any axis.
If you want to iterate over different combinations of mirrors, you need to do the mirror operations separately for each axis
```
for(z=[0:1])
{
for(y=[0:1])
{
for(x=[0:1])
{
mirror([x,0,0]) mirror([0,y,0]) mirror([0,0,z]) color([x,y,z]) cube([1,1,1]);
}
}
}
```

closed 2018-05-29 by thehans

thehans, 2018-05-29:
An alternate option is that you can scale each axis by -1(mirrored) or 1(not mirrored) to achieve the same effect.

Such as this:
```
for(z=[-1,1])
{
for(y=[-1,1])
{
for(x=[-1,1])
{
scale([x,y,z]) color([(1-x)/2,(1-y)/2,(1-z)/2]) cube([1,1,1]);
}
}
}
```