openscad/openscad #2412: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation!

thomaskilian, 2018-05-30:
When rendering the below code (using 2017.12.23 (git 39823be1)) I get the error

> ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: ss_circle.has_on(sp) File: /Users/kintel/code/OpenSCAD/openscad/../libraries/install/include/CGAL/Nef_3/polygon_mesh_to_nef_3.h Line: 263 

The preview appears fine, though.

```
// ISO Metric Thread Implementation
// Trevor Moseley
// 15/07/2012

// For thread dimensions see
//   http://en.wikipedia.org/wiki/File:ISO_and_UTS_Thread_Dimensions.svg

//$fn = 32;
//ThreadOut(mOut(8), 10);
//HexBolt(10, 16);
//HexNut(10);
//HexBolt(8, 16);
//HexNut(8);
//HexBolt(6, 12);
//translate([0,0,20])
//HexNut(6);

// thread pitch approximations for distinct diameters
function coarsePitch(diameter) = lookup(diameter, [[1, 0.25], [1.2, 0.25], [1.4, 0.3], [1.6, 0.35], [1.8, 0.35], [2, 0.4], [2.5, 0.45], [3, 0.5], [3.5, 0.6], [4, 0.7], [5, 0.8], [6, 1], [7, 1], [8, 1.25], [10, 1.5], [12, 1.75], [14, 2], [16, 2], [18, 2.5], [20, 2.5], [22, 2.5], [24, 3], [27, 3], [30, 3.5], [33, 3.5], [36, 4], [39, 4], [42, 4.5], [45, 4.5], [48, 5], [52, 5], [56, 5.5], [60, 5.5], [64, 6], [78, 5]]);

// hex nut diameter from thread size
function hexNutdiameter(diameter) = lookup(diameter, [[3, 6.4], [4, 8.1], [5, 9.2], [6, 11.5], [8, 16.0], [10, 19.6], [12, 22.1], [16, 27.7], [20, 34.6], [24, 41.6], [30, 53.1], [36, 63.5]]);

// hex nut height from thread size
function hexBoltHeight(diameter) = lookup(diameter, [[3, 2.4], [4, 3.2], [5, 4], [6, 3], [8, 5], [10, 5], [12, 10], [16, 13], [20, 16], [24, 19], [30, 24], [36, 29]]);

// hex bolt head diameter from thread size
function hexBoltdiameter(diameter) = lookup(diameter, [[3, 6.4], [4, 8.1], [5, 9.2], [6, 11.5], [8, 14.0], [10, 16], [12, 22.1], [16, 27.7], [20, 34.6], [24, 41.6], [30, 53.1], [36, 63.5]]);

// hex bolt head height from thread size
function hexBoltHeadHight(diameter) = lookup(diameter, [[3, 2.4], [4, 3.2], [5, 4], [6, 3.5], [8, 4.5], [10, 5], [12, 10], [16, 13], [20, 16], [24, 19], [30, 24], [36, 29]]);

HOLEOVER = 0.5; // overextrusion when bore is printed
// find the bore diameter for e M-thread
function mBore(m) = lookup (m, [[2, 2.4], [2.5, 2.9], [3, 3.4], [4, 4.5], [5, 5.5], [6, 6.6], [8, 9], [10, 11]]) - HULLOVER;

HULLOVER = 0.4; // overextrusion when hull is printed
// find diameter of center when printing a thread
function mOut(m) = lookup (m, [[2, 1.51], [2.5, 1.95], [3, 2.39], [4, 3.14], [5, 4.02], [6, 4.77], [8, 6.47], [10, 8.16]]) - HULLOVER;

// thread at outside of a bolt
module ThreadOut(diameter, height, threadSegments=$fn) {
  segmentsInCircumferrence = (threadSegments == 0) ? 32 : threadSegments;
	pitch = coarsePitch(diameter);
	h = (cos(30)*pitch)/8;
	rMin = (diameter/2) - (5*h);	// angleStart wiki Dmin
	segmentAngle = 360/segmentsInCircumferrence;
	turns = (height-pitch)/pitch;
	segments = turns*segmentsInCircumferrence;
  cylinder(r=rMin, h=height);
	for (segmentNumber=[0:segments])
		ThreadOutSegmentAtPitch(rMin, pitch, segmentAngle, segmentNumber, segmentsInCircumferrence, h, (height-pitch)/segments);
}

// create a segment of a thread outside with a certain pich
module ThreadOutSegmentAtPitch(threadRadius, pitch, segmentDegrees, segmentNumber, segmentsInCircumferrence, h /* ISO h of thread / 8 */, segmentHeight) {
	angleStart = (segmentNumber % segmentsInCircumferrence) * segmentDegrees;			// angle to start of seg
	angleEnd = angleStart + segmentDegrees;		// angle to end of seg
	z = segmentHeight*segmentNumber;

	//   1, 4
	//   |\
	//   | \  2, 5
 	//   | / 
	//   |/
	//   0, 3
	//  view from front (x & z) extruded in y by segmentNumber

	polyhedron(
		points = [
			[cos(angleStart)*threadRadius, sin(angleStart)*threadRadius, z], 								// 0
			[cos(angleStart)*threadRadius, sin(angleStart)*threadRadius, z+(3/4*pitch)], 						// 1
			[cos(angleStart)*(threadRadius+(5*h)), sin(angleStart)*(threadRadius+(5*h)), z+(3/8*pitch)], 		// 2
			[cos(angleEnd)*threadRadius, sin(angleEnd)*threadRadius, z+segmentHeight], 							// 3
			[cos(angleEnd)*threadRadius, sin(angleEnd)*threadRadius, z+(3/4*pitch)+segmentHeight], 					// 4
			[cos(angleEnd)*(threadRadius+(5*h)), sin(angleEnd)*(threadRadius+(5*h)), z+segmentHeight+(3/8*pitch)]], 	// 5
		faces = [
			[0, 1, 2], 			// near face
			[3, 5, 4], 			// far face
			[0, 3, 4, 1], 	// left face
			[0, 3, 5, 2], 	// bottom face
			[1, 4, 5, 2]]);	// top face
}

// thread inside of a bore
module ThreadIn(diameter, height, threadSegments=$fn) {
  segmentsInCircumferrence = (threadSegments == 0) ? 32 : threadSegments;
	pitch = coarsePitch(diameter);
	h = (cos(30)*pitch)/8;
	rMin = (diameter/2) - (5*h);	// as wiki Dmin
	segmentAngle = 360/segmentsInCircumferrence;
	turns = (height-pitch)/pitch;			// number of turns
	segments = turns*segmentsInCircumferrence;				// number of segments
	difference() {
		cylinder(r=(diameter/2)+0.5, h=height);
		translate([0, 0, -1]) cylinder(r=(diameter/2)+0.1, h=height+2);
	}
	for(segmentNumber=[0:segments])
		ThreadInSegmentAtPitch(rMin, pitch, segmentAngle, segmentNumber, segmentsInCircumferrence, h, (height-pitch)/segments);
}

module ThreadInSegmentAtPitch(threadRadius, pitch, segmentDegrees, segmentNumber, threadSegments, h, segmentHeight) {
	angleStart = ((segmentNumber % threadSegments) * segmentDegrees - 180);	// angle to start of seg
	angleEnd = angleStart + segmentDegrees;		// angle to end of seg
	z = segmentHeight*segmentNumber;

	//         2, 5
	//          /|
	//  1, 4   / | 
 	//         \ |
	//          \|
	//         0, 3
	//  view from front (x & z) extruded in y by sg

	polyhedron(
		points = [
			[cos(angleStart)*(threadRadius+(5*h)), sin(angleStart)*(threadRadius+(5*h)), z], 				//0
			[cos(angleStart)*threadRadius, sin(angleStart)*threadRadius, z+(3/8*pitch)], 						//1
			[cos(angleStart)*(threadRadius+(5*h)), sin(angleStart)*(threadRadius+(5*h)), z+(3/4*pitch)], 		//2
			[cos(angleEnd)*(threadRadius+(5*h)), sin(angleEnd)*(threadRadius+(5*h)), z+segmentHeight], 			//3
			[cos(angleEnd)*threadRadius, sin(angleEnd)*threadRadius, z+(3/8*pitch)+segmentHeight], 					//4
			[cos(angleEnd)*(threadRadius+(5*h)), sin(angleEnd)*(threadRadius+(5*h)), z+(3/4*pitch)+segmentHeight]], 	//5
		faces = [
			[0, 1, 2], 			// near face
			[3, 5, 4], 			// far face
			[0, 3, 4, 1], 	// left face
			[0, 5, 3, 2], 	// bottom face
			[1, 4, 5, 2]]);	// top face
}

module HexNut(diameter) {
	h = hexBoltHeight(diameter);
	difference() {
		cylinder(r=hexNutdiameter(diameter)/2, h=h, $fn=6);
		translate([0, 0, -0.1])	cylinder(r=diameter/2, h=h + 0.2);
	}
	translate([0, 0, 0.1])	ThreadIn(diameter, h-0.2);
}

module HexBolt(diameter, height) {
	h = hexBoltHeadHight(diameter);
	cylinder(r=hexBoltdiameter(diameter)/2, h=h, $fn=6);
	translate([0, 0, h-0.1])	ThreadOut(diameter, height+0.1);
}

ThreadOut(46,20);

```

t-paul, 2018-05-30:
See [FAQ](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#Why_is_my_model_showing_up_with_F5_but_not_F6?).

Some faces have wrong winding order.

thomaskilian, 2018-06-03:
Yes, the bottom face was wrong. Changing
`			[0, 5, 3, 2], 	// bottom face
`
to
`			[0, 2, 5, 3], 	// bottom face
`
fixed it, but the error still shows up.

13rac1, 2018-11-19:
Just received this same error message running the last full release `2015.03-2` on Ubuntu 18.04. Confirmed the object provided by @thomaskilian above produces the same error message.

```
ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: ss_circle.has_on(sp) File: /usr/include/CGAL/Nef_3/polygon_mesh_to_nef_3.h Line: 263 
```

Solution is to upgrade. The object renders correctly in `2018.11.18`. OpenSCAD complains multiple times during rendering:
```
PolySet has nonplanar faces. Attempting alternate construction
```
and ends with:
```
WARNING: Object may not be a valid 2-manifold and may need repair!
```
but the STL loads and renders in Blender:
[image]

I believe this issue was fixed in #2139

lukehutch, 2019-05-30:
I received the same error message when I had an invalid (out of range) vertex index in the faces field of a polyhedron. OpenSCAD did not complain, but GCAL failed with this assertion error. OpenSCAD should really check the vertex indices before passing them to CGAL.

markmaker, 2020-02-24:
I had the same issue and traced it back to using 2D translate/rotate on 3D bodies. OpenSCAD did not complain and just used them as expected (in the X-Y plane), but GCAL failed. 

The assertion is not consistent, it may or may not happen, looks like an unchecked vector access. Sometimes the STL generated will be faulty.

burunduk3, 2023-12-27:
I had the same issue (unsurprisingly, also when trying to draw helixes for screw threads), and I think I understood what is going on. And while it would be nice for **openscad** to express the error properly (not like “some cryptic¹ assertion failed in some under-the-hood geometry library”) it's indeed a problem with the code. Let me explain. 

Actually, the faces are wrong. For example, this one:
```
[0, 2, 5, 3], 	// bottom face
```

The culprit here is not the vertices order. It is that **these four points don't lie on the same plane**. You can imagine it: if you would extrude¹ your triangle linearly, this side will become just a rectangle, in case of a simple rotation it would be a trapezoid, but we need rotation with a little vertical shift, which bends this trapezoid out of it being a two-dimensional figure.

The solution is trivial now: just split these faces into triangles, e.g. the `[0, 2, 5, 3]` may become `[0, 2, 5], [0, 5, 3]`. At least, such an approach solved the issue for me.

¹ Well, actually not exactly cryptic. Some point didn't appear on some circle is very close to what happened, and it was looking into `polygon_mesh_to_nef_3.h` code which sparkled the idea for me.
² I wish **openscad** had an option to do rotational extrude with a helix shift, so we won't have to manually play with coordinates and faces.

kintel, 2024-03-29:
Status as of 2024-03-24: Works when using Manifold