openscad/openscad #2413: Geometry ignored with some floating point values

yohcop, 2018-05-31:
I stumbled upon this bug which is hard to believe. I did not manage to make a simpler example, as commenting pretty much any those lines make things work.

It's basically:
- a square
- out of this square, I subtract 2 "notches", which are made of a square and 2 circles.
- the whole thing (which is 2d) is linear-extruded

Depending where the circles are exactly, only the first notch is drawn, then the rest of the geometry seems to be ignored. I had a for loop for notches, and same thing: only the first one was drawn correctly.

But, if some operations are slightly modified, things appear to work as expected. For example (those are commented with `XXX` in the code):
- removing the linear_extrude()
- removing the difference()
- changing the "f" variable in the notch() function to something else (but not _anything_)

The `f` variable in the `notch` module at the bottom seems to be the issue. I originally found the problem by using `f = 1/sqrt(2)*drillDiam/2`. Using this value, things fail as described. If I change the value to say 0.07, things suddenly work. 0.06, and things fail again (note that using 0.07 and 0.06 draw the circles in a location that can't be seen, but you can still see them with `#`, etc).

We looked at it with @justbuchanan but we didn't really have any more ideas at this point.

```
// OpenSCAD version 2015.03-2 
$fn = 60;
woodDepth = 0.72;  // 3/4 plywood. ish. Double check.
// 58 inches can be done in 2 halves.
tableSize = 58;  // 29
tableDepth = 32; // 16
shelfHeight = 10;
boxClearance = 1/8;
boxWidth = tableSize / 2 - tableDepth / 2 - woodDepth - 2*boxClearance;
boxHeight = shelfHeight - boxClearance;

//===========================================================

boxFront();

module boxFront() {
  // XXX: Just commenting the linear_extrude makes things work.
  linear_extrude(woodDepth) {
    // XXX: Also removing the difference() makes things work.
    difference() {
      square([boxWidth, boxHeight]);
      notches(0.125, 1, 1, boxWidth);
    }
  }
}

// ==============================================

module notches(drillDiam, size, spacing) {
  // XXX: Alternatively, commenting this first notch drawing also makes things work.
  translate([0,0]) notch(drillDiam, size);
  translate([size+spacing, 0]) notch(drillDiam, size);
}

module notch(drillDiam, size) {
  square([size, woodDepth]);
  f = 1/sqrt(2)*drillDiam/2;  // doesn't work
  // f = 0.707*drillDiam/2;  // doesn't work
  // f = 0.06; // doesn't work.
  // f = 0.07; // works (circles are outside the material though, but it works).
  echo(f);
  // XXX: Commenting those 2 lines then work (the square above is correctly drawn).
  translate([f, woodDepth-f]) circle(d=drillDiam);
  translate([size-f, woodDepth-f]) circle(d=drillDiam);
}

```

Thanks!

thehans, 2018-05-31:
I tested some variations and tried creating a simplified example:

```
$fn = 20;
//$fn = 50; // some $fn values work ok with the geometry

y_extension = 0; // nonzero value such as 0.01 here fixes it

linear_extrude(1) // comment linear_extrude and run F6 for an interesting display
  difference() {
    square(40,40);
    union() // commenting union fixes it
    { 
      square(10,10);
      translate([0,10]) circle(1);
    }
    // not being differenced from larger square in some cases
    translate([20,-y_extension]) square(10,10+y_extension);
  }
```

The issue seems to be something related to the squares being perfectly on the border, as using a y offset to extend beyond the edge avoids the issue.  I didn't think coincident edges was particularly a problem for 2D csg operations, but maybe so?  

Also interesting is that if you only comment out the linear extrude, the preview looks good, but the F6 render shows the second square border but the center is filled in.

thehans, 2018-05-31:
Its a strange interaction that seems to involve the circle somehow, despite it not directly touching the second small square directly.

MichaelAtOz, 2018-05-31:
First, ignore the linear extrude, garbage-in-out:
[image]

MichaelAtOz, 2018-06-01:
Normally I would put this down to coincident lines, and as thehans shows it is easily fixed that way.
But the 'spooky action at a distance' of the circles is interesting, and should be pursued as it may point to some fundamental bug.

kintel, 2018-07-01:
This is definitely due to coincident lines. Offsetting the notches a tiny bit fixes the problem.

It can be a bit confusing to diagnose this, as evaluation is done differently in preview and in render mode:
* In 2D preview mode, we just give all 2D object a thickness, and let OpenGL wrangle the coincident surfaces, which usually looks ok, but may come with some flickering due to z-fighting.
* In 2D render mode, we fully evaluate the 2D object. This is done using fixed point arithmetic, so coincident surfaces won't play well
* With linear extrude in preview mode, we actually perform a full 2D evaluation, so the rendering issues will correspond to what you'd see in 2D render mode.

I believe the weird circle interaction that @thehans pointed out is just some artifact of how ClipperLib (http://www.angusj.com/delphi/clipper.php) handles such edge cases.

thehans, 2021-06-10:
I just tested this again (the example from the original post) and it now causes a crash from nightly.

Running a debug build, i found that it hits the assert here, related to @redlizard's recent context/garbage collection changes:
https://github.com/openscad/openscad/blob/5e1a7cddd26de6fcfee753ee1d0fde5c90f785cd/src/context-mm.cc#L350

The value I saw for the size looked like it had underflowed to 2^64-1 or so.

thehans, 2021-06-14:
An update about the crash, it only happens when "Stop on first warning" preference is enabled (or `--hardwarnings` option from command line).  So it is maybe something about the stack trace that attempts to print.  

Also requires a debug build, where `NDEBUG` is *not* defined, for the assert to trigger.  Not sure if a crash would still occur without assertions active.

kintel, 2024-03-29:
No longer crashes

kintel, 2024-03-29:
Clipper coincident line issue still applies - let's keep this open and re-test against Clipper2 when we're ready.