openscad/openscad #2414: CGAL error when performing difference()

JorgDr, 2018-06-02:
Hi there,

I am fairly new at using OpenSCAD, so my knowledge is pretty limited on what it can and can't do, but I am getting the following F6 output, whereas F5 seems to work fine.

Compiling design (CSG Tree generation)...
Rendering Polygon Mesh using CGAL...
ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: e->incident_sface() != SFace_const_handle() File: /opt/mxe/usr/x86_64-w64-mingw32.static/include/CGAL/Nef_S2/SM_const_decorator.h Line: 326 
ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: e->incident_sface() != SFace_const_handle() File: /opt/mxe/usr/x86_64-w64-mingw32.static/include/CGAL/Nef_S2/SM_const_decorator.h Line: 326 
Geometries in cache: 4
Geometry cache size in bytes: 38160032
CGAL Polyhedrons in cache: 2
CGAL cache size in bytes: 0
Total rendering time: 0 hours, 0 minutes, 30 seconds
Rendering finished.

My code is:

module MyObject()
{
  import ("D:/Downloads/Dining Vase_fixed.stl");
}

// MyObject();
 

difference () {
  translate([0.001,0,0]) MyObject();
    scale ([0.95,0.95,1.01]) MyObject();

}

When I only render MyObject() it works with both F5 and F6, so the object itself seems to be renderable, yet when that same object is shrunk and subtracted from itself (I am trying to make the object hollow by shrinking the x & y axis).  The object is an STL file that was fixed with NetFabb online and looking at it with Meshmixer seems to be solid.  It's a geometric solid shape.  Rendering it with F6 works as well, the problem is in the difference() function.

Any help would be greatly appreciated.
[Dining Vase_fixed.zip](https://github.com/openscad/openscad/files/2064805/Dining.Vase_fixed.zip)

t-paul, 2018-06-02:
Please have a look at this FAQ entry:
[Why is my imported STL file only showing up with F5 but not F6?](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ#Why_is_my_imported_STL_file_only_showing_up_with_F5_but_not_F6?)

Specifically the MeshLab part for removing zero faces as this type of STL issue is not considered a problem in all mesh fixing tools.

JorgDr, 2018-06-02:
Thanks for the quick reply t-paul.  I do search the internet high and low for problems I encounter prior to asking for help and had come across the FAQs.

The reason I thought my stl file was valid is because F6 does render it when I don't use the difference() function on it.  I thought it was weird that F6 can render the object by itself but when performing a difference calculation on it, F6 fails.  That led me to believe the problem was with the difference calculation not the object per se.

I just installed MeshLab and after performing a Filters -> Cleaning and Repairing -> Remove T-Vertices by Edge-Flip it tells me successfully removed 0 t-vertices.  After saving the file, OpenSCAD fares no better.  MeshLab does not show faults with the model.

Even after trying the Cleaning and Repairing functions of remove duplicate vertices, remove faces from non manifold edges and remove isolated folded edges by edgeflip, nothing seems to be able to create an stl file acceptable by OpenSCAD for F6 processing.   I probably need to spend more time with Meshlab to become fully proficient in it, but there don't seem to be any obvious problems with the model.   Compute geometric measures has no problems computing volume, surface area, etc.

nophead, 2018-06-02:
My understanding is OpenSCAD snaps vertices to a grid when it imports STL files. If vertices are closer than the grid spacing they merge and the topology is corrupted. So you can't import some manifold STL files.

JorgDr, 2018-06-03:
Thanks for that explanation.  It would explain why something that has no mesh errors still does not import.

MichaelAtOz, 2018-06-03:
If that is the case you should be able to scale it up in a tool, import, scale down in OpenSCAD.

JorgDr, 2018-06-03:
If that is the case, low facet count large object STL files would have a better chance of importing than highly detailed small objects.

nophead, 2018-06-03:
I have had problems with parts of an extruder that were generated by a CAD program. Not particularly small or having small features. They just happened to have some close vertices due to being exported at fairly high resolution.

OpenSCAD only imports a small subset of manifold STL files that doesn't even include all the ones it can make.

t-paul, 2018-06-03:
Blender does flag 28 zero faces with this model.

t-paul, 2018-06-03:
I could not get it working with MeshLab either, the following file is created with Blender by selecting the zero faces flagged by the (built-in add-on) 3d printing toolkit, removing those and then using the "make manifold" function of the 3d printing toolkit.

[Dining Vase_fixed2.zip](https://github.com/openscad/openscad/files/2065840/Dining.Vase_fixed2.zip)

From technical perspective that seems to work now in OpenSCAD, I may need a close inspection that the procedure did not cause any bad results in the model though.

JorgDr, 2018-06-04:
Hi guys,
thank you very much for looking into my rendering problem.   I guess I should also install Blender and learn it, so I have more tools at my disposal identifying problems.  So far I have always relied on MeshMixer for quickly checking STL consistency, which obviously isn't enough.

Even the Blender fixed Dining Vase from t-paul errors depending on what action is performed on it with a 
ERROR: CGAL error in CGALUtils::applyBinaryOperator difference: CGAL ERROR: assertion violation! Expr: itl != it->second.end() File: /opt/mxe/usr/x86_64-w64-mingw32.static.posix/include/CGAL/Nef_3/SNC_external_structure.h Line: 1152 

This error is easily achieved by trying to slice the bottom of the Vase after trying to make it hollow using this code:

difference () {
    MyObject();
    scale ([0.95,0.95,1]) MyObject();
    translate ([-200,-200,0]) cube([400,400,20]); 
}

At this stage I think I will not use externally created STL files with OpenSCAD, but rather use it for projects that require no existing external input, which is most of them anyway.  I obviously have a lot to learn so I know which tool is best to use for the job at hand.

Thanks again for investigating this.

stuartpb, 2018-10-23:
Similar to https://github.com/openscad/openscad/issues/1081#issuecomment-351283652, I'm having this issue again, but with a file that, by all accounts, should be fine. See [this model](https://github.com/openscad/openscad/files/2503875/Cultist-cleanest.zip) - it appears to be manifold, and yet I get `ERROR: CGAL error in CGAL_Nef_polyhedron3(): CGAL ERROR: assertion violation! Expr: e->incident_sface() != SFace_const_handle() File: /usr/include/CGAL/Nef_S2/SM_const_decorator.h Line: 330` with something as simple as `difference() { import("Cultist.stl"); cube(); }`

Philius, 2019-02-24:
I'm getting this error with internally generated meshes
[hypocycloid-one.scad.tar.gz](https://github.com/openscad/openscad/files/2898296/hypocycloid-one.scad.tar.gz)
I included the json file for presets.
I tried scaling the model up but that didn't work.
In the Customizer the 10:1 preset renders fine but "100:1" and "100:1 scaled" presets doesn't.

Philius, 2019-02-27:
As of version 2019.02.26.nightly (git b6aa7c8) the intel driver has problems.
The NVidia driver doesn't render the preview when it opens the file, otherwise it works.

nophead, 2019-02-27:
Does it render if you hit F5? I think the current master doesn't automatically render the file when you open it.

Philius, 2019-02-27:
Sorry I meant to say "draw the preview".
When I open a (problem?) file the display doesn't change until I select a preset from the customizer.

nophead, 2019-02-27:
Yes the current master doesn't preview when you open a file with "auto reload and preview" selected. Also F4 does nothing but F5 will render. I fixed it with this PR https://github.com/openscad/openscad/pull/2821

codekansas, 2021-02-14:
Had this error as well, fixed by making my faces point inwards - remember to follow right-hand rule when specifying the indices in each face

kintel, 2024-03-30:
Status as of 2024-03-24: 
* Dining Vase Fixed: Works with Manifold, still broken with CGAL
* Cultist cleanest: Broken with both CGAL and Manifold
* Hypocycloid one: Works with both Manifold and CGAL

stefano2734, 2026-09-20:
Last test here in 2024 with CGAL 6.0 or previous.
So perhaps a new test can show a new solved behavior.