openscad/openscad #2415: QtMultimedia5 missing in Xubuntu 16.04

ednisley, 2018-06-02:
This seems to be due to the recent "play a happy sound" change (4b6587f031ec4d778e6e4b35703d8067a765af8c):

`src/mainwin.cc:99:31: fatal error: QtMultimedia/QSound: No such file or directory
compilation terminated.
Makefile:2951: recipe for target 'objects/mainwin.o' failed
make: *** [objects/mainwin.o] Error 1`

The `check-dependencies.sh` script seems happy and running `uni-get-dependencies.sh` didn't make any changes, but even after manually installing `qtmultimedia5-dev` and a few other likely packages, the compile step still fails.

I'm still running Xubuntu 16.04 LTS, so perhaps I must bump this box to 18.04 LTS.

Any other hints or recommendations?

Thanks ...

t-paul, 2018-06-03:
Did you re-run `qmake`? Only other idea I have is to make sure that the compilation is actually using Qt5 by running the matching `qmake`, e.g. `QT_SELECT=5 qmake` or whatever Xubuntu uses.

The package name looks correct. That is also used on [OBS](https://build.opensuse.org/package/show/home:t-paul/OpenSCAD) where the compilation works for Ubuntu 16.04.

ednisley, 2018-06-03:
Forcing QT_SELECT=5 nudges the error down a few lines:

`In file included from src/MainWindow.h:6:0,
                 from src/mainwin.cc:31:
objects/ui_MainWindow.h:13:25: fatal error: QtGui/QAction: No such file or directory`

I'll move 18.04 LTS higher on my to-do list, which should solve the problem and eliminate any reason to dig deeper into this steaming pile of rubble.

t-paul, 2018-06-03:
That is usually caused by a mix of both Qt4 and Qt5, so somewhere there may still be some Qt4 generated files around.

DeuxVis, 2018-06-03:
FYI I just built openscad on an ubuntu 14.04 - yeah I should really update that - featuring a mix of QT4 and QT5 development packages.
My procedure was to manually install qtmultimedia5-dev and then :
```
qmake -qt=qt5
make -j6
sudo make install
```

ednisley, 2018-06-03:
*Eeek*: there's an unholy mix of 32 bit Qt4 and 64 bit Qt5 inside. I'll restart from scratch after installing 18.04.

Let this be a warning to anybody else behind the times.

Thanks ...

t-paul, 2018-06-03:
Yes, Debian (and therefor Ubuntu too) does normally support both Qt4 and Qt5 development packages at the same time. So with clean environment it should be possible to work with both.

That said Qt4 support in OpenSCAD will be phased out now as newer distributions will not ship that anymore.

DeuxVis, 2018-06-05:
On another machine running Ubuntu 14.04.2 LTS I have had total success without any code modification.
Procedure used :
```
git clone https://github.com/openscad/openscad.git
cd openscad/
sudo apt update
sudo ./scripts/uni-get-dependencies.sh
./scripts/check-dependencies.sh
sudo apt install qtmultimedia5-dev
QT_SELECT=5 qmake -qt=qt5 CONFIG+=experimental
make -j2
sudo make install
```

brycepg, 2018-07-07:
I had a similar issue for Fedora 27. Using qt4 is a dead end. I had to do:

```
make clean
QT_SELECT=5 qmake-qt5 CONFIG+=experimental
```

Cqoicebordel, 2018-08-10:
In 18.04 the `qmake openscad.pro` step failed.
I ran `sudo apt-get install qtmultimedia5-dev libqt5multimediawidgets5 libqt5multimedia5-plugins libqt5multimedia5` and then everything worked.
Yeah, I went overboard with all those packages, but just adding `qtmultimedia5-dev` to the dependency check might be enough.

t-paul, 2018-08-10:
Yes, WIP pull request in #2418.

t-paul, 2018-09-30:
Closed via d1c999f4afc80f33e1075fee3778e6989c8516d9

closed 2018-09-30 by t-paul