openscad/openscad #2417: defaults to AkrutiMal1 font on KDE Manjaro Linux

andpeterson, 2018-06-06:
I just installed Manjaro in my laptop and downloaded OpenSCAD. When I started typing it defaulted to some garbage text. When investigating the settings I found the font was on AkrutiMal1 which is a symbol based font.

donbright, 2018-08-04:
i have tested OpenSCAD on a large number of distros late last year, and there are a lot of messed up systems with this same kind of problem. 

i think we should stop defaulting to monospace on linux/bsd. monospace is not that important and a lot of system just do not do it properly 'out of the box'.

t-paul, 2018-11-10:
Hum, turns out the font lookup is not really the problem. The  `Q_OS_X11` does not actually exist so the lookup effectively just used an empty string causing it to use the first matching font.
https://github.com/openscad/openscad/blob/89b649c312f7ddd708eda2d49256f5c49e66895b/src/Preferences.cc#L109-L111

commit cc506620 (2018-11-11): Fix non-existing Q_OS_X11 and add Qt5.2 based systemFont() lookup (fixes #2417).
commit 3bc18f05 (2018-11-12): Fix non-existing Q_OS_X11 and add Qt5.2 based systemFont() lookup (fixes #2417).
closed 2018-11-12 by kintel