openscad/openscad #2419: No syntax highlighting in macOS 10.13.4

maccesch, 2018-06-09:
This morning I had a build from January and syntax highlighting didn't work. So I built the latest version with Homebrew but still in the preferences in the dropdown for syntax highlighting there's only the option "Off". I'm using qcsintilla of course. Is there sth wrong they way I build it? I couldn't find an error or warning that seems to be related.

t-paul, 2018-06-09:
Maybe can't find the `color-schemes/editor/` folder with the json files.

kintel, 2018-06-09:
See https://github.com/openscad/openscad/blob/master/src/PlatformUtils.cc#L29

The resources are searched based on the location of the executable. If you built in manually, you may end up with the application path not being registered correctly relative to to where these resources are looked up.

maccesch, 2018-06-10:
Yes that actually was the problem. In the .app folder that is created during building the subolder `Resources` was missing `color-schemes`. Also `fonts`, `libraries` and `locale` were missing. After copying all that manually in there everything worked just fine.

I did a little bit of searching and found that there is a script https://github.com/openscad/openscad/blob/21ac6f2298b083c30e7df301205d85429144a32d/scripts/release-common.sh#L335 which has code to copy all this into the `Resources` folder but when I tried to run it I got 

```
src/SparkleAutoUpdater.mm:15:10: fatal error: 'Sparkle/Sparkle.h' file not found
```

kintel, 2018-06-10:
If you're ok with running OpenSCAD from the build directory, you shouldn't even need to copy any resources into the app bundle.
Sparkle is needed if you build deployment builds. It's possible to run without Sparkle, but you'll need to tweak various scripts. That's not really a use-case we've optimized for.

maccesch, 2018-06-11:
Sure. Wouldn't it make sense to extract the resource copying part of that script to maybe reuse it in the `scripts/macosx-build-*` scripts?

kintel, 2018-12-07:
The supported workflow right now is to run `./scripts/publish-macosx.sh buildonly`.
If you're looking for specific convenience for e.g. homebrew builds, please open a new ticket with a suitable title.

closed 2018-12-07 by kintel