openscad/openscad #2421: Trouble compiling on Solus - fatal error: stdlib.h: No such file or directory

arcooke, 2018-06-22:
Hey all.  I'm not sure how much this has to do with openscad but I was hoping someone might be able to point me in the right direction.  I'm not a c++ developer

None of the v2018 packages (snap/appimage) work in Solus so I'm trying to compile from source and I've hit a snag. All my dependencies are met and I have build tools installed (equivalent to build-essential), but I'm getting this error 

```
In file included from /usr/include/boost/config/platform/linux.hpp:15:0,
                 from /usr/include/boost/config.hpp:57,
                 from /usr/include/boost/filesystem/config.hpp:28,
                 from /usr/include/boost/filesystem.hpp:15,
                 from src/openscad.h:29,
                 from src/openscad.cc:27:
/usr/include/c++/7.3.0/cstdlib:75:15: fatal error: stdlib.h: No such file or directory
 #include_next <stdlib.h>
               ^~~~~~~~~~
compilation terminated.
make: *** [Makefile:3736: objects/openscad.o] Error 1
```

```
depname     minimum     found       OKness      
qt          4.4         5.11.0      OK          
qscintilla2 2.7         2.10.2      OK          
cgal        3.6         4.9         OK          
gmp         5.0         6.1.2       OK          
mpfr        3.0         3.1.6       OK          
boost       1.35        1.66        OK          
opencsg     1.3.2       1.4.2       OK          
glew        1.5.4       1.7.0       OK          
eigen       3.0         3.3.3       OK          
glib2       2.0         2.56.1      OK          
fontconfig  2.10        2.13.       OK          
freetype2   2.4         2.8.0       OK          
harfbuzz    0.9.19      1.7.5       OK          
bison       2.4         3.0.4       OK          
flex        2.5.35      2.6.4       OK          
make        3           4.2.1       OK
```

```
locate stdlib.h
/usr/include/bits/stdlib.h
/usr/include/c++/7.3.0/stdlib.h
/usr/include/c++/7.3.0/tr1/stdlib.h
/usr/include/freetype2/freetype/config/ftstdlib.h
/usr/include/stdlib.h
/usr/include/wine/msvcrt/stdlib.h
```

Any thoughts how I can get past this?  Thanks

kintel, 2018-06-22:
This sounds a bit like an issue with your compiler environment.
I would try to build a simple C++ program that uses the `<algorithm>` header and try to see how that compilation cmd-line differs from the OpenSCAD one.

arcooke, 2018-06-22:
Thanks.  I didn't think this was important at first, but since you mentioned `<algorighm>` specifically.. it might be.

While troubleshooting I thought I would try installing `qt4-devel` and running `qmake-qt4` instead to see if that helped.. that's when I got the error mentioned above.  Before I installed qt4, I was getting the same error from a different source file

```
In file included from /usr/include/boost/config/platform/linux.hpp:15:0,
                 from /usr/include/boost/config.hpp:57,
                 from /usr/include/boost/filesystem/config.hpp:28,
                 from /usr/include/boost/filesystem.hpp:15,
                 from src/openscad.h:29,
                 from src/openscad.cc:27:
/usr/include/c++/7.3.0/cstdlib:75:15: fatal error: stdlib.h: No such file or directory
 #include_next <stdlib.h>
               ^~~~~~~~~~
compilation terminated.
make: *** [Makefile:3736: objects/openscad.o] Error 1
```

If this isn't related to openscad in any way I can ask elsewhere, I don't mean to misplace my question

kintel, 2018-06-22:
Still looks related to you compiler config. Did you try compiling anything else from source?

justbuchanan, 2018-06-22:
Could this be relevant?

https://www.reddit.com/r/SolusProject/comments/57o8sq/fatal_error_stdioh_no_such_file_or_directory/

They suggest running `sudo eopkg it -c system.devel` to install the right bits for compiling.

arcooke, 2018-06-23:
@kintel I just tested [compiling GZDoom](https://zdoom.org/wiki/Compile_GZDoom_on_Linux) and that went fine

@justbuchanan I do already have that installed

rewolff, 2018-06-24:
the "stdlib.h" is a standard (that's what the std stands for!) part of the compiler. 

It SHOULD be located in a place where the compiler can find it. Say for instance on most computers it will be in "/usr/include/stdlib.h". It is on your system! That leaves: your computer is not looking in the standard locations for the header files. The "my first C program" looks like: 
`#include <stdio.h>
#include <stdlib.h>
int main (int argc, char **argv)
{
  printf ("Hello world\n");
  exit (0);
}`
I just checked: This also counts as valid C++. So put this in a file, and try to compile that. When that compiles, you're in better shape to compile anything.....

t-paul, 2018-11-10:
I believe this is caused by broken Qt/qmake setup. On Solus I get
```
qmake -query | grep QT_INSTALL_HEADERS
QT_INSTALL_HEADERS:/usr/include
```
which translates to a `-isystem /usr/include` and that breaks the g++ `#include_next` logic as it moves the `/usr/include` path before the g++ stuff instead of the default position at the end.

You can workaround the problem by editing the `Makefile` generated by `qmake`. In the line that starts with `INCPATH` remove the `-isystem /usr/include`.

I don't see any way to fix that on our side, so for a proper fix this needs to be reported to Solus and fixed there.

closed 2018-11-10 by t-paul

t-paul, 2018-11-10:
FWIW, I added Solus to the `uni-get-dependencies.sh` script so on a fresh Solus installation, OpenSCAD compliation works with
```
sudo eopkg -y install git
git clone https://github.com/openscad/openscad
cd openscad
sudo bash scripts/uni-get-dependencies.sh
qmake CONFIG-=debug CONFIG+=experimental CONFIG+=snapshot
sed -ie 's,-isystem /usr/include ,,' Makefile
make
```
The `sed` call is the "automatic" version of the workaround mentioned above.

Looks like Solus has also the font issue (#2417) but that can be solved by just changing the font in preferences.

sixtyfive, 2019-01-02:
Works for me! #34C3 ;-) (Solus w/ latest updates applied).