openscad/openscad #2422: Feature request: Customizer.

rewolff, 2018-06-24:
Suppose I have an openscad program:
```
blocksize = 10;
holesize = 3;
difference () {
  cube (blocksize, center=true);
  translate ([0,0,-blocksize/2-.1]) cylinder (d=holesize,h=blocksize+0.2, $fs=0.1);
}
```
Now I would like to change the parameters at the beginning. (of course in practice, things will quickly become more complex). It would be nice that openscad could offer an extra window that would provide me with a slider for "blocksize" and "holesize". 
Similarly, I'm working on an extruder for a 3D printer. The different parts can be enabled and disabled in the openscad code, but just having a little gui box that allows me to turn them on and off at will would make this project much easier to work with. 

I would suggest that developers can annotate their code with hints for the customizer module. Say: 
`# blocksize slider 1 100` or 
```
# attachment_type dropdown
# attachment_type option 1 none
# attachment_type option 2 horizontal
# attachment_type option 3 vertical
```
When the user has customized his object, the customizer gui needs to be able to save the settings. I've thought about editing the source to write the current settings there. I don't think that'd be nice. I normally edit the design in "emacs", which would warn me about the file having changed on disk. With the way the openscad language is structured, reading a settings file after reading the main file (so after reading test.scad, you automatically read test.settings that contains something like `blocksize=20;` overruling the blocksize=10; in the test.scad file.)

I think this addition would  smoothen entry into openscad usage for a bunch of people. They would be able to do their familiar point-and-click to customize openscad objects and then be drawn to do slightly more interesting work on the model once the customizer no longer is enough....

But still, power users like me would also benefit! 

What do you guys think?

rewolff, 2018-06-27:
Thanks. 
In response to your suggestion to get a newer openscad: 
I upgraded my machine to 18.04 and..... openscad gone! 
I went to openscad download and followed the directions to use the PPA for openscad releases..... openscad back to 15.03!
I cloned the source, installed 20 dependencies, hacked Makefiles, created symbolic links to trick the include path to  find the proper includes and... now I have a recent openscad! :-)

t-paul, 2018-06-27:
Not sure why you needed any hacks, but recent OpenSCAD sounds good :-).

Note that there is an automatically updated [nightly-build](http://www.openscad.org/downloads.html#snapshots) on [OBS](https://build.opensuse.org/package/show/home:t-paul/OpenSCAD) for Ubuntu 18.04 which tracks the latest git master (and has the experimental features enabled).

rewolff, 2018-06-27:
In the root of openscad I now have: 
  ./QtGui -> /usr/include/x86_64-linux-gnu/qt5/QtWidgets
qscintilla.h was including stuff that on my system lives in Qt5/QtWidgets/<name> , while it was looking for QtGui/<name> 
Because -I. , this link local to openscad fixes this for openscad. 
After upgrading I few systems I got a bit more bold and upgraded an in-use workstation. That one suddenly started calling user-2 by the user-1 name. (changed to protect the innocent). So I rebooted and... Bricked. Sigh. God I love upstart.