openscad/openscad #2424: OpenSCAD 2018.03.17 (git 3e7fbfff) crashes when rendering text()

thrasher, 2018-07-01:
The latest development version of OpenSCAD crashes when it encounters text().  The latest release version (~2015) does not crash.

Here's an example file that will crash it:
crash-test.scad
text("hello world"); // there's only one line in this file

=== OpenSCAD Help/Library Info ========
OpenSCAD Version: 2018.03.17 (git 3e7fbfff)
System information: Mac OS X Version 10.13.5 (Build 17F77) MacBookPro12,1 2 CPUs 16.00 GB RAM 
Compiler: Clang "9.0.0 (clang-900.0.39.2)" 64bit
Boost version: 1_65_1
Eigen version: 3.3.3
CGAL version, kernels: 4.11, Cartesian, Extended_cartesian, Epeck
OpenCSG version: OpenCSG 1.4.2
Qt version: 5.10.0
QScintilla version: 2.9.3
MingW build: No
GLib version: 2.54.2
libzip version: 
Application Path: /Applications/OpenSCAD.app/Contents/MacOS
Documents Path: /Users/jason/Documents
Resource Path: /Applications/OpenSCAD.app/Contents/Resources
User Library Path: /Users/jason/Documents/OpenSCAD/libraries
User Config Path: /Users/jason/Library/Application Support/OpenSCAD
Backup Path: /Users/jason/Documents/OpenSCAD/backups
OPENSCADPATH: 
OpenSCAD library path:
/Users/jason/Documents/OpenSCAD/libraries
/Applications/OpenSCAD.app/Contents/Resources/libraries

OPENSCAD_FONT_PATH: 
OpenSCAD font path:
GLEW version: 1.13.0
OpenGL Version: 2.1 INTEL-10.34.27
GL Renderer: Intel(R) Iris(TM) Graphics 6100
GL Vendor: Intel Inc.
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes

Qt graphics widget: QGLWidgetGL Extensions:
GL_APPLE_aux_depth_stencil
GL_APPLE_client_storage
GL_APPLE_element_array
GL_APPLE_fence
GL_APPLE_float_pixels
GL_APPLE_flush_buffer_range
GL_APPLE_flush_render
GL_APPLE_object_purgeable
GL_APPLE_packed_pixels
GL_APPLE_pixel_buffer
GL_APPLE_rgb_422
GL_APPLE_row_bytes
GL_APPLE_specular_vector
GL_APPLE_texture_range
GL_APPLE_transform_hint
GL_APPLE_vertex_array_object
GL_APPLE_vertex_array_range
GL_APPLE_vertex_point_size
GL_APPLE_vertex_program_evaluators
GL_APPLE_ycbcr_422
GL_ARB_color_buffer_float
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_draw_buffers
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_instanced
GL_ARB_fragment_program
[... 585 lines cut ...]
Stack                             11.6M        9 
VM_ALLOCATE                        384K       21 
__DATA                            28.4M      299 
__FONT_DATA                          4K        2 
__GLSLBUILTINS                    2588K        2 
__LINKEDIT                       201.4M       42 
__TEXT                           244.2M      294 
__UNICODE                          560K        2 
mapped file                       59.4M       40 
shared memory                     3256K       19 
===========                     =======  ======= 
TOTAL                            960.2M      850 

Model: MacBookPro12,1, BootROM MBP121.0176.B00, 2 processors, Intel Core i7, 3.1 GHz, 16 GB, SMC 2.28f7
Graphics: Intel Iris Graphics 6100, Intel Iris Graphics 6100, Built-In
Memory Module: BANK 0/DIMM0, 8 GB, DDR3, 1867 MHz, 0x02FE, 0x4544464232333241314D412D4A442D460000
Memory Module: BANK 1/DIMM0, 8 GB, DDR3, 1867 MHz, 0x02FE, 0x4544464232333241314D412D4A442D460000
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x133), Broadcom BCM43xx 1.0 (7.77.37.31.1a9)
Bluetooth: Version 6.0.6f2, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
Serial ATA Device: APPLE SSD SM0256G, 251 GB
USB Device: USB 3.0 Bus
USB Device: Internal Memory Card Reader
USB Device: Bluetooth USB Host Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 27.1

joshcarter, 2018-07-28:
Same here, stack trace looks the same. Same version of OpenSCAD, also running on MacOS. Any call to text() is an insta-crash.

jmckeown2, 2018-09-05:
Additional data point: text() did not crash on my 2014 MBP running version 2018.03 but DOES crash on my new 2018 MPB. and I'll confirm reverting to the OpenSCAD ~2015 release does not crash.

biggreenogre, 2018-12-06:
MacBookPro 2010 (Sierra 10.12.6) and MacBookPro 14,2 Intel Iris Plus Graphics 650 1536 MB (High Sierra 10.13.6). X is installed on both with Xcode tools plus MacPorts.  Same issue. Tried 18.03.17 and some interim nightlies. Still present 2018.11.25. Works fine on 2015.03-3 but no success with any later versions that I have tried.
What information should I provide?
Hints on debugging actions would be greatly appreciated.  Temp files/cache/libs that should removed? Config/debug settings to tweak? Any other tests to try?

kintel, 2018-12-06:
I still haven't managed to reproduce this. Just tested on a fresh Mojave install, but no dime.
Could you try running a nightly build of OpenSCAD on the cmd-line like this and report the output?
```
./OpenSCAD.app/Contents/MacOS/OpenSCAD --debug=FontCache,FreetypeRenderer text.scad -o out.png
```

biggreenogre, 2018-12-10:
Similar result on MacBookPro 14,2 High Sierra:

~/scratch/openscad/OpenSCAD.app/Contents/MacOS/OpenSCAD --debug=FontCache,FreetypeRenderer text.scad -o out.png
Debug on. --debug=FontCache,FreetypeRenderer
FreetypeRenderer: Detected script 'Latn' for hello world
FreetypeRenderer: Detected direction 'ltr' for hello world
Fontconfig warning: line 144: blank doesn't take any effect anymore. please remove it from your fonts.conf
FontCache: font = "", lookup = "Liberation Sans:style=Regular"
Segmentation fault: 11

biggreenogre, 2018-12-18:
On further testing, does NOT crash on MacBookPro 2010 (Sierra 10.12.6). Apologies for the misinformation.

jmckeown2, 2018-12-21:
I also retested. Same 2018 MBP running version 2018.03 as mentioned in my previous post. MacOS is fully-patched Mojave (10.14.2)
text("Hello World"); does NOT crash anymore. Also opened the previous project that was crashing and it works fine now too. This used to be 100% reproducible on my Mac. So it may a macOS patch change triggered then corrected the crash?

kintel, 2018-12-27:
Looking at the code: This would probably crash if the font wasn't found when OpenSCAD is build in debug mode: https://github.com/openscad/openscad/blob/master/src/FontCache.cc#L324

This should be fixed. I'll see if I can force it to fail now that I see the issue.

kintel, 2018-12-27:
Update: The crash is fixed in #2657.

The underlying problem is that the requested font couldn't be found, which is a bit harder to track down. Please open a separate issue if any issues related to font selection are reproducible.

jmckeown2, 2018-12-27:
The one thing I would add is that it was crashing when the scad simply contained `text(“hello world”);` 
So it’s not just that a font couldn’t be found but that the DEFAULT font could not be found. There should be a usable font if nothing is specified.

kintel, 2018-12-27:
Yup, the default font should be managed by fontconfig, so this smells like a broken fontconfig setup. I'll keep an eye out for that, but it's still hard if it cannot be reproduced : /

closed 2018-12-28 by kintel

nospam2000, 2019-01-09:
I called `sudo font_cache -s -v `on my Mac which gave the following error message:
```
font_cache is already running with PID 0.
If you believe this to be erroneous, please remove /opt/X11/var/run/font_cache.lock.
```

After deleting `/opt/X11/var/run/font_cache.lock` (it was from 2015) and calling the command again, it succeeded.
Then I started OpenSCAD and used the command `text("abc");`. This time it didn't crash, but it updated the fontconfig and then it worked.

bgrumbine, 2019-05-11:
In case this helps someone else, I had the same or similar problem on Ubuntu. It also turned out to be a corrupt font cache.

Core dump:
`text("OpenSCAD");`
Core dump:
`text("OpenSCAD", font = "Liberation Sans");`
Core dump:
`text("OpenSCAD", font = "Liberation Sans:style=Italic");`
But this was successful:
`text("OpenSCAD", font = "DejaVu Sans");`

The Ubuntu fix is:
`$ sudo fc-cache -s -v`
or something to that effect... I actually tried a few things including:
`$ sudo rm -r /var/cache/fontconfig/*`
then
`$ sudo fc-cache -frs -v`

`$ uname -a`
`Linux M79 5.0.0-13-generic #14-Ubuntu SMP Mon Apr 15 14:59:14 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux`

`$ ./OpenSCAD-2019.01-RC4-x86_64.AppImage --version`
`OpenSCAD version 2019.01-RC4`