openscad/openscad #2425: GNU MP exception.

loricott, 2018-07-03:
Hello. I stumbled into a bug, I think. The following simple file:

$fn=12;

angle=45;
steps=3;
branches=3;
ratio=sqrt(2);
ratio=1.5;
dev=0.7;

module branch( l, iter ) {
    a=angle;
    rot=0;
    count=branches;
    factor=ratio * ( rands(1,1+dev,1)[0] ) ;
    d = l / 10;
    
    cylinder( d1=d, d2=d/factor, h=l );
    for( rot = [0:360/count:360] ) {
        //if( l > 5 ) {
        if( iter < steps ) {
            // Do something
            translate([0,0,l]) rotate([a,0,rot]) branch( l=l/factor, iter=iter+1 );
        } else {
            translate([0,0,l]) rotate([a,0,rot]) cylinder( d=d/factor, h=l/factor);
        }
    }
}

branch( l=10, iter=0 );

does not render on my Linux box using the latest released version (2015.03-2) and fails with GNU MP: Cannot allocate memory (size=18446744070406745120). It does render (albeit slowly) with the latest nightly build.

My HW/SW setup is: 

Ubuntu 16.04, 4.4.0-130-generic #156-Ubuntu SMP Thu Jun 14 08:53:28 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

i5-6600 CPU @ 3.30GHz, 4 cores, 16TByte RAM, NVidia Quadro K620

t-paul, 2018-07-03:
Renders successfully in about 4 minutes on my notebook with 2018.06.10.nightly (git c6a4856).

I'm not sure what you are asking for. If it's not happening with the latest dev snapshots, it seems to be already fixed.

loricott, 2018-07-03:
Thanks for the check. I merely wanted to report it because it is not working in the latest stable version. I thought it was useful to have it because it could point out some interesting aspects in the code. I was also a bit puzzled by the time it took to render.

kintel, 2018-07-03:
Closing as most likely fixed. @loricott if you experience this again, please reopen - this was probably some uninitialized memory somewhere that was resolved sometime in the past.

closed 2018-07-03 by kintel