openscad/openscad #2426: Customizer isn't usable until a preview has been made

DrLex0, 2018-07-03:
OS: OS X 10.11.6
Version: 2018.03.17 (git 3e7fbfff)

When I load a file, even if it has a valid .json file with customizer presets, the customizer interface remains grayed out until I hit the Preview or Render button. This is very unintuitive especially for first time users, and annoying for complex scad files that take a long time to render. I am the author of [this](https://www.thingiverse.com/thing:2812634) and I would like to make the list of instructions simpler instead of having to describe a different magic dance with every new release.
The customizer UI should be enabled after loading a customizable file.

kintel, 2018-07-03:
What do you mean by "The customizer UI should be enabled after loading a customizable file"?
..just make sure that the customizer window is available in a tab somewhere?

Note that the Features preferences will go away when built in release mode, once the customizer is generally available.

t-paul, 2018-07-03:
I think the solution would be to allow parsing only process so that the AST is generated but no preview/render is triggered.

DrLex0, 2018-07-03:
> What do you mean by "The customizer UI should be enabled after loading a customizable file"?
..just make sure that the customizer window is available in a tab somewhere?

I'm not talking about the requirement to enable the feature in Preferences. Even when it has already been enabled, opening a file that is customizer-compatible yields this:

[image]

while it should look like this:

[image]

kintel, 2018-07-03:
OK, makes sense. @MichaelPFrey Are you still working through your list of Customizer enhancements? Does this belong on that list?

MichaelAtOz, 2018-07-03:
> allow parsing only process so that the AST is generated but no preview/render is triggered.

Or, if Customizer Automatic Preview is ticked, do a Preview on load, just like Automatic Reload and Preview.

t-paul, 2018-07-03:
That does work, but could be massively annoying if the file has a big `minkowski` and preview needs a couple of minutes. Separating the parser would also later help picking up module names for auto-complete and similar stuff.

t-paul, 2018-11-09:
Tested with the included examples. Works nicely.

closed 2018-11-09 by t-paul