openscad/openscad #2431: Specify the cases of each "Advanced" string

rffontenelle, 2018-07-14:
The string "Advanced" is used twice in the UI: One as [a Tab in Preferences dialog](https://github.com/openscad/openscad/blob/master/src/Preferences.ui#L1532) and another as [a type of examples in the welcome dialog](https://github.com/openscad/openscad/blob/master/examples/examples.json#L19). For my language (Brazilian Portuguese), the translation of each "Advanced" should be different. Please add a [context to solve this ambiguity](https://www.gnu.org/software/gettext/manual/html_node/Contexts.html) and allow translators to translate accordingly.

MichaelPFrey, 2018-08-15:
[image]
[image]

MichaelPFrey, 2018-09-19:
Technically, this specific issue should now be fixed.
But going trough, I realize that that part of the underlying issue is still there.

I have document it for my own reference.

The effect of the change can be seen in the auto generated `objecs/ui_Preferences.h` 
A normal text looks like this:
`        prefsAction3DView->setText(_("3D View", Q_NULLPTR));`
and with the translation context "preferences" it looks like
`        prefsActionAdvanced->setText(_("Advanced", "preferences"));`

Note that this is
`msgid, msgctxt`

Now we also need a reference: https://www.gnu.org/software/gettext/manual/gettext.html where this can be found as `msgctxt` in 
> 11.2.5 Using contexts for solving ambiguities

So, for me to fully resolve this issue I also want to update printutils.h so that I can use a msgctxt for #2325 but that is currently a bit over my head.

MichaelPFrey, 2018-11-24:
Okay:
https://github.com/openscad/openscad/blob/4f58918f321fdf0e501a28ada0c5c5384c5dbcdd/src/qtgettext.h#L26-L30

https://github.com/openscad/openscad/blob/4f58918f321fdf0e501a28ada0c5c5384c5dbcdd/src/printutils.h#L8-L10

So we need pgettext:

https://www.gnu.org/software/gettext/manual/html_node/Contexts.html

To get pgettext, we need.

https://www.gnu.org/software/gettext/manual/html_node/lib_002fgettext_002eh.html

... and for that, we need to add gettext.h into the project and refractor some code.

t-paul, 2018-11-25:
I think the issue is that the current way tries to match `gettext` macros and Qt translation to look the same. I think https://github.com/openscad/openscad/pull/2604/commits/a46648992ca9cb81e6712547664ca7446e1b03d0 is a better solution. It makes clear there are 2 different path though translation and one (Qt) mapps to the other (gettext).

MichaelPFrey, 2018-11-25:
It is possible to emulate pgettext with just a few lines of code.
https://github.com/openscad/openscad/blob/ff859191bec45e63e94bc27f370970d31b31ef70/src/printutils.h#L11-L24
I saw that in an PHP example and double checked by locking into _gettext.h_.
Not the cleanest way, but QT using nullptr to indicate "no context" makes things tricky - so we could not use pgettext anyway.

I also prefer inlined c++ (I assume that gcc will optimize it further) over macro-magic.

I closed the original PR as that became a messy, as I tried to abuse context for comments.
Knowing now that context comes at a performance penalty (debatable how much it is, but still reason enough to not abuse it for comments).

I should write a note about context in the doc. (only use when needed, keep it short, performance penalty...)

PR #2604 (2018-12-27): updating the translation 
**Todo:**
- [x] add german translation
   - [x] defuzzy the messages
- [x] update translation doc
   - [x] only use when needed
   - [x] keep it short
   - [x] performance penalty
   - [x]  do not abuse as comment
- [x] tweak xgettext option?
   - [x] --language=c++
      - resolves `xgettext: Warnung: Typ der Datei »src/CocoaUtils.mm« mit Suffix »mm« ist unbekannt; C wird versucht`
   - [x] https://www.gnu.org/software/gettext/manual/html_node/xgettext-Invocation.html
   - -k, --keyword[=WORD]        additional keyword to be looked for **(without
WORD means not to us [... cut]
closed 2018-12-27 by kintel