openscad/openscad #2432: Nightly Build 2277.1 AppImage does not work on Ubuntu 18.04

probonopd, 2018-07-14:
```
root@host:/home/me# '/home/me/Downloads/OpenSCAD-Nightly-0-Build2277.1.glibc2.14-x86_64.AppImage'
This application failed to start because it could not find or load the Qt platform plugin "xcb"
in "".

Reinstalling the application may fix this problem.
Aborted

root@host:/home/me# cat /etc/os-release 
NAME="Ubuntu"
VERSION="18.04 LTS (Bionic Beaver)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 18.04 LTS"
VERSION_ID="18.04"
```

probonopd, 2018-07-14:
`usr/plugins/platforms/` is empty - this cannot work. How was the AppImage generated?

probonopd, 2018-07-14:
Looking at the build log at https://build.opensuse.org/public/build/home:t-paul/AppImage/x86_64/OpenSCAD/_log,

```
[  643s] ERROR: findDependencyInfo: "ldd: /usr/lib64/qt4/plugins/platforms/libqxcb.so: No such file or directory\n"
[  643s] ERROR: getBinaryRPaths: "objdump: '/usr/lib64/qt4/plugins/platforms/libqxcb.so': No such file\n"
[  643s] ERROR: file copy failed from "/usr/lib64/qt4/plugins/platforms/libqxcb.so"
[  643s] ERROR:  to "/usr/src/packages/OpenSCAD-Nightly-0-Build2277.1-x86_64/usr/bin/..//plugins/platforms/libqxcb.so"
[  648s] ERROR: findDependencyInfo: "ldd: /usr/lib64/qt4/plugins/printsupport/libcupsprintersupport.so: No such file or directory\n"
[  648s] ERROR: getBinaryRPaths: "objdump: '/usr/lib64/qt4/plugins/printsupport/libcupsprintersupport.so': No such file\n"
[  648s] ERROR: file copy failed from "/usr/lib64/qt4/plugins/printsupport/libcupsprintersupport.so"
[  648s] ERROR:  to "/usr/src/packages/OpenSCAD-Nightly-0-Build2277.1-x86_64/usr/bin/..//plugins/printsupport/libcupsprintersupport.so"
[  649s] ERROR: Could not find any translations in  "/usr/share/qt4/translations"  (developer build?)
[  649s] + linuxdeployqt /usr/src/packages/OpenSCAD-Nightly-0-Build2277.1-x86_64/usr/share/applications/openscad.desktop -bundle-non-qt-libs
[  649s] linuxdeployqt  (commit ), build <local dev build> built on 2018-07-09 18:53:20 UTC
[  649s] Desktop file as first argument: "/usr/src/packages/OpenSCAD-Nightly-0-Build2277.1-x86_64/usr/share/applications/openscad.desktop"
[  649s] desktopExecEntry: "openscad"
[  649s] desktopIconEntry: "openscad"
[  649s] Found binary from desktop file: "/usr/src/packages/OpenSCAD-Nightly-0-Build2277.1-x86_64/usr/bin/openscad"
[  649s] FHS-like mode with PREFIX, fhsPrefix: "/usr/src/packages/OpenSCAD-Nightly-0-Build2277.1-x86_64/usr"
[  649s] app-binary: "/usr/src/packages/OpenSCAD-Nightly-0-Build2277.1-x86_64/usr/bin/openscad"
[  649s] appDirPath: "/usr/src/packages/OpenSCAD-Nightly-0-Build2277.1-x86_64"
[  649s] relativeBinPath: "usr/bin/openscad"
[  649s] Found icons from desktop file: ("/usr/src/packages/OpenSCAD-Nightly-0-Build2277.1-x86_64/openscad.png", "/usr/src/packages/OpenSCAD-Nightly-0-Build2277.1-x86_64/usr/share/pixmaps/openscad.png")
[  649s] preExistingToplevelIcon: "/usr/src/packages/OpenSCAD-Nightly-0-Build2277.1-x86_64/openscad.png"
[  662s] ERROR: findDependencyInfo: "ldd: /usr/lib64/qt4/plugins/platforms/libqxcb.so: No such file or directory\n"
[  662s] ERROR: getBinaryRPaths: "objdump: '/usr/lib64/qt4/plugins/platforms/libqxcb.so': No such file\n"
[  662s] ERROR: file copy failed from "/usr/lib64/qt4/plugins/platforms/libqxcb.so"
[  662s] ERROR:  to "/usr/src/packages/OpenSCAD-Nightly-0-Build2277.1-x86_64/usr/bin/..//plugins/platforms/libqxcb.so"
[  668s] ERROR: findDependencyInfo: "ldd: /usr/lib64/qt4/plugins/printsupport/libcupsprintersupport.so: No such file or directory\n"
[  668s] ERROR: getBinaryRPaths: "objdump: '/usr/lib64/qt4/plugins/printsupport/libcupsprintersupport.so': No such file\n"
[  668s] ERROR: file copy failed from "/usr/lib64/qt4/plugins/printsupport/libcupsprintersupport.so"
[  668s] ERROR:  to "/usr/src/packages/OpenSCAD-Nightly-0-Build2277.1-x86_64/usr/bin/..//plugins/printsupport/libcupsprintersupport.so"
[  669s] WARNING: "/usr/src/packages/OpenSCAD-Nightly-0-Build2277.1-x86_64/usr/bin/qt.conf" already exists, will not overwrite.
[  669s] ERROR: Could not find any translations in  "/usr/share/qt4/translations"  (developer build?)
```

thehans, 2018-07-14:
I've run into issues of mixed up Qt4/Qt5 on Linux Mint in the past, and was able to clear them up with `export QT_SELECT=qt5` before the build.

I don't really know what the source script for AppImage builds looks like or where exactly it resides, but maybe setting this at the top would help?

I also had to set this in `.travis.yml` to force ubuntu test builds to use Qt5:
https://github.com/openscad/openscad/blob/master/.travis.yml#L50

t-paul, 2018-07-14:
Looks like `-qmake=` did the trick, [`OpenSCAD-Nightly-0-Build2279.1.glibc2.14-x86_64.AppImage`](https://build.opensuse.org/package/binaries/home:t-paul/OpenSCAD/AppImage) works for me on Debian/Testing again.

OBS seems to have some trouble currently, as this build is not yet published to the [download repo](https://download.opensuse.org/repositories/home:/t-paul/AppImage) which still shows build 2277.1.

t-paul, 2018-07-14:
Ahh, OBS finally synced [the latest build](https://download.opensuse.org/repositories/home:/t-paul/AppImage/OpenSCAD-Nightly-latest-x86_64.AppImage) to the download repo, @probonopd and @linupi would you give it a try.

linupi, 2018-07-15:
For me it is now working on both Debian 8 and Ubuntu xenial.  Thank you very much!

probonopd, 2018-07-15:
I can confirm the latest nightly is running for me on xubuntu-18.04-desktop-amd64.iso .

t-paul, 2018-07-15:
Great! Thanks @probonopd for tracking down the root cause.

closed 2018-07-15 by t-paul