openscad/openscad #2434: Description and tab title translation selection in Customizer

PRouzeau, 2018-07-16:
label:Customizer

It is tedious work to translate the user interface of a program and it might be much simpler in if a langage selecting system is added in Customizer.
Translation of field description could be done simply by having multiple comments lines and a selector as first chars in the comments
so for the field description it can looks like:
```
//es: Esta es una descripción en español
//fr: Ceci est une description en français
//This is the default description – if no langage selected
name = value; //[min:max] 
```

Alternatively, the selector tag could be separated with a double colon instead of a simple one for simpler discrimination:
```
//fr::Mon commentaire
```

And for tabs titles, a comma or a semi-colon could be used to separate the different descriptions
```
/*[Tab title; fr:Titre de l’onglet; es:Título de la pestaña ]*/
```
And there shall be a special variable used as a langage selector e.g. $language
If there is no translation for the selected langage, the default description shall be displayed and if no description at all, the variable name (as of today behaviour).

It might be useful to propose to start customizer in ‘description only’ mode to hide all the program innards. This does not drive to problem as if there is no description, variable named is used instead.

t-paul, 2018-07-16:
Should we find some other tag to mark requests that do not fit to the thingiverse style comment markup?

MichaelAtOz, 2018-07-17:
Or (perhaps this is paraphrasing the above) do we make language handling in the source a more generic feature, not part of Customizer, but which works with it too. So messages in echo() or text() can be easily handled.
Tho, for it to work with Customizer, it has to handle comments, so would be like a pre-processor....??

MichaelPFrey, 2018-08-15:
As long as we keep the comment syntax, we need to keep thingiverse.com customizer compatibility in mind, making expanding the syntax kind of tricky.

PRouzeau, 2018-08-19:
So I followed the suggestion to use a pre-processor, which allow selecting translation for the whole program, not only Customiser.
I slightly modified the syntax from my first post, and the modified syntax may be compatible with the Thingiverse Customizer, both for original app and translated version. 
I did translate the tab title, but as they are not stored in the json files, the json files are compatible (but you have to translate the texts and variables which are localized, if any.

Input fields description 
```
//lg:Localized field description
//Field original description 
myfield=0; // [0:2:10]
```
will became:
```
//Localized field description
myfield=0; // [0:2:10]
```
For tabs (note that tabs with no contents are not displayed), the following 
```
/*lg:[Localized tab]*/ 
/*[Default tab]*/ 
```
will became:
```
/*[Localized tab]*/ 
```
and when variables use localized names, the translation shall be AFTER and commented – to not became a description-  Note the doubled colon
```
name = "Peter"; // ["Peter", "John", "Edward"]
//fr::name = "Pierre"; //["Pierre", "Jean", "Edouard"]
```
will became (original line removed and translated line uncommented):
```
name = "Pierre"; //["Pierre", "Jean", "Edouard"]
```
To be usable, you will need to define in a  [Hidden] tab something like that
```
ar_name = ["Peter", "John", "Edward"];
//lg::ar_name = ["Pierre", "Jean", "Edouard"];
```
To be used with something like that:
```
if (name==ar_name[i]) {do something}
```
Large comments to be removed are tagged with a '+' symbol  the last line shall only have two char comment closing:
```
/*lg+ Ceci est un commentaire
multi-ligne pour test
*/
```
Original english comments to be removed (if there is an equivalent in local langage)
```
/*en+ this is a multiline comment
for test
*/
```
I did a preprocessor, but needing only two languages, I made serious shortcuts, and the pre-processor did not look the language code, it only eliminates the unneeded lines (after or before) and remove the comment marker when needed. Also, it does not remove spaces and syntax shall be exact.

I just published an app with this syntax, including a libre Office file with a dedicated macro (with name and language harcoded) here:
[https://github.com/PRouzeau/Protected-crossing](url)
My Libre office macro (with said simplifications):
```
sub trans
	dim l1$, l2$, l3$, tag2$, tag1$, tag3$,tag4$, token$, filein$, fileout$, prefix$
	dim txtin as integer, txtout as integer, lrgcomment as boolean
	lrgcomment=false
	filein = "d:/Loisirs/_Dessins/Openscad/Crossing.scad"
	fileout = "d:/Loisirs/_Dessins/Openscad/Translated.scad"
	'filein = "/home/mydir/test.scad"
	'fileout = "/home/mydir/translated.scad"
	prefix="fr"
	tag1 = "//"+prefix+"::" 'prefix will remove previous line
	tag2 = "//"+prefix+":" 'prefix will remove following line
	tag3 = "/*"+prefix+":" 'prefix will remove following line
	tag4 = "language ="
	token = "ø" 'any rare character could do
	txtin = FreeFile()
	open filein for input as #txtin 
	txtout = FreeFile()
	open fileout for output as #txtout 
	do while not eof(txtin)
		line input #txtin, l3
		if Left(l2,5) = "/*en+" Then 
			lrgcomment = true	
		endif
		if lrgcomment then
			if Left(l2,2) = "*/" Then 
				lrgcomment = false	
			endif
			l2= token
		endif
		if Left(l2,6) = tag1 Then 
	 		l1=token 'remove former line
	 		l2 =Mid(l2,7)
	 	endif
	 	if Left(l2,10) = tag4 Then 'detect language selection
	 		l2 ="language = "+chr(34)+prefix+chr(34)+";"+Mid(l2,17)
	 	endif
	 	if l1<>token then
		    Print #txtout, l1
		endif    
	    l1=l2
	    l2=l3
	    if Left(l1,5) = tag2 Then
	    	l1 =  "//"&Mid(l1,6) 
	 		l2=token 'don't print empty
	 	endif
	 	if Left(l1,5) = tag3 Then
	    	l1 =  "/*"&Mid(l1,6) 
	 		l2=token 'don't print empty
	 	endif
	loop
	Print #txtout, l1
	Print #txtout, l2
	close #txtin
	close #txtout
	msgbox "OpenSCAD translation done"
end sub
```

t-paul, 2018-08-19:
I think that new features like this should be handled in the normal language. Parsing comments will need to remain for compatibility with the Thingiverse customizer, but we should not start maintaining two different languages.

PRouzeau, 2018-08-19:
I don't get your point. 
What do you want to tell ? 
That translated message shall not be in the body of the main program ? 
or that the interpretation shall not be done in a pre-processor but in Openscad itself? Sure, but I had to solve an immediate problem, hence I did what I can do.
I am confused by the term "language", it referring to the user interface language or Openscad ?

t-paul, 2018-08-19:
I mean OpenSCAD should not need any knowledge about the content in comments. There is an exception for handling compatibility with the Thingiverse customizer but that should stay focused on just that.

What is done externally, like external pre-processors or editors can't be controlled by OpenSCAD anyway, so everyone can just use what they need.

PRouzeau, 2018-08-19:
Ok, I get it, you don't want to create a 'Comment language' and prefer to have integrated instructions.
I think handling stuff in comments is not a bad solution and you free yourself from a lot of intrinsic constraints of OpenSCAD. I don't feel the proposed syntax is that complex.
Customizer is a pre-processor, so you already have it.
Before writing the pre-processing macro, I was handling the translation in my app with a language selector variable, arrays and if or ? selections. 
Handling that as comments made the stuff simpler and cleaner.  This is also related to the fact that OpenSCAD dont like the 'if' and  you may prefer using the ternary operator.
Also, there is another macro that I did not show, to remove the translated part, which can make an 'english only' version, not to be used when modifying the app, but easier to read at first for newcomers.

MethylBromide, 2025-02-11:
The Thingiverse implementation is **WRETCHED** and is not being maintained. Please do not try to maintain compatibility with it if that in _any_ way limits desirable functionality in the client. Coordinate with Makerspace, yes, because they have a respectable online implementation, but Thingiverse can go f*** themselves for all the effort they are putting into this currently.