openscad/openscad #2436: text() metrics / glyph size problem

arpruss, 2018-07-16:
The easiest way to see this problem is to run `text("This is a string",font="Times New Roman",valign="top");` and notice that top of the "h" and the dot over the "i" slightly stick out above the x-axis. This is despite the fact that `FreetypeRenderer::calc_y_offset()` aligns using the actual bounds of the glyphs, which _should_ guarantee that the letters are correctly located (modulo some minor issues if a curve sticks out past a control point--but I've checked that this isn't the cause of the issue). 

What seems to be happening is that the font metrics, including x-advances and bounding boxes, are scaled slightly differently from the scaling of the actual glyph outlines in rendering. As a result the glyphs are (in my testing with several fonts) slightly too large as compared to what the code thinks the bounding box sizes to be. This would both throw off the `valign="top"` rendering and make the x-advances slightly smaller than they should be.

I checked this hypothesis by drawing the same string in Inkscape on top of the  OpenSCAD rendering and scaling it so that the bottoms and left and right edges of the text were aligned between Inkscape and OpenSCAD:

[image]

In the image, you can see that the top edge of the yellow OpenSCAD render sticks out past the top edge of the red Inkscape render of the text. Moreover, the Inkscape text when scaled horizontally to be the same width as the OpenSCAD text and to align with the bottom of the OpenSCAD text nicely fits just below the x-axis, just as the OpenSCAD text should (as I put `valign="top"`).

I also examined the actual metrics and outlines inside the TTF files to verify that OpenSCAD was calculating things wrong.

Empirically, it looks like OpenSCAD consistently scales metrics (e.g., x-advances) to make the em-size be equal to `2048/1510*size` (I verified this with several fonts), but scales glyphs a little bit bigger, with different fonts resulting in somewhat different scaling of the glyphs (e.g., Times New Roman glyphs are a little larger relative to the metrics than in the case of Liberation Sans; Liberation Sans is only slightly too big).

I have not been able to figure out what exactly the cause of this is in the OpenSCAD code. Maybe there is some sort of a slight mismatch between the FreeType2 rendering and the HarfBuzz placements?

I noticed the problems with the metrics and glyph sizes when trying to write OpenSCAD code to calculate text bounding boxes using metric data extracted from ttf files, and finding that when I made the x-advances correct, ascenders, descenders and side-bearings would stick out of the bounding boxes I calculated. (See https://www.thingiverse.com/thing:3004457 for the code.)

LMLB, 2020-07-30:
I found that the font size is scaled by a factor of `100/72`, and that the advance widths are scaled by a factor of `1000/1024` relative to the scaled font size.

I believe the `100/72` factor comes from this line (changing `100, 100` to `72, 72` would fix it):
https://github.com/openscad/openscad/blob/4dcd5a9a46fc89b11e6616c6a1324d4050b9ff6c/src/FreetypeRenderer.cc#L215

I believe the `1000/1024` factor comes from this line (changing `1.024e5` it to `1e5` would fix it):
https://github.com/openscad/openscad/blob/4dcd5a9a46fc89b11e6616c6a1324d4050b9ff6c/src/FreetypeRenderer.cc#L52

arpruss, 2020-07-30:
I now wonder if this should be fixed. We have many, many OpenSCAD files out in the wild, and if we tweak the font sizes, they will render differently. But perhaps the difference is so slight that that's OK.

jordanbrown0, 2023-11-06:
See [this thread](https://lists.openscad.org/empathy/thread/DMTDCROXR4JPAO24I3226KSSUERF4VDP) for a discussion of the 2.4% spacing problem.  I fixed it in PR#3684.

See also #4304 for how OpenSCAD text is ~30% larger than it "should" be based on conventional font measurements.

kintel, 2023-11-07:
Closing as #3684 addressed this. We can continue in #4304 for scaling issues.

closed 2023-11-07 by kintel