openscad/openscad #2437: Feature Request: Voronoi Generator

Sislar, 2018-07-19:
New here but not new to programming.  Been enjoy openscad for a while now and made a few things But I'm stumped so far trying to make a voronoi pattern.  Been looking into it for a few weeks and I think i have found a the algorithms but they are fairly complex. Would it be possible to write this in openscad script or would it need to be a new feature/primitive in Openscad?

What i was thinking was the function or primitive would be for 2d at least to start applying it to 
an arbitrary surface... well I'm not that advanced yet. 
voronoi( P = set of points, r = Rounding, t = thickness)

First step is to convert the set of points to a set of non overlapping triangles via
https://en.wikipedia.org/wiki/Delaunay_triangulation

Then you find the circumcenter of each triangle. (This is fairly easy and I have openscad script for this)

Then you need to create the pologons around each point using the right set of circumcenters. I haven't thought through this point yet.

from there is easy,  you can just offset (-r) and minoski with the rounding parameter.

Sorry if this isn't the right place to post.  Where should i go from here, is there someone that ca comment if this is possible in the existing openscad and if so willing to help code it. Or will it need
a new primitive (and I'm sure my nomenclature is off here as well),

MichaelAtOz, 2018-07-19:
> Sorry if this isn't the right place to post. Where should i go from here

You should generally raise discussion topics on [the forum ](http://forum.openscad.org/)first.
When you think you have a bug report or a semi-firm change request you raise them here.

Also [see this.](http://lmgtfy.com/?q=voronoi+openscad) There are some solutions.

tkphd, 2018-08-01:
@Sislar, if you're referring to @felipesanches' [OpenSCAD_Voronoi_Generator](https://github.com/felipesanches/OpenSCAD_Voronoi_Generator), you're wrong: it places nuclei at random within rotated squares, then performs a valid Poisson-Voronoi tessellation of the 2D surface from those nuclei. Give it a shot.

stefano2734, 2026-09-20:
Label Meshing?