openscad/openscad #2439: Customizer: Out of place tab if double allocation of a variable of this group

PRouzeau, 2018-08-03:
label [Customizer]
Bug:
If a variable is allocated a first time, then the same variable is allocated a second time within a customizer group, the group tab is moved to the position of the first allocation, whatever its original position is supposed to be. In principle, it is not normal to allocate a variable twice when using customizer, but it can be difficult to track it without a message. This bug may not need to be removed, provided a message about the double allocation is sent.

MichaelPFrey, 2018-08-25:
Code showing the issue:
```
a=5;
/* [Group 1] */
a=10;
b=10;
/* [Group 2] */
a=20;
```
screenshot:
[image]
hmmm...

t-paul, 2018-08-25:
Reassingment should ideally be an error. Right now it's not even a warning. It's not as easy as it sounds as there are some usecases that historically mis-use that feature (e.g. the -D command line parameter).

This also seems to be a duplicate of #1770.

MichaelPFrey, 2018-08-25:
every assignment knows its location:
`const Location loc =assignment.location();`
which includes `firstLine()`. `ParameterExtractor::setParameters` gets the assignments in order of first assignment, but with the information of the last assignment. We could reorder the assignments by location. That would at least fix the "out of place" placement of the group. std::sort with a custom function could be used for that.

@t-paul good catch - I can see it in two ways: Fixing #1770 also solves #2439. But #2439 could be solved/worked around without resolving #1770, as I see some issues with the customizer code as well.
Well, I start working on #1770. I see away to deal with include and the -D command line parameter.

t-paul, 2018-08-25:
Right, good point. Let's leave this open then as separate topic to sync data and location of the assignments.

MichaelPFrey, 2018-08-26:
Testing my patch, I found `value-reassignment-tests.scad`. This test case has a similar root cause to this issue. But thinking about it, it shows how complicated rearranging the statements would get.

closed 2026-03-09 by github-actions[bot]