openscad/openscad #2441: Cannot build windows version due to bzip2 missing

stepman0, 2018-08-08:
Trying to cross-build windows version on Debian machine. 

When running `./scripts/mingw-x-build-dependencies.sh 64` it fails due to bzip2 and protobuf cannot be downloaded:

`Download failed or wrong checksum of package bzip2!
------------------------------------------------------------
HTTP-Anforderung gesendet, auf Antwort wird gewartet … 200 OK
Länge: 271 [text/html]
Wird in »»/home/user/openscad_deps/mxe/pkg/.tmp-bzip2-1.0.6.tar.gz«« gespeichert.
     0K                                                       100% 62,6M=0s
2018-08-08 17:10:02 (62,6 MB/s) - »»/home/user/openscad_deps/mxe/pkg/.tmp-bzip2-1.0.6.tar.gz«« gespeichert [271/271]
+ cat /home/user/openscad_deps/mxe/pkg/.tmp-bzip2-1.0.6.tar.gz
+ rm /home/user/openscad_deps/mxe/pkg/.tmp-bzip2-1.0.6.tar.gz
------------------------------------------------------------
[log]      /home/user/openscad_deps/mxe/log/bzip2-download

Makefile:533: die Regel für Ziel „download-only-bzip2-1.0.6.tar.gz“ scheiterte
make: *** [download-only-bzip2-1.0.6.tar.gz] Fehler 1
make: *** Es wird auf noch nicht beendete Prozesse gewartet...
[done]      cmake-conf             x86_64-pc-linux-gnu                              12 KiB         0m4,401s
[done]      mxe-conf               x86_64-pc-linux-gnu                              328 KiB        0m6,293s

Download failed or wrong checksum of package protobuf!
------------------------------------------------------------
  4000K .......... .......... .......... .......... .......... 95%  293K 1s
  4050K .......... .......... .......... .......... .......... 96%  318K 1s
  4100K .......... .......... .......... .......... .......... 97%  328K 0s
  4150K .......... .......... .......... .......... .......... 99%  329K 0s
  4200K .......... .......... .......... .....                100%  346K=24s
2018-08-08 17:10:27 (177 KB/s) - »»/home/user/openscad_deps/mxe/pkg/.tmp-protobuf-3.3.2.tar.gz«« gespeichert [4337495/4337495]
+ cat /home/user/openscad_deps/mxe/pkg/.tmp-protobuf-3.3.2.tar.gz
+ rm /home/user/openscad_deps/mxe/pkg/.tmp-protobuf-3.3.2.tar.gz
------------------------------------------------------------
[log]      /home/user/openscad_deps/mxe/log/protobuf-download

Makefile:533: die Regel für Ziel „download-only-protobuf-3.3.2.tar.gz“ scheiterte
make: *** [download-only-protobuf-3.3.2.tar.gz] Fehler 1
`

Logfile /home/user/openscad_deps/mxe/log/bzip2-download shows:
`+ mkdir -p /home/user/openscad_deps/mxe/pkg
+ wget --user-agent=Wget/1.18 -T 30 -t 3 -O /home/user/openscad_deps/mxe/pkg/.tmp-bzip2-1.0.6.tar.gz http://www.bzip.org/1.0.6/bzip2-1.0.6.tar.gz
--2018-08-08 17:32:11--  http://www.bzip.org/1.0.6/bzip2-1.0.6.tar.gz
Auflösen des Hostnamens »www.bzip.org (www.bzip.org)« … 208.91.197.44
Verbindungsaufbau zu www.bzip.org (www.bzip.org)|208.91.197.44|:80 … verbunden.
HTTP-Anforderung gesendet, auf Antwort wird gewartet … 200 OK
Länge: 272 [text/html]
Wird in »»/home/user/openscad_deps/mxe/pkg/.tmp-bzip2-1.0.6.tar.gz«« gespeichert.

     0K                                                       100% 54,8M=0s

2018-08-08 17:32:11 (54,8 MB/s) - »»/home/user/openscad_deps/mxe/pkg/.tmp-bzip2-1.0.6.tar.gz«« gespeichert [272/272]

+ cat /home/user/openscad_deps/mxe/pkg/.tmp-bzip2-1.0.6.tar.gz
+ rm /home/user/openscad_deps/mxe/pkg/.tmp-bzip2-1.0.6.tar.gz
`
It seems that www.bzip.org is not valid anymore.

donbright, 2018-08-12:
this is because the build mingw dependency script is using a 'frozen' version of MXE that is a little bit old, 

if you go inside the mingw-x-build-dependencies.sh script and change github.com/openscad.mxe to github.com/mxe/mxe then also comment out the part that does checkout of openscad-snapshot branch, then it should work. 

```
if [ ! -e $MXEDIR ]; then
        mkdir -p $MXEDIR
        cd $MXEDIR/..
        echo "Downloading MXE into " $PWD
#       git clone git://github.com/openscad/mxe.git $MXEDIR
        git clone https://github.com/mxe/mxe.git
fi

echo "entering" $MXEDIR
cd $MXEDIR
#echo 'checkout openscad-snapshot-build branch'
#git checkout openscad-snapshot-build

```

@t-paul is there an easy way to upgrade the openscad mxe fork or ...

t-paul, 2018-08-12:
Currently rebuilding on romilly...

donbright, 2018-08-12:
it loks like MXE master is pretty broken... its still using very old gcc 5.5 which has internal compiler errors segfaulting on some libraries

but if you try to upgrade gcc, well, gcc 6, and 7, has this bad interaction between -isystem and 'include_next' which broke a bunch of stuff, including cross compilers

https://gcc.gnu.org/bugzilla/show_bug.cgi?id=70129

https://gitlab.kitware.com/cmake/cmake/issues/16291

https://bugs.webkit.org/show_bug.cgi?id=161697

donbright, 2018-08-12:
yes for now we can probably do a workaround..... but also.... 

we cannot depend on staying on old gcc 5 forever...and mxe cant really either. .. .  what i think we might be able to do is to upgrade mxe to gcc 7.3.0 but then also somehow patch mxe's CGAL and/or mxe's cmake so it stops inserting '-isystem' into its cmake builds. ... i havent quite figured out which one is the better one to change.

t-paul, 2018-08-12:
Build did work so far and looked ok in `wine`. I can't do a full test though as I don't have a Windows machine at hand.

http://files.openscad.org/snapshots/OpenSCAD-2018.08.12-x86-32-Installer.exe
http://files.openscad.org/snapshots/OpenSCAD-2018.08.12-x86-32.zip
http://files.openscad.org/snapshots/OpenSCAD-2018.08.12-x86-64-Installer.exe
http://files.openscad.org/snapshots/OpenSCAD-2018.08.12-x86-64.zip

```
OpenSCAD Version: 2018.08.12 (git bbc99b8)
System information: Unknown Windows(TM) 8 CPUs 15.39 GB RAM
Compiler: GCC "4.9.4" 64bit
Boost version: 1_60
Eigen version: 3.2.5
CGAL version, kernels: 4.6.3, Cartesian, Extended_cartesian, Epeck
OpenCSG version: OpenCSG 1.4.2
Qt version: 5.7.0
QScintilla version: 2.8.4
MingW build: MingW64
GLib version: 2.44.1
```

t-paul, 2018-08-12:
There's some [WIP pull request](https://github.com/mxe/mxe/pull/1860) for gcc6, that looks like it might be related to  the `-isystem` issue...

MichaelAtOz, 2018-08-13:
>  have a Windows machine at hand.

I'll give it a spin, anything in particular I should look for, or just that it seems to run?

MichaelAtOz, 2018-08-13:
Well AVG took this 'very unique file' away for 73 minutes while they look at it...

t-paul, 2018-08-13:
Yeah, nothing special I could point to, it's just updated dependencies. So a general "works as normal" check would be great.

donbright, 2018-08-13:
that is excellent news.... is the gcc version "4.9.4" because it uses old MXE? the thing i think about is, this. gcc 5 -> segfaults during build. gcc 6,7,8-> all have next_include conflict with -isystem which is used by CGAL, so cross-build breaks. we might be OK for a few years but....

t-paul, 2018-08-13:
Restarted build, this time it really should be master. This seems to be using gcc-5.5, lets see if that works...

MichaelAtOz, 2018-08-13:
That version, after I disentangled AVG, run, previews & renders normally.

Library info.
```
OpenSCAD Version: 2018.08.12 (git bbc99b8)
System information: Windows(TM) 6.1 SP 1.0 NTW 1 MSDN 724833 2 CPUs 3.94 GB RAM
Compiler: GCC "4.9.4" 64bit
Boost version: 1_60
Eigen version: 3.2.5
CGAL version, kernels: 4.6.3, Cartesian, Extended_cartesian, Epeck
OpenCSG version: OpenCSG 1.4.2
Qt version: 5.7.0
QScintilla version: 2.8.4
MingW build: MingW64
GLib version: 2.44.1
libzip version: 
Application Path: C:/Program Files/openscad-2018.08.12
Documents Path: C:\Users\Testing\Documents
Resource Path: C:/Program Files/openscad-2018.08.12
User Library Path: C:/Users/Testing/Documents/OpenSCAD/libraries
User Config Path: C:\Users\Testing\AppData\Local/OpenSCAD
Backup Path: C:/Users/Testing/Documents/OpenSCAD/backups
OPENSCADPATH: 
OpenSCAD library path:
C:/Users/Testing/Documents/OpenSCAD/libraries
C:/Program Files/openscad-2018.08.12\libraries

OPENSCAD_FONT_PATH: 
OpenSCAD font path:
GLEW version: 1.12.0
OpenGL Version: 3.3.0
GL Renderer: ION/integrated/SSE2
GL Vendor: NVIDIA Corporation
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes
```

tonytheodore, 2018-08-13:
> There's some WIP pull request for gcc6, that looks like it might be related to the -isystem issue...

I wasn't happy with that approach, but have cherry-picked the `cgal` fix into mxe master so all the [deps](https://github.com/openscad/openscad/blob/master/scripts/mingw-x-build-dependencies.sh#L87) build with gcc7 (and gcc5 - I haven't seen the internal compiler errors).

To test, you can add `MXE_PLUGIN_DIRS=plugins/gcc7` to your [`make` invocation](https://github.com/openscad/openscad/blob/master/scripts/mingw-x-build-dependencies.sh#L98).

The bzip2 issue is also fixed since Nov 17, if you need to run an older version of mxe, you can try adding `bzip2_URL=foo.org` to your `make` invocation - that will force an error (instead of the `200 OK`) and download from one of our backup mirrors.

t-paul, 2018-08-14:
Hmm, I'm still getting gcc-5.5 when just adding the plugin variable

```
make MXE_PLUGIN_DIRS=plugins/gcc7 qtbase qtmultimedia qscintilla2 mpfr eigen opencsg cgal glib libxml2 freetype fontconfig harfbuzz libzip MXE_TARGETS=x86_64-w64-mingw32.static.posix -j 2 JOBS=2
```

```
$ ls -1 /opt/mxe/usr/bin/*-gcc
/opt/mxe/usr/bin/i686-w64-mingw32.static.posix-gcc
/opt/mxe/usr/bin/x86_64-w64-mingw32.static.posix-gcc

$ /opt/mxe/usr/bin/x86_64-w64-mingw32.static.posix-gcc -v
Using built-in specs.
COLLECT_GCC=/opt/mxe/usr/bin/x86_64-w64-mingw32.static.posix-gcc
COLLECT_LTO_WRAPPER=/opt/mxe/usr/libexec/gcc/x86_64-w64-mingw32.static.posix/5.5.0/lto-wrapper
Target: x86_64-w64-mingw32.static.posix
Configured with: /opt/mxe/tmp-gcc-x86_64-w64-mingw32.static.posix/gcc-5.5.0/configure ac_cv_prog_HAVE_DOXYGEN=false --enable-doc=no --enable-gtk-doc=no --enable-gtk-doc-html=no --enable-gtk-doc-pdf=no --docdir=/opt/mxe/tmp-gcc-x86_64-w64-mingw32.static.posix/gcc-5.5.0.build_.sink --infodir=/opt/mxe/tmp-gcc-x86_64-w64-mingw32.static.posix/gcc-5.5.0.build_.sink --mandir=/opt/mxe/tmp-gcc-x86_64-w64-mingw32.static.posix/gcc-5.5.0.build_.sink --with-html-dir=/opt/mxe/tmp-gcc-x86_64-w64-mingw32.static.posix/gcc-5.5.0.build_.sink --disable-doxygen --target=x86_64-w64-mingw32.static.posix --build=x86_64-pc-linux-gnu --prefix=/opt/mxe/usr --libdir=/opt/mxe/usr/lib --with-sysroot=/opt/mxe/usr/x86_64-w64-mingw32.static.posix --enable-languages=c,c++,objc,fortran --enable-version-specific-runtime-libs --with-gcc --with-gnu-ld --with-gnu-as --disable-nls --disable-shared --disable-multilib --without-x --disable-win32-registry --enable-threads=posix --enable-libgomp --with-gmp=/opt/mxe/usr/x86_64-pc-linux-gnu --with-isl=/opt/mxe/usr/x86_64-pc-linux-gnu --with-mpc=/opt/mxe/usr/x86_64-pc-linux-gnu --with-mpfr=/opt/mxe/usr/x86_64-pc-linux-gnu --with-as=/opt/mxe/usr/bin/x86_64-w64-mingw32.static.posix-as --with-ld=/opt/mxe/usr/bin/x86_64-w64-mingw32.static.posix-ld --with-nm=/opt/mxe/usr/bin/x86_64-w64-mingw32.static.posix-nm
Thread model: posix
gcc version 5.5.0 (GCC) 

```

And with that the library build works but OpenSCAD compilation crashes like @donbright mentioned earlier.

@tonytheodore is there any additional setting needed to get gcc7 built and used?

```
i686-w64-mingw32.static.posix-g++ -c -pipe -fno-keep-inline-dllexport -DSTACKSIZE=8388608 -DNOGDI -fno-strict-aliasing -frounding-math -DEIGEN_DONT_ALIGN -I/opt/mxe/usr/i686-w64-mingw32.static.posix/include
/glib-2.0 -I/opt/mxe/usr/i686-w64-mingw32.static.posix/lib/glib-2.0/include -I/opt/mxe/usr/i686-w64-mingw32.static.posix/include/libxml2 -I/opt/mxe/usr/i686-w64-mingw32.static.posix/include/harfbuzz -I/opt/
mxe/usr/i686-w64-mingw32.static.posix/include/freetype2 -I/opt/mxe/usr/i686-w64-mingw32.static.posix/include/fontconfig -std=c++0x -fpermissive -O2 -std=gnu++11 -Wall -W -Wextra -Wno-unused-local-typedefs -
Wno-unused-local-typedefs -fexceptions -mthreads -DUNICODE -D_UNICODE -DWIN32 -DOPENSCAD_VERSION=2018.08.14 -DOPENSCAD_SHORTVERSION=2018.08.14 -DOPENSCAD_YEAR=2018.0 -DOPENSCAD_MONTH=08.0 -DOPENSCAD_DAY=14.
0 -DOPENSCAD_COMMIT=bbc99b8 -DOPENSCAD_DEPLOY -DOPENSCAD_SNAPSHOT -DSTACKSIZE=8388608 -DENABLE_EXPERIMENTAL -DENABLE_CGAL -DENABLE_OPENCSG -DGLEW_STATIC -DBOOST_THREAD_USE_LIB -DBOOST_STATIC -DBoost_USE_STA
TIC_LIBS -DLIBXML_STATIC -DUSE_SCINTILLA_EDITOR -DQT_NO_DEBUG -DQT_OPENGL_LIB -DQT_PRINTSUPPORT_LIB -DQT_WIDGETS_LIB -DQT_MULTIMEDIA_LIB -DQT_GUI_LIB -DQT_CONCURRENT_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -DQT_
NEEDS_QMAIN -I../../openscad-branch -I. -I../src -I/opt/mxe/usr/i686-w64-mingw32.static.posix/qt5/include -I../src/ext/libtess2/Include -I/opt/mxe/usr/i686-w64-mingw32.static.posix/qt5/include/QtOpenGL -I/o
pt/mxe/usr/i686-w64-mingw32.static.posix/qt5/include/QtPrintSupport -I/opt/mxe/usr/i686-w64-mingw32.static.posix/qt5/include/QtWidgets -I/opt/mxe/usr/i686-w64-mingw32.static.posix/qt5/include/QtMultimedia -
I/opt/mxe/usr/i686-w64-mingw32.static.posix/qt5/include/QtGui -I/opt/mxe/usr/i686-w64-mingw32.static.posix/qt5/include/QtConcurrent -I/opt/mxe/usr/i686-w64-mingw32.static.posix/qt5/include/QtNetwork -I/opt/
mxe/usr/i686-w64-mingw32.static.posix/qt5/include/QtCore -Iobjects -isystem /opt/mxe/usr/i686-w64-mingw32.static.posix/include/eigen3 -isystem /opt/mxe/usr/i686-w64-mingw32.static.posix/include -I/opt/mxe/u
sr/i686-w64-mingw32.static.posix/qt5/mkspecs/win32-g++  -o objects/CSGTreeEvaluator.o ../src/CSGTreeEvaluator.cc
In file included from /opt/mxe/usr/i686-w64-mingw32.static.posix/include/boost/detail/sp_typeinfo.hpp:20:0,
                 from /opt/mxe/usr/i686-w64-mingw32.static.posix/include/boost/smart_ptr/detail/sp_counted_base_gcc_x86.hpp:27,
                 from /opt/mxe/usr/i686-w64-mingw32.static.posix/include/boost/smart_ptr/detail/sp_counted_base.hpp:51,
                 from /opt/mxe/usr/i686-w64-mingw32.static.posix/include/boost/smart_ptr/detail/shared_count.hpp:29,
                 from /opt/mxe/usr/i686-w64-mingw32.static.posix/include/boost/smart_ptr/shared_ptr.hpp:28,
                 from /opt/mxe/usr/i686-w64-mingw32.static.posix/include/boost/shared_ptr.hpp:17,
                 from /opt/mxe/usr/i686-w64-mingw32.static.posix/include/boost/format/alt_sstream.hpp:21,
                 from /opt/mxe/usr/i686-w64-mingw32.static.posix/include/boost/format/internals.hpp:23,
                 from /opt/mxe/usr/i686-w64-mingw32.static.posix/include/boost/format.hpp:38,
                 from ../src/printutils.h:6,
                 from ../src/CSGTreeNormalizer.cc:3:
/opt/mxe/usr/i686-w64-mingw32.static.posix/include/boost/smart_ptr/detail/sp_counted_impl.hpp: In member function 'virtual void* boost::detail::sp_counted_impl_pd<P, D>::get_deleter(const sp_typeinfo&)':
/opt/mxe/usr/i686-w64-mingw32.static.posix/include/boost/smart_ptr/detail/sp_counted_impl.hpp:158:22: internal compiler error: in typeid_ok_p, at cp/rtti.c:319
         return ti == BOOST_SP_TYPEID(D)? &reinterpret_cast<char&>( del ): 0;
```

tonytheodore, 2018-08-14:
> is there any additional setting needed to get gcc7 built and used?

You probably need to start with a `make clean` - the script looks like it always uses a fresh clone and it's usually better to build the toolchain in a single pass. You could trigger the build with `touch src/gcc.mk` but then you'd likely get conflicts with some of the native deps like `isl`.

tonytheodore, 2018-08-15:
So starting with `make clean` myself, these changes:
```diff
diff --git a/qscintilla2.prf b/qscintilla2.prf
index 0326430d..535eabed 100644
--- a/qscintilla2.prf
+++ b/qscintilla2.prf
@@ -10,7 +10,11 @@ INCLUDEPATH += $$[QT_INSTALL_HEADERS]

 LIBS += -L$$[QT_INSTALL_LIBS]

-QT5LIB=qt5scintilla2
+CONFIG(mingw-cross-env) {
+    QT5LIB=qscintilla2_qt5
+} else {
+    QT5LIB=qt5scintilla2
+}

 unix:linux* {
     FEDORA32LIBS=/usr/lib/libqscintilla2-qt5.so
@@ -40,7 +44,7 @@ CONFIG(debug, debug|release) {
     mac: {
        LIBS += -lqscintilla2
     } else {
-      win32: {
+      win32!CONFIG(mingw-cross-env): {
         LIBS += -lqscintilla2
       } else {
         greaterThan(QT_MAJOR_VERSION, 4) {
```
and this build command:
```sh
export mxe_dir=/Users/tonyt/dev/mxe
export mxe_target=i686-w64-mingw32.static.posix.gcc7
export openscad_src=$mxe_dir/gits/openscad
export openscad_build=$mxe_dir/gits/openscad.build
rm -rf $openscad_build && \
mkdir -p $openscad_build && \
cd $openscad_build && \
ln -s $mxe_dir/usr/$mxe_target mingw-cross-env && \
OPENSCAD_LIBRARIES=$mxe_dir/usr/$mxe_target \
$mxe_target-qmake-qt5 $openscad_src \
  VERSION=2018.08.15 \
  CONFIG+='mingw-cross-env link_pkgconfig' \
  PKGCONFIG+=libzip
```
it succeeds with a `$openscad_build/release/openscad.exe` that opens and runs. I used pkg-config for `libzip` since it installs headers to weird locations that aren't in `libzip.pri`.

tonytheodore, 2018-08-15:
Actually, my [error above](https://github.com/openscad/openscad/issues/2441#issuecomment-412950105) only happens when `libzip` is enabled - setting `PKGCONFIG+=libzip` doesn't set `DEFINES += ENABLE_LIBZIP` so the build succeeds.

t-paul, 2018-08-15:
Awesome, thanks. I'll try to get that integrated into the scripts (slowly due to work being a bit crazy lately). We seem to have something in the scripts that forces to revert back to default gcc.

I guess the `.gcc7` suffix is one of the important tips, I was not aware of that. The env build with gcc7 plugin seems to have built fine over night, it did produce only `x86_64-w64-mingw32.static.posix` as output folder. Should that also have the `.gcc7` suffix or is that only needed for the application build?

tonytheodore, 2018-08-15:
Updating mxe to the [latest libzip](https://github.com/mxe/mxe/commit/8a8b14070c981715d1d55b2485505413cb79b31f) fixes it.

tonytheodore, 2018-08-15:
The `.gcc7` suffix is just a naming convention I use, it could be anything like `.my.test.gcc`. It also allows me to have side-by-side versions to test with.

tonytheodore, 2018-08-15:
> The .gcc7 suffix is just a naming convention

Thinking about it some more, I'll make this functional so there's no need to additionally specify `MXE_PLUGIN_DIRS` - makes it much more obvious what's going on.

I've done a PR #2447 to remove hard-coded entires in *.prl files and the need to set OPENSCAD_LIBRARIES etc.

MichaelAtOz, 2018-08-30:
Latest 64 bit exe works.

```
OpenSCAD Version: 2018.08.18 (git 5169fb1)
System information: Windows(TM) 6.1 SP 1.0 NTW 1 MSDN 724833 6 CPUs 8.00 GB RAM
Compiler: GCC "7.3.0" 64bit
Boost version: 1_60
Eigen version: 3.2.5
CGAL version, kernels: 4.11, Cartesian, Extended_cartesian, Epeck
OpenCSG version: OpenCSG 1.4.2
Qt version: 5.11.1
QScintilla version: 2.10.4
MingW build: MingW64
GLib version: 2.50.2
libzip version: 
...

OPENSCAD_FONT_PATH: 
OpenSCAD font path:
GLEW version: 1.12.0
OpenGL Version: 1.1.0   <- RDP
GL Renderer: GDI Generic
GL Vendor: Microsoft Corporation
RGBA(8888), depth(16), stencil(8)
GL_ARB_framebuffer_object: no
GL_EXT_framebuffer_object: no
GL_EXT_packed_depth_stencil: no

Qt graphics widget: QGLWidgetGL Extensions:GL_EXT_bgra
GL_EXT_paletted_texture
GL_WIN_swap_hint
```

closed 2018-11-09 by t-paul