openscad/openscad #2442: qt5 causes certain windows machines to crash on start

donbright, 2018-08-12:
on machines with certain Intel graphics chips, a file called **ig4icd32.dll** causes OpenSCAD to immediately crash directly after clicking "new" or trying to load a file, on Windows

see also

https://bugreports.qt.io/browse/QTBUG-40991

A workaround is as follows, to have Qt use its software renderer, 'angle'

```
    c:\program files\openscad
    set QT_OPENGL=angle 
    openscad.exe
```
windows crash report files:

[WEREAED.tmp.txt](https://github.com/openscad/openscad/files/2280731/WEREAED.tmp.txt)
[WERI8880.tmp.txt](https://github.com/openscad/openscad/files/2280732/WERI8880.tmp.txt)

Note that this does not affect openscad on the command line - rendering to a PNG from command line works fine. This issue is an interaction between certain versions of Qt5 and the graphics drivers on some windows machines

donbright, 2022-12-02:
closing, qt5 is nearing end of life if not already there

closed 2022-12-02 by donbright

MichaelAtOz, 2022-12-02:
It is in the current OpenSCAD release 2021.01. (Comment for posterity...)