openscad/openscad #2444: Customizer automatic calculation while typing and fields losing focus

PRouzeau, 2018-08-14:
Tag: Customizer
Recent versions of customizer run automatic recalculations as long as you type. 
Why not? 
- First problem is after a recalculation, the field you are typing in lose the focus, so you further typing goes nowhere. So you have to type as fast as possible to try to beat the recalculation process. 
- Second problem is that as you haven't finished to type your new data, the intermediate recalculated result is unpredictable and may crash.
If you have some text, this is unmanageable, so I open an editor, type in the editor and copy/paste. Copy/paste may also be a faster process for numbers. 
Another side effect is the the mouse wheel operation for spinboxes no longer works. Considering the very poor ergonomy of QT spinboxes (too small targets), this is real bad.
So, either the focus came back automatically to the field during and after a recalculation, or you simply wait the [enter] key as before, which is generally a much better solution when handling numbers.

MichaelPFrey, 2018-08-15:
The slider feels mostly right. When the user is using the scroll wheel, it would be nice if the slider would keep focus after a new preview has been generated.

The spin boxes feel okay to some extend, but are an issue in some cases. One case being changing it for e.g. from 200 to 120 while typing slowly. editingFinished() once again to the rescue.

One of the problem is, that we rebuild the UI, when updating it would be enough.
(This is partially the cause for lost focus)

An other side note is, that "automatic preview"  can be disabled.

t-paul, 2019-11-18:
Closed via  #2446.

closed 2019-11-18 by t-paul