openscad/openscad #2450: Customizer: Creation of a group [Reserved] for recorded data not shown to user

PRouzeau, 2018-08-17:
Tag: [Customizer] [New Feature]
Variables under the [Hidden] group are no longer recorded, which is fine.
However, you may want to record some variables without showing them to the user.
These special groups which may contains data for future development, data record version, language, unit coefficient,etc. and may be named either (or both)  [Reserved] or [Future]

t-paul, 2018-08-17:
Why would that need any special feature?

MichaelPFrey, 2018-08-18:
You mean recorded as in being stored in the .json file?

My concern is, that stored hidden variables can cause a confusing state that can not be seen nor be fixed within OpenSCAD.
Coming from that, I would only see a point in "read only variables", that are shown in the Customizer UI, stored in the JSON File, but can not be changed in the Customizer UI. But I do not know what "read only variable" is supposed to mean, as in: Does the SCAD File take priority or the JSON File? What if the user "needs to change" the read-only variable for some reason?

The question is still: What is the use case? I know that the use cases evolve once a feature exists, but having a specific use case in mind while developing makes the feature more consistent.

Keeping track which JSON File Version belongs to which SCAD File Version is best done by using a Versioning System (Git/SVN/Mercurial/...).

If someone shares a json file without the scad file: Well that is an issue and knowing the Version of the SCAD file that created the json File would help. But always sending scad & json file would be easier.

If someone upgrades the scad file, but uses an old json file:  Knowing the Version of the SCAD file that created the json File could allow the scad file to use some logic - but that gets messy quickly.
New features being off by default would be easier.

Last but not least, we need to keep the thingiverse customizer in mind.

PRouzeau, 2018-08-30:
Use case is for version info and other parameters which shall not yet be modified by users.
Example: On my last app, I created the tab named "_reserved-do not modify" (need to be the first due to the parameters involved) 
which contains:
*version number 
*language digram
*unit coefficient
*another unit coefficient
[image]

t-paul, 2018-08-30:
Why have them in customizer then?

PRouzeau, 2018-08-30:
To be recorded in the dataset.
In fact, this is a provision for future evolution

t-paul, 2018-08-30:
I don't know what that means. Non-editable values are in the *.scad file, why would those need to be somewhere else too?

MichaelAtOz, 2018-08-30:
From a holistic view I would expect variables in [hidden] to be in the json file, just like other 'tabs'.
I see @MichaelPFrey's point on versioning, but just because someone can misuse a feature, doesn't make it bad.
Being able to distribute a json file with custom hidden variables, without the .scad file changing is a worthwhile capability IMHO.

MichaelAtOz, 2018-08-30:
BTW @PRouzeau from the small glimpse above, that looks like an impressive project.

MichaelPFrey, 2018-09-02:
Quoting MichaelAtOz:
> I see @MichaelPFrey's point on versioning, but just because someone can misuse a feature, doesn't make it bad.

I do not mind when people _intentionally_ exploit a feature - that is fine by me.

My issue is, that stored hidden variables can create **accidental** problems, that require opening the json file just to diagnose. For e.g. someone is developing a SCAD file, makes some preset, changes a hidden variable (e.g. factor from 1 to 25.4) in the SCAD file, make some more preset and is latter wondering, why the factor is overwritten by some presets.

Quoting MichaelAtOz:

> Being able to distribute a json file with custom hidden variables, without the .scad file changing is a worthwhile capability IMHO.

I think I now start to see where you are going and how we can compromise.
Would "hidden variables get stored but are not applied by the customizer" be okay?
The newer version of the SCAD Model can always add a formerly hidden variable to a group, making it visible and thus applicable. A person can always manually review the json file to figure out which version of the model has created that json file.

Quoting PRouzeau:

> version number

There are two possible use cases for a version number, both requiring a different approach:
- The Version Number of the SCAD file takes priority and overwrites the Version Number in the JSON File (acting as "last saved with version" indicator)
- The Version Number of the json files takes priority, allowing the SCAD file to respond to legacy presets

which one do you want/need?

Allowing the scad file to both respond to the version in the JSON and to overwrite it does not work, as this would require true variables.

MichaelAtOz, 2018-09-03:
I'm taking about
```
/* [Global] */
g1=1;

/* [Hidden] */
h1=1;
h2="h2";
```

These variables are not saved to the json file.

I may have missed some of the discussion, is there a reason why?
Or are you both talking about something else?

t-paul, 2018-09-03:
If `g1` is missing, I'd call that a bug. Everything in "Hidden" is declared as not being shown to the user, I think it then should also not be saved as there is no way to change that value, not even by changing it in the source code after selecting a preset which has that variable.

I don't think it's a good idea to make other tab names special as that can potentially collide with Thingiverse.

All that applies to the "thingiverse way of defining customization via comments". If we add some native markup, we can define anything we want, as we own the specification.

For reference, [Thingiverse docs](http://customizer.makerbot.com/docs) describe to those sections as:

* *[Global]* Parameters in the global tab will always be shown on every tab no matter which tab is selected. Note: there will be no tab for global params, they will just always be shown.
* *[Hidden]*  Parameters in the hidden tab will never be displayed. Not even the tab will be shown. This provides a more natural way to have parameters in your code that you don't want the user to see.

MichaelPFrey, 2018-09-04:
there are for different possible implementation for hidden variables:

1. hidden variables are stored and retrieved from the json file
1. hidden variables **NOT stored** in json file, but are retrieved from the json file
1.  hidden variables are stored in the json file, but are **NOT retrieved** from the json file
1. hidden variables are **neither** stored and  nor retrieved from the json file

Originally, the OpenSCAD customizer has behaved as in 1.

Currently, the OpenSCAD customizer behaves as in 2. The customizer applying variables from the hidden group when found in a json file does not make sense.

Implementing 3 would be a reasonable compromise. Hidden variables are not applied, but still stored. Meaning: If a once hidden variable becomes visible, it also becomes applicable. This allows a designer to use the hidden section for reserved variables, that become customizable (and assigend with a different default) in a future version, without breaking existing preset. 

Implementing 4 would be consequent.

MichaelAtOz, 2018-09-05:
What's wrong with 1.?

t-paul, 2018-09-05:
It creates variables whose values can't be changed in scad source anymore once the value is stored in the parameter set and that set is selected.

MichaelAtOz, 2018-09-07:
As there is now a Default set, how about a From Source Code set?
Then it can be 1. which is useful IMO, and you can load the source values easily to over rule anything.

t-paul, 2018-09-07:
I think that is exactly what the "design default" is - the values from the source code.