openscad/openscad #2451: update the documentation for $preview

MichaelPFrey, 2018-08-18:
- [ ]  http://www.openscad.org/cheatsheet/ https://github.com/openscad/openscad.github.com/blob/master/cheatsheet/index.html
- [ ]  https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Other_Language_Features#$preview
- [ ] https://github.com/openscad/openscad/blob/master/contrib/OpenSCAD.plist

for reference, see
   * #1963
   * #149
   * 2429

nophead, 2018-08-18:
Not sure what the Test case is for. There is no interaction between $preview and render(). $preview is only set once at the top scope.

MichaelPFrey, 2018-08-18:
@nophead you are right of course. Still it is worth a regression test. The reason why I would like to have a regression test for that is to make that behavior documented and prevent it from accidentally (or due to an misunderstanding) being changed later.

To some extend, all regression test only test obvious and seemingly simple things - but in summary, they catch a lot of issues.

MichaelPFrey, 2018-09-02:
Thank you very much.
I would like to keep the issue open until the Pull Request are merged just to keep track.

nophead, 2018-09-06:
They are all merged now except the cheat sheet, so you can tick off three of the four tasks.

I am not sure if the live version of the cheat sheet is supposed to match the latest release or the current snapshot. Since the release is so old I think it would be better merging it sooner.

t-paul, 2018-10-06:
Cheet sheet changes moved to [snapshot.html](http://www.openscad.org/cheatsheet/snapshot.html), so all done now, thanks.

closed 2018-10-06 by t-paul