openscad/openscad #2453: Customizer:Windows-dataset not read nor written if directory name use accentuated letter

PRouzeau, 2018-08-19:
tag: [Customizer][bug]
Platform : Windows 10: not tested on another platform
Testing my last app, I expanded it in a directory of the form (note the accent on the e):
```
D:/whatever/vélo/
```
My app woks perfectly well, but I cannot see any of the prerecorded dataset, and if I create new datasets, they will be lost at next opening and json file remains unmodified.
Now, changing the directory name to 
```
D:/whatever/velo/
```
and everything is fine...
That fails also if the accentuated letter is used in an upper directory.

t-paul, 2018-08-19:
I think the general issue is converting (or lack of) from local encoding (e.g. filesystem) to UTF-8 and back. This tends to fail on Windows more often due to the file system usually not using UTF-8.

donbright, 2018-09-03:
it is most definitely a duplicate... boost is very misleading about what its filesystem module can and cannot do. it does not provide any help with the encoding of filenames, which is the problem here.