openscad/openscad #2458: snap version fails to create symlinks

cbednarski, 2018-08-24:
I just installed the nightly on Ubuntu 18.04. It seems like there is a circular symlink reference somewhere in the snap package.

Step to repro:

```
sudo snap install openscad-nightly
snap run openscad-nightly -h
ln: failed to create symbolic link '/home/username/snap/openscad-nightly/8/snap/openscad-nightly/8/Documents': No such file or directory
ln: failed to create symbolic link '/home/username/snap/openscad-nightly/8/snap/openscad-nightly/8/Desktop': No such file or directory
ln: failed to create symbolic link '/home/username/snap/openscad-nightly/8/snap/openscad-nightly/8/Downloads': No such file or directory
ln: failed to create symbolic link '/home/username/snap/openscad-nightly/8/snap/openscad-nightly/8/Music': No such file or directory
ln: failed to create symbolic link '/home/username/snap/openscad-nightly/8/snap/openscad-nightly/8/Pictures': No such file or directory
ln: failed to create symbolic link '/home/username/snap/openscad-nightly/8/snap/openscad-nightly/8/Videos': No such file or directory
ln: failed to create symbolic link '/home/username/snap/openscad-nightly/8/snap/openscad-nightly/8/Templates': No such file or directory
...
```

Notice in the path that `/snap/openscad-nightly/8` is repeated. It appears the correct path should be e.g. `/home/username/snap/openscad-nightly/8/Templates`.

Details:

```
$ snap version
snap    2.34.3
snapd   2.34.3
series  16
ubuntu  18.04
kernel  4.15.0-32-generic
```
```
$ openscad-nightly --version
...
OpenSCAD version 2018.06.10
```
```
$ snap info openscad-nightly
...
snap-id:      U0G6xWcDnmIByhlnpmuxM2zmzsLwmm60
tracking:     stable
refresh-date: today at 16:44 PDT
channels:                            
  stable:    0+git.c6a4856 (8)  89MB -
  candidate: ↑                       
  beta:      ↑                       
  edge:      0+git.ca6d36d (58) 90MB -
installed:   0+git.c6a4856 (8)  89MB -
```

t-paul, 2018-08-24:
Could that be caused by the snap package being installed in the home directory? The OpenSCAD snap uses the `home` plugin which might be the reason for that effect.

I don't see the issue on my Debian system (same snap version) which installs to `/snap`.

cbednarski, 2018-08-26:
Thanks! Ubuntu does indeed install snaps to `~/snap`, but beyond that I don't really know anything about snap/d. Maybe I would need to read the developer docs for snap to figure out how to troubleshoot this, but I ended up using the AppImage version on the website instead.

t-paul, 2018-09-16:
Snap version `0+git.5b737cb (131)` is now working on Ubuntu 18.04.

closed 2018-09-16 by t-paul

dprangnell, 2024-06-26:
This still ongoing with RHEL 9:
```
# snap version
snap    2.63-0.el9
snapd   2.63-0.el9
series  16
rhel    9.4
kernel  5.14.0-427.22.1.el9_4.x86_64
```

```
# cat /etc/os-release 
NAME="Red Hat Enterprise Linux"
VERSION="9.4 (Plow)"
ID="rhel"
ID_LIKE="fedora"
VERSION_ID="9.4"
PLATFORM_ID="platform:el9"
PRETTY_NAME="Red Hat Enterprise Linux 9.4 (Plow)"
ANSI_COLOR="0;31"
LOGO="fedora-logo-icon"
CPE_NAME="cpe:/o:redhat:enterprise_linux:9::baseos"
HOME_URL="https://www.redhat.com/"
DOCUMENTATION_URL="https://access.redhat.com/documentation/en-us/red_hat_enterprise_linux/9"
BUG_REPORT_URL="https://bugzilla.redhat.com/"

REDHAT_BUGZILLA_PRODUCT="Red Hat Enterprise Linux 9"
REDHAT_BUGZILLA_PRODUCT_VERSION=9.4
REDHAT_SUPPORT_PRODUCT="Red Hat Enterprise Linux"
REDHAT_SUPPORT_PRODUCT_VERSION="9.4"
```

```
# snap run unison-jz.unison -help
ln: failed to create symbolic link '/root/snap/unison-jz/19/snap/unison-jz/19/Documents': No such file or directory
ln: failed to create symbolic link '/root/snap/unison-jz/19/snap/unison-jz/19/Desktop': No such file or directory
ln: failed to create symbolic link '/root/snap/unison-jz/19/snap/unison-jz/19/Downloads': No such file or directory
ln: failed to create symbolic link '/root/snap/unison-jz/19/snap/unison-jz/19/Music': No such file or directory
ln: failed to create symbolic link '/root/snap/unison-jz/19/snap/unison-jz/19/Pictures': No such file or directory
ln: failed to create symbolic link '/root/snap/unison-jz/19/snap/unison-jz/19/Videos': No such file or directory
ln: failed to create symbolic link '/root/snap/unison-jz/19/snap/unison-jz/19/Templates': No such file or directory
```