openscad/openscad #2459: linear_extrude() with angle (XY displacement of top face)

pwnage2apps, 2018-08-24:
My suggestion is to allow to change the direction of a linear_extrude(), instead of "z only", allow to send a vector where Z is the current height and X and Y are the displacement of the end of extrusion. For example: [3,2,5] will create a extrusion where the top face is displaced 3 and 2 units in X and Y, the height is 5 units.

I know that there is a "generalized extrusion" proposal, but my idea would be more user friendly:
https://github.com/openscad/openscad/issues/114

The only problem is how to manage when Z is 0.

At this moment, we can use "twist" from a very far point, but it's leaves a small twist.

thehans, 2018-09-06:
If i'm understanding correctly, what you want would be the same as a regular linear_extrude plus a skew operation.  There's no native skew module but you can perform skew operation by using a transform matrix, documented here: https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Transformations#multmatrix

pwnage2apps, 2018-09-06:
Thanks, it worked well.

l29ah, 2018-09-06:
@pwnage2apps can you show the code you ended up using?

pwnage2apps, 2018-09-07:
Yes, but I haven't uploaded the latest version yet:
https://www.thingiverse.com/thing:3043241

I hope that I can upload it tomorrow, I will notify you.

pwnage2apps, 2018-09-10:
I have updated the code, the multimatrix is at line 181:
https://www.thingiverse.com/thing:3043241

t-paul, 2019-11-18:
Closing as it's working without core language extensions.

closed 2019-11-18 by t-paul