openscad/openscad #2461: Assert failure while loading customizer preset

nyponen, 2018-08-26:
This occurred while loading a customizer preset on OpenSCAD version 2018.04.01 (git e7c0851)

[image]

t-paul, 2018-08-26:
Please add some example files to reproduce.

nyponen, 2018-08-26:
This occurs randomly while switching between different presets. I can't reproduce it at will.

t-paul, 2018-08-26:
So it happens with different models at random times when changing the preset selection box to a different value?

nyponen, 2018-08-26:
> So it happens with different models at random times when changing the preset selection box to a different value?

Yes.

t-paul, 2018-08-26:
Please check with latest snapshot version which has a fix that could be related (#2404).

nyponen, 2018-08-26:
I have a different error with the latest snapshot. When I open a preset, an asterisk appears next to the name in the drop down list, and if I try to load a different preset, I get this dialog box, even though I made no changes to the preset. When I tried to simplify the code to make a test case, the problem went away.

[image]

t-paul, 2018-08-26:
Maybe the code has a new variable that is not yet in the presets? Would it be possible to share that original code?

nyponen, 2018-08-26:
No, the code didn't have any new variables in it. This simplified code shows the problem on my WIndows 10 machine. Here's the .scad file:
`/* [1] */
Element1 = true;

/* [2] */
Elements = 12;              // [1:1:34]   

design( 
    element1 = Element1 
);`

Here's the .json file:
`{
    "parameterSets": {
        "Feb22": {
            "Elements": "13",
            "outer_diameter": "30"
        }
    },
    "fileFormatVersion": "1"
}
`