openscad/openscad #2463: Recent AppImage binaries fail with "no version information available"

cbednarski, 2018-08-28:
I downloaded several AppImage builds from <https://download.opensuse.org/repositories/home:/t-paul/AppImage/>, dated `26-Aug-2018`. When executed they all fail with the following error:

```
$ ./OpenSCAD-Release-0-Build35.32.glibc2.14-x86_64.AppImage 
openscad: openscad: no version information available (required by openscad)
openscad: openscad: no version information available (required by openscad)
...
some variable number of repeated lines
...
openscad: openscad: no version information available (required by openscad)
openscad: openscad: no version information available (required by openscad)
openscad: relocation error: openscad: symbol  version  not defined in file  with link time reference
```

Here are the binaries I tried.
```
$ find . -name "OpenSCAD*"
./OpenSCAD-Branch-InputDriver-0-Build13.2.glibc2.14-x86_64.AppImage
./OpenSCAD-Release-0-Build35.32.glibc2.14-x86_64.AppImage
./OpenSCAD-Nightly-0-Build2306.1.glibc2.14-x86_64.AppImage
```

I have an older version from Jan 14/15 that works: `OpenSCAD-Branch-InputDriver-0-Build8.42.glibc2.14-x86_64.AppImage`, so I don't think there's anything untoward with my OS (I'm using Ubuntu 18.04).

t-paul, 2018-08-28:
See https://github.com/probonopd/linuxdeployqt/issues/313

karlvirgil, 2018-08-30:
I'm having this issue too on Centos 7
openscad: openscad: no version information available (required by openscad)
openscad: relocation error: openscad: symbol , version  not defined in file  with link time reference
make: *** [create] Error 127

t-paul, 2018-09-03:
Technically we can, but it's quite some extra effort and we lose the ability to create ARM images (which is currently broken on OBS too after Qt5 switch, but it did work in the past).

Maybe we can update our systems sometime and see if we can get stuff (cross-)built on Docker/Vagrant/Qemu via Jenkins or our own OBS instance. Basically I don't want to maintain lots of different systems with lots of different build styles.

The public hosted OBS has some limitations regarding what can be built, but it's still a great service with normally low maintenance effort - if there's a new distribution out, I just click on the repo selection and all what's needed in addition is to add it to our download page :-).

donbright, 2018-09-03:
@t-paul i see, bummer. https://github.com/AppImage/AppImageKit/wiki/Creating-AppImages ... not exactly 'configure && make install'

closed 2018-09-10 by t-paul