openscad/openscad #2477: UBUNTU 18.04 Fails to compile

salvato, 2018-09-06:
`gdate.c: In function 'g_date_strftime':
gdate.c:2497:7: error: format not a string literal, format string not checked [-Werror=format-nonliteral]
       tmplen = strftime (tmpbuf, tmpbufsize, locale_format, &tm);
       ^~~~~~
cc1: some warnings being treated as errors
Makefile:1782: recipe for target 'libglib_2_0_la-gdate.lo' failed
`

salvato, 2018-09-06:
Yes it fails in running:

./scripts/uni-get-dependencies.sh

t-paul, 2018-09-06:
That's not building, just trying to load dependencies - which would be the ok (you probably will also need to install `qtmultimedia5-dev`, that patch is not merged yet).

I guess you are running `uni-build-dependencies.sh`. There should be only some rare cases where that is really needed. Using `-dev` packages from the distribution should be fine.

Note that if you just want to install the latest version, you can add the snapshot repo. This will install `openscad-nightly` which is always automatically updated against the latest code on github (master).

salvato, 2018-09-07:
Thank you very much. I'll try to install the nightly build asap.

closed 2018-09-07 by salvato