openscad/openscad #2482: Unknown errors when checking the STL file

frankenstein91, 2018-09-08:
I have these two errors and I don't know how to solve them before printing. Could you please help me? I used this website for the check: https://www.3dhubs.com/3dprint#/

The file is https://github.com/frankenstein91/number-plate-eGolf/blob/hotfix/IntersectingFaces/nrPlate.scad

[image]

Thank you for the help

nophead, 2018-09-10:
I took the liberty of simplifying it a bit to speed up the rendering.
```
 $fa = 6;
$fs = 0.25;
car = "eGolf";
carnumber = 987654;
difference() {
    union(){
        union() {
            linear_extrude(height = 1){
                text(text=str(car));
                
                translate([0.3, -10])
                    text(text=str("Number: ",carnumber));

            }
            translate(v = [-1, -13, -2])
                color([1,0,0]) 
                    linear_extrude(height = 2.01)
                        offset(2)
                            square([105,26]);
        }
        translate([45,5]){
            union(){
                translate([0,1.5])
                    cube([3,1,1]);

                translate([0,-1.5])
                    cube([3,1,1]);
                
                hull() {
                    translate([3,-3])
                        cube([1,7,1]);

                    translate([7,0.5])
                        cylinder(h = 1, r = 3.5);
                }
                //        Kabel
                translate([10,-4]) {
                    difference() {
                        sector(1, 10, 30, 90);
                        
                        translate([0,0,-1]) sector(4, 8, 29, 92);
                    }
                    translate([7.67,4.7,0]) rotate([0, 0, 180]) {
                        difference() {
                            sector(1, 10, 30, 147);
                            
                            translate([0,0,-1]) sector(4, 8, 29, 148);
                        }
                    }
                }
            }
        }
    }

    translate([3,9,-2.7]) cylinder(h=1, d=6.7);
    translate([3,-9,-2.7]) cylinder(h=1, d=6.7);
    translate([100,9,-2.7]) cylinder(h=1, d=6.7);
    translate([100,-9,-2.7]) cylinder(h=1, d=6.7);
}

//https://gist.github.com/plumbum/78e3c8281e1c031601456df2aa8e37c6
module sector(h, d, a1, a2) {
    if (a2 - a1 > 180) {
        difference() {
            cylinder(h=h, d=d);
            
            translate([0,0,-0.5]) sector(h+1, d+1, a2-360, a1); 
        }
    } else {
        difference() {
            cylinder(h=h, d=d);
            
            rotate([0,0,a1]) translate([-d/2, -d/2, -0.5])
                cube([d, d/2, h+1]);
            
            rotate([0,0,a2]) translate([-d/2, 0, -0.5])
                cube([d, d/2, h+1]);
        }
    }
} 
```

When viewed in NetFabb it reports 4 holes but I can only see one between the plug and the colon. Removing either the plug body, or the colon in the text fixes it. I changed the plug to be a hull rather than a union of two objects touching but that still doesn't fix it.

It seems to be a bug in OpenSCAD.

[image]

nophead, 2018-09-10:
Removing the cable also fixes the holes and the cable on its own is two volumes, so that could the real issue.

nophead, 2018-09-10:
Here is a version that works. I re-implemented sector but I don't really know why it fixes it.

```
$fa = 6;
$fs = 0.25;
car = "eGolf";
carnumber = 987654;
difference() {
    union(){
        linear_extrude(height = 1){
            text(text=str(car));
            
            translate([0.3, -10])
                text(text=str("Number: ",carnumber));
        }
        translate(v = [-1, -13, -2])
            color([1,0,0]) 
                linear_extrude(height = 2.01)
                    offset(2)
                        square([105,26]);
        
        translate([45,5]){
            translate([0,1.5])
                cube([3.5,1,1]);

            translate([0,-1.5])
                cube([3.5,1,1]);
            
            hull() {
                translate([3,-3])
                    cube([1,7,1]);

                translate([7,0.5])
                    cylinder(h = 1, r = 3.5);
            }
            //        Kabel
            translate([10,-4]) {
                difference() {
                    sector(1, 10, 30, 90);
                    
                    translate([0,0,-1]) sector(4, 8, 29, 92);
                }
                translate([7.67,4.7,0]) rotate([0, 0, 180]) {
                    difference() {
                        sector(1, 10, 30, 147);
                        
                        translate([0,0,-1]) sector(4, 8, 29, 148);
                    }
                }
            }
        }
    }

    translate([3,9,-2.7]) cylinder(h=1, d=6.7);
    translate([3,-9,-2.7]) cylinder(h=1, d=6.7);
    translate([100,9,-2.7]) cylinder(h=1, d=6.7);
    translate([100,-9,-2.7]) cylinder(h=1, d=6.7);
}

module pie_slice(r, start_angle, end_angle) {
    R = r * sqrt(2) + 1;
    a0 = (4 * start_angle + 0 * end_angle) / 4;
    a1 = (3 * start_angle + 1 * end_angle) / 4;
    a2 = (2 * start_angle + 2 * end_angle) / 4;
    a3 = (1 * start_angle + 3 * end_angle) / 4;
    a4 = (0 * start_angle + 4 * end_angle) / 4;
    if(end_angle > start_angle)
        intersection() {
        circle(r);
        polygon([
            [0,0],
            [R * cos(a0), R * sin(a0)],
            [R * cos(a1), R * sin(a1)],
            [R * cos(a2), R * sin(a2)],
            [R * cos(a3), R * sin(a3)],
            [R * cos(a4), R * sin(a4)],
       ]);
    }
}

module sector(h, d, a1, a2)
    linear_extrude(height = h)
        pie_slice(d / 2, a1, a2);
```

nophead, 2018-09-10:
A version that does as much as possible in 2D before extruding to 3D. That is much faster and safer and makes sense for a 2.5D design like this.

```
$fa = 6;
$fs = 0.25;
car = "eGolf";
carnumber = 987654;
union(){
    difference() {
        translate([-1, -13, -2])
            color([1,0,0]) 
                linear_extrude(height = 2.01)
                    offset(2)
                        square([105, 26]);
        
         translate([0, 0, -2.7])
            linear_extrude(height = 1) {
                translate([3, 9]) circle(d=6.7);
                translate([3, -9]) circle(d=6.7);
                translate([100, 9]) circle(d=6.7);
                translate([100, -9]) circle(d=6.7);
            }
        }
        linear_extrude(height = 1){
            text(car);
            
            translate([0.3, -10])
                text(str("Number: ", carnumber));
            
            translate([45, 5]){
                for(side = [-1, 1])
                    translate([0, side * 1.5])
                        square([3.5, 1]);
                
                hull() {
                    translate([3, -3])
                        square([1, 7]);

                    translate([7, 0.5])
                        circle(3.5);
                }
                //        Kabel
                translate([10, -4]) {
                    difference() {
                        pie_slice(5, 30, 90);
                        
                        pie_slice(4, 29, 92);
                    }
                    translate([7.67, 4.7]) rotate(180) {
                        difference() {
                            pie_slice(5, 30, 147);
                            
                            pie_slice(4, 29, 148);
                        }
                    }
                }
            }
        }
}

module pie_slice(r, start_angle, end_angle) {
    R = r * sqrt(2) + 1;
    a0 = (4 * start_angle + 0 * end_angle) / 4;
    a1 = (3 * start_angle + 1 * end_angle) / 4;
    a2 = (2 * start_angle + 2 * end_angle) / 4;
    a3 = (1 * start_angle + 3 * end_angle) / 4;
    a4 = (0 * start_angle + 4 * end_angle) / 4;
    if(end_angle > start_angle)
        intersection() {
        circle(r);
        polygon([
            [0,0],
            [R * cos(a0), R * sin(a0)],
            [R * cos(a1), R * sin(a1)],
            [R * cos(a2), R * sin(a2)],
            [R * cos(a3), R * sin(a3)],
            [R * cos(a4), R * sin(a4)],
       ]);
    }
}
```

frankenstein91, 2018-09-16:
thank you for your help.
Maybe you could add it as a fork, that your work is comprehensibly for others?

nophead, 2018-09-16:
Sorry I don't understand. A fork of what?

frankenstein91, 2018-09-16:
> Sorry I don't understand. A fork of what?

of the small github projekt. I don't like to just copy your solution, so it would be very good for me as a fork and then PullRequest.

nophead, 2018-09-16:
I have done as you requested but do you know  you can create your own pull requests? If you simply edit the file on GitHub and paste in my changes you can select to make a pull request rather than updating it directly. It would have achieved the same effect without me having to fork a project that I am not involved in. The only difference would be the PR came from you instead of me.

frankenstein91, 2018-09-16:
that's what my goal was, to have your name recognized in the project.

closed 2018-09-16 by frankenstein91