openscad/openscad #2486: Openscad - 2D Border in SVG files

Ziggy2013, 2018-09-19:
Re : version 2015.03-2 

When using Openscad to create 2D files, exporting the 2D content to an SVG file automatically adds a 0.5mm border around all of the 2D objects in the SVG file.  The SVG xml which generates the border is normally the last line in the SVG file

" stroke="black" fill="lightgray" stroke-width="0.5"/></svg>

This means that when the SVG file is opened in a vector based editing program (eg Inkscape), the actual dimensions of all 2D objects are not correct. Consequently Openscad created 2D SVG files can not be used accurately by laser cutters/engravers etc without  direct modification of the SVG file xml to remove the border.

A suggested solution is to remove the 0.5mm border entirely when exporting SVG files (or make adding the border an option).

nophead, 2018-09-19:
A laser cutter will normally cut down the centre of the stroke, so the size is accurate, less the kerf of the laser beam.

In Inkscape you have the choice of whether it includes the boarder width in the reported dimensions. It's a setting that can be changed.

nophead, 2018-09-19:
Edit / Preferences / Tools / Geometric bounding box.

Ziggy2013, 2018-09-20:
It is not clear to me the reason openscad automatically adds a border of 0.5mm to a 2D object exported to an SVG file. 

Of course the border can be removed or compensated for in some vector drawing apps and  by some laser cutters. But why add the border in the first place? 3D objects exported as STL files do not have anything like a border added. Perhaps someone can explain why the border is necessary for 2D objects?

Rendering a 2D object in openscad does not show the border. The border is only added during the export to SVG and there is no control over the width of the border.

IMO the 2D border should be an export option.

Ziggy2013, 2018-09-20:
Chris
I understand your point that openscad is currently not perfect for all 2D objects.  It is however, very useful for some 2D objects because of its programming like features.

Even given the openscad 2D limitations, I still don't see any reason why openscad should automatically add an arbitary 0.5mm border only when exporting to SVG.  IMO adding an arbitrary border doesn't make sense and is a potential source of problems if the SVG is used directly or as a source to a CAM layer. 

In my case, as an interim solution I have simply patched openscad.exe to stop the border being automatically added to SVG.

Ziggy2013, 2018-09-20:
I used the term "border" because of its appearance on the Inkscape screen (ie a 0.5mm wide black border around all 2D objects).  Why pick 0.5mm as a stroke width? Even a stroke width of 0.0mm presents well in Inkscape.

in my case when the svg file was saved in Inkscape and sent to a manufacturer, the dimensions they saw on their systems (CorelDraw based) were wrong. (BTW they requested SVG). The dimensions did not agree with the openscad source and were wrong for both their membrane switch printer and laser cutter. Perhaps the "Geometric bounding box" option in Inkscape may also have been an issue?

nophead, 2018-09-20:
Did you attempt to change the size in InkScape? If so not having it in geometric bounding mode would make the actual size smaller than it displays. If you didn't change the dimension I would expect it to be correct for a laser cutter. Is the membrane switch printer a graphics printer or a cutter? If it actually prints an image I would expect the boarder to be included in the output.

Did they actually cut from the file or just report the dimensions looked wrong in Corel?

Yes a stroke with of 0 will leave the object visible if it also has a fill.

Ziggy2013, 2018-09-20:
No. There were no Inkscape changes to the imported svg background. Only two icons (also imported svg files) were added before the svg was saved in Inkscape. The membrane switch printer is basically a precision colour digital printer.
This manufacturer has a process of using the source SVG to prepare detailed Corel artwork for approval prior to production. That artwork showed incorrect dimensions (not one dimension was correct). Hence the investigation as to why?.

The issue of "stroke width" in svg prompted me to check Inkscape. It is possible to easily vary the stroke width using the "fill and stroke dialog". Inkscape actually sees the stroke width of an openscad svg file as 0.5 pixels not mm (ie 0.5px = 0.132mm). Of course the stroke width can be changed  to zero, in which case the "border" disappears.

So in openscad a potential solution is to provide an option to specify the stroke width when exporting to an svg file. The stroke width value should be in the same mm units as openscad.

t-paul, 2018-09-20:
Can you check if this maybe has nothing to do with the stroke parameter but with #1597 instead.

Ziggy2013, 2018-09-20:
"Can you check if this maybe has nothing to do with the stroke parameter but with #1597 instead."

I believe the svg xml specifies the correct units (mm) for the 2D objects. However the stroke width is in pixels.

nophead, 2018-09-20:
No the stroke width and all the other numbers are interpreted as mm if the height and width are defined in mm, unless they specify their own units.

If I open an SVG created with OpenSCAD in InkScape with the preference set to Geometric then it reports the correct dimensions. If it is set to Visual then it adds 0.5mm to the dimensions. If you send it to a laser cutter it should use the geometric dimensions of the vector path. If you send it to a printer or a screen it should print the 0.5mm border, so it will occupy 0.5mm more.

[image]

```
<?xml version="1.0" standalone="no"?>
<!DOCTYPE svg PUBLIC "-//W3C//DTD SVG 1.1//EN" "http://www.w3.org/Graphics/SVG/1.1/DTD/svg11.dtd">
<svg width="10mm" height="10mm" viewBox="0 -10 10 10" xmlns="http://www.w3.org/2000/svg" version="1.1">
<title>OpenSCAD Model</title>
<path d="
M 0,-0 L 10,-0 L 10,-10 L 0,-10 z
" stroke="black" fill="lightgray" stroke-width="0.5"/>
</svg>
```

[image]

t-paul, 2018-09-20:
And the release version does generate the SVG below. Note the `width="10" height="10"`.

```
<?xml version="1.0" standalone="no"?>
<!DOCTYPE svg PUBLIC "-//W3C//DTD SVG 1.1//EN" "http://www.w3.org/Graphics/SVG/1.1/DTD/svg11.dtd">
<svg width="10" height="10" viewBox="0 -10 10 10" xmlns="http://www.w3.org/2000/svg" version="1.1">
<title>OpenSCAD Model</title>
<path d="
M 0,-0 L 10,-0 L 10,-10 L 0,-10 z
" stroke="black" fill="lightgray" stroke-width="0.5"/></svg>
```

nophead, 2018-09-22:
Yes 2015.03-2 does not specify units, so all dimensions will be in pixels. And the number of pixels per inch varies between different versions of InkScape. Yo need a recent snapshot to get correct dimensions in mm. 

I don't think the stroke width is a problem. It is ignored by laser cutters.

Ziggy2013, 2018-09-22:
Stroke width may be ignored by some laser cutters but not by  some manufacturer's laser cutting preprocessing (as I experienced). In any case laser cutting is a rather narrow application of openscad svg files.

Stroke width is not ignored by graphics printers.

t-paul, 2018-09-22:
I'm going to close this now, @nophead already explained the current situation with great detail and stroke with is specifically mentioned as option for SVG in #649 so that will be covered via that issue.

closed 2018-09-22 by t-paul

UltraMej, 2020-10-05:
Hi (2 years latter),

Actually, since in OpenSCAD all primitives have to be closed volumes or closed polygons, line shouldn't exist (only surfaces)... no?

By no existing, I mean that the default line thickness (when OpenSCAD exports in svg) should be "Lines of zero width" aka "vector-effect=non-scaling-stroke" aka "line width= Hairline". It is a type of line that are always visible but never get any thickness no matter how much you zoom on it.

NB1: When using Epilog or Trotec laser-cutter if you want a line to be cut, it has to be "zero width" or "infinitely thin" or "hairline" depending on the editor you are using. I dunno about the model of laser-cutter used by Ziggy2013.

NB2: In https://gitlab.com/inkscape/inbox/-/issues/1922 they actually say that "non-scaling-stroke" is a SVG 2.0-feature but not a SVG 1.2, also it is supported by Inkscape... I didn't even know there was diferent svg "version". What is the one used by OpenSCAD svg export?..