openscad/openscad #2487: Segfault when rendering an imported stl with a difference

JPFrancoia, 2018-09-20:
Hi all,

I've a problem rendering a simple scad file. The file contains a stl import and a difference.

Here is the imported stl file:

[bord_male.stl.txt](https://github.com/openscad/openscad/files/2401109/bord_male.stl.txt)

and here is the scad code:

```
$fn = 75;

difference(){
    resize([44.3967, 3.5, 30]) import("piece.stl");
    color([1, 0, 0]) translate([-60, 1.4, -60]) rotate([90, 0, 0]) cylinder(h=1.5, r=2.25);
}
```
When I try to render, here is what I get:

```
openscad bord_male_hole.scad -o bord_male_hole.stl  
[1]    11711 segmentation fault (core dumped)  openscad bord_male_hole.scad -o bord_male_hole.stl
```

I get the segfault with openscad 2015.03.03-17 installed from the community repo on Archlinux, and from openscad git (20180713-1).

What can I do to solve the issue?

Cheers

t-paul, 2018-09-20:
It does not seem to crash with the AppImage, so the crash may need to reported to Archlinux.

It still does not product the expected model due to some error in mesh calculation. I suspect the STL has some issue - see [FAQ](https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/FAQ) for some possible way to fix those.

JPFrancoia, 2018-09-20:
The stl was indeed faulty. Thank you for pointing me in the right direction.

closed 2018-09-20 by JPFrancoia